lvgl/lvgl #2899: lv_cont_create replacement in 8.1

elgerg, 2021-12-14:
### Perform all steps below and tick them with [x]
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I am (was) using:
lv_cont_create(lv_disp_get_scr_act(NULL), NULL);
but that doesnt work in 8.1
What is the replacement for this in 8.1?

### To Reproduce
```c
lv_cont_create(lv_disp_get_scr_act(NULL), NULL);
```

### Expected behavior
Creation of a container 

### Screenshots or video

Miguel0101, 2021-12-14:
Use `lv_obj_create`. Now the lv_cont function has been replaced by lv_obj.

kisvegabor, 2021-12-14:
> Use lv_obj_create. Now the lv_cont function has been replaced by lv_obj.

Confirmed.

closed 2021-12-14 by kisvegabor

elgerg, 2021-12-14:
Thanks, is the same for pages?

Thanks
Alex

embeddedt, 2021-12-16:
Yes; scrolling is now part of `lv_obj` as well.

embeddedt, 2022-09-05:
@shahawi-sumup I believe it was mentioned in the release changelog for v8 [here](https://docs.lvgl.io/master/CHANGELOG.html#widget-changes).

> lv_page removed, scroll features are moved to lv_obj