lvgl/lvgl #2900: Merging lv_demos into lvgl

kisvegabor, 2021-12-14:
In the name of simplifying the idea of merging lv_demos to lvgl came up [here](https://github.com/lvgl/lvgl/issues/2790#issuecomment-976674652). 
I agree with that as the only costs are the larger disk space usage and longer cloning time. 

The only thing I can't decide is where to move lv_demos?
1. into root, next to src, scripts, tests, ...
2. into src

1) would be cleaner but 2) would allow e.g. for Arudino users to call the demos directly (Ardunio complies only the src folder).

I'm not sure if it's worth choosing 2) only for this specific case. It just came to my mind we can tell Ardunio users to simply manually copy/paste the demos and examples folder into src. If they update lvgl they need to copy/paste again but it doesn't happen that often. 

What do you think?

cc @xiaoxiang781216, @embeddedt

embeddedt, 2021-12-15:
I'm fine with either.

kisvegabor, 2021-12-16:
I've merged to the root.

A few things are still missing:
- Update make
- Update CMake
- Update Kconfig
- Use demos in tests

kisvegabor, 2021-12-16:
@xiaoxiang781216 Adding the demo configs to lv_conf_template.h seemingly revealed  an inconsistency between `astyle` and `lv_conf_internal_gen.py` again. Could you take a look?

embeddedt, 2022-01-04:
@kisvegabor I updated the docs so the demos are included in the Emscripten binary, so the CMakeLists file also supposrts them now. However, I wasn't able to understand what needed to be added in `docs/example_list.py` to get them to show up in the examples list on the website.

kisvegabor, 2022-01-05:
Great! Do we need `index.rst` files fro the demos too?

embeddedt, 2022-01-05:
I think so. I should probably have done a better job of documenting this back when I set it up for v7/v8... :laughing:

kisvegabor, 2022-01-05:
Familiar feeling... If git didn't show that something was written by me, I'd disown it :laughing: 

---

Is there something I can help with?

kisvegabor, 2022-01-20:
The build related things are added and only "Use demos in tests" remained. 

In tests (at least in the first iteration) I think we should do 2 things:
- call `lv_demo_widgets()` to see if it can be built and run
- call `lv_demo_stress()` to see if it can be built and let run for a few seconds and check for memory leaks

kisvegabor, 2022-01-24:
@embeddedt 
With #3037 nothing else left, right?

embeddedt, 2022-01-24:
I think that's everything, yes.

closed 2022-01-24 by embeddedt