lvgl/lvgl #2903: distorted pixel lines at animation, STM32F

epikao, 2021-12-15:
Hello

I have always distorted pixel lines with your examples, especially with animations, see pictures below. I have no idea why, but I think it is because of the simple STM32 (F746) driver code, see code below. Does somebody have a better/another drivercode?

```
void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
  int16_t x1, y1,x2,y2;
	x1=area->x1;
	x2=area->x2;
	y1=area->y1;
	y2=area->y2;
	if (x2>479)
		x2=479;
	if (y2>271)
		y2=271;
	if (y1<0)
		y1=0;
	if (y2<0)
		y2=0;
	if (x1<0)
		x1=0;
	if (x2<0)
		x2=0;
	//lcd_fb = ( uint16_t*) (0xC0000000);
	for(int y = y1; y <= y2; y++) {
		for(int x = x1; x <= x2; x++) {
			tft.buffer[x+(y * 480)]=(color_p++->full);  //pixelColor;
		}
	}
	lv_disp_flush_ready(disp);
}

```
[image]
[image]
[image]

embeddedt, 2021-12-15:
I'm not 100% sure how to fix it in your case, but it looks like exactly the type of glitched rendering I see when the CPU cache is not being invalidated at the right time or isn't configured correctly.

epikao, 2021-12-15:
> I'm not 100% sure how to fix it in your case, but it looks like exactly the type of glitched rendering I see when the CPU cache is not being invalidated at the right time or isn't configured correctly.

ok, so it looks it is more something with the STM32 setup (LTDC) and not with the driver code above?

epikao, 2021-12-20:
solved, see: https://github.com/lvgl/lv_demos/issues/143

closed 2021-12-20 by epikao