lvgl/lvgl #2905: Code stops mid startup

elgerg, 2021-12-15:
### Perform all steps below and tick them with [x]
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I am porting my code from 7.5 to 8.1
I have a loop that sets up my weather output for 20 segments. However on the 17th loop it just stops. no errors or anything,  

### To Reproduce
```c
    for (int i = 1; i < 20; i++)
    {
      printf("Adding Row%i\n", i);

      weatherItems[i].dt = lv_label_create(weatherPage);
      lv_obj_align_to(weatherItems[i].dt, weatherItems[i-1].dt, LV_ALIGN_TOP_RIGHT, 42, 0);
      lv_label_set_text(weatherItems[i].dt, "");
      lv_label_set_long_mode(weatherItems[i].dt, LV_LABEL_LONG_WRAP);
      lv_obj_set_width(weatherItems[i].dt, 238);
      lv_obj_set_align(weatherItems[i].dt, LV_ALIGN_CENTER);

      printf("Add Icon\n");
      weatherItems[i].icon = lv_img_create(weatherPage);
      lv_obj_align_to(weatherItems[i].icon, weatherItems[i].dt, LV_ALIGN_OUT_BOTTOM_MID, 5, 5);
      lv_img_set_src(weatherItems[i].icon, &owm_01d_64px);

      printf("Add Feels\n");
      weatherItems[i].feels = lv_label_create(weatherPage);
      printf("Add Feels AlignTo\n");
      lv_obj_align_to(weatherItems[i].feels, weatherItems[i].dt, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 60);
      printf("Add Feels mode\n");
      lv_label_set_long_mode(weatherItems[i].feels, LV_LABEL_LONG_WRAP);
      printf("Add Feels Width\n");
      lv_obj_set_width(weatherItems[i].feels, 225);
      printf("Add Feels Aign\n");
      lv_obj_set_align(weatherItems[i].feels, LV_ALIGN_CENTER);
      printf("Add Feels text\n");
      lv_label_set_text(weatherItems[i].feels, "");

      printf("Add Min\n");
      weatherItems[i].min = lv_label_create(weatherPage);
      lv_obj_align_to(weatherItems[i].min, weatherItems[i].feels, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 5);
      lv_label_set_long_mode(weatherItems[i].min, LV_LABEL_LONG_WRAP);
      lv_obj_set_width(weatherItems[i].min, 225);
      lv_obj_set_align(weatherItems[i].min, LV_ALIGN_CENTER);
      lv_label_set_text(weatherItems[i].min, "");

      printf("Add Max\n");
      weatherItems[i].max = lv_label_create(weatherPage);
      lv_obj_align_to(weatherItems[i].max, weatherItems[i].min, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 5);
      lv_label_set_long_mode(weatherItems[i].max, LV_LABEL_LONG_WRAP);
      lv_obj_set_width(weatherItems[i].max, 225);
      lv_obj_set_align(weatherItems[i].max, LV_ALIGN_CENTER);
      lv_label_set_text(weatherItems[i].max, "");

      printf("Add Rain\n");
      weatherItems[i].rain = lv_label_create(weatherPage);
      lv_obj_align_to(weatherItems[i].rain, weatherItems[i].max, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 5);
      lv_label_set_long_mode(weatherItems[i].rain, LV_LABEL_LONG_WRAP);
      lv_obj_set_width(weatherItems[i].rain, 225);
      lv_obj_set_align(weatherItems[i].rain, LV_ALIGN_CENTER);
      lv_label_set_text(weatherItems[i].rain, "");

      printf("Add Main\n");
      weatherItems[i].bmain = lv_label_create(weatherPage);
      lv_obj_align_to(weatherItems[i].bmain, weatherItems[i].rain, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 5);
      lv_label_set_long_mode(weatherItems[i].bmain, LV_LABEL_LONG_WRAP);
      lv_obj_set_width(weatherItems[i].bmain, 225);
      lv_obj_set_align(weatherItems[i].bmain, LV_ALIGN_CENTER);
      lv_label_set_text(weatherItems[i].bmain, "");

      printf("Add Desc\n");
      weatherItems[i].desc = lv_label_create(weatherPage);
      lv_obj_align_to(weatherItems[i].desc, weatherItems[i].bmain, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 5);
      lv_label_set_long_mode(weatherItems[i].desc, LV_LABEL_LONG_WRAP);
      lv_obj_set_width(weatherItems[i].desc, 225);
      lv_obj_set_align(weatherItems[i].desc, LV_ALIGN_CENTER);
      lv_label_set_text(weatherItems[i].desc, "");

    }

```

Terminal output:
xxxxx
Adding Row17
Add Icon
Add Feels
Add Feels AlignTo
Add Feels mode
Add Feels Width
Add Feels Aign
Add Feels text
Add Min
Add Max
Add Rain
Add Main
Add Desc
Adding Row18
Add Icon
Add Feels
Add Feels AlignTo

### Expected behavior
Code to run all of the way through and not just hang

### Screenshots or video

Any suggestions would be appreciated.
Thanks
Alex

elgerg, 2021-12-15:
I think I am hitting some memory limit (although I have loads of RAM free, I'm running on a PI 400)..

Any help is appreciated,
Thanks
Alex

embeddedt, 2021-12-16:
Can you send your `lv_conf.h` file?

You can also try enabling logging and asserts; then you should hopefully see exactly where the issue is.

elgerg, 2021-12-16:
LV_conf attached. It looks like logging is enabled as is asserts..
[lv_conf.h.gz](https://github.com/lvgl/lvgl/files/7730619/lv_conf.h.gz)

Thanks
Alex

elgerg, 2021-12-16:
after further playing i believe it's to do with the screen size.

After changing:
#  define MONITOR_HOR_RES     800
#  define MONITOR_VER_RES     600

to:
#  define MONITOR_HOR_RES     1024
#  define MONITOR_VER_RES     600

it seems t get past the issue point.

I'm now getting some really strange issues with json-c that I dint get running under eclipse. Further investigation is required..

elgerg, 2021-12-17:
After trying the same on a new build of buster it works no problem. I suspect something is hinkey with bullseye...

closed 2021-12-17 by elgerg