lvgl/lvgl #2906: ESP32-S2. CPU halted from a deleting the lv_dropdown widget.

SinglWolf, 2021-12-16:
There is a code for creating and deleting a list of radio stations, which works great when using a touchscreen.
```c
lv_obj_t *ListStation;
static bool liststation = false;
static void event_list(lv_event_t *e)
{
	lv_event_code_t code = lv_event_get_code(e);
	if ((code == LV_EVENT_VALUE_CHANGED) || (code == LV_EVENT_KEY && (lv_event_get_key(e) == LV_KEY_ENTER)))
	{
		char buf[64];
		lv_obj_t *obj = lv_event_get_target(e);
		uint16_t ID = lv_dropdown_get_selected(obj);
		lv_dropdown_get_selected_str(obj, buf, sizeof(buf));
		setStationNum(ID + 1);
		startStop();
	}
}

bool status_list_station(void)
{
	return liststation;
}

void del_list_station(void)
{
	if (!liststation)
	{
		return;
	}

	liststation = false;
	vTaskDelay(10);
	lv_group_remove_obj(ListStation);
	vTaskDelay(10);
	lv_dropdown_clear_options(ListStation);
	lv_obj_del(ListStation);
	vTaskDelay(10);
}

// V8
void list_station(void)
{
	if (GetTotalStations() == 0)
		return;

	if (liststation)
	{
		del_list_station();
		liststation = false;
		return;
	}
	/*Create a normal drop down list*/
	ListStation = lv_dropdown_create(main_screen);
#if defined CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9488
	lv_obj_set_style_text_font(lv_dropdown_get_list(ListStation), &lv_font_radiola_18, LV_PART_MAIN);
#elif CONFIG_LV_TFT_DISPLAY_CONTROLLER_ILI9341
	lv_obj_set_style_text_font(lv_dropdown_get_list(ListStation), &lv_font_radiola_16, LV_PART_MAIN);
#endif
	lv_dropdown_clear_options(ListStation);
	lv_obj_set_width(ListStation, 316);

	//	lv_dropdown_set_show_selected(ListStation, false);
	const char *list_st[] = {" СПИСОК СТАНЦИЙ ", " LIST OF STATIONS "};
	lv_dropdown_set_text(ListStation, list_st[MainConfig->translate]);
	station_slot_s *si = NULL;
	for (uint8_t i = 0; i < GetTotalStations(); i++)
	{
		si = getStation(i);
		lv_dropdown_add_option(ListStation, si->name, i);
	}
	lv_dropdown_set_selected(ListStation, GetCurrentStation());
	lv_obj_align(ListStation, LV_ALIGN_TOP_MID, 0, 0);
	lv_obj_add_event_cb(ListStation, event_list, LV_EVENT_ALL, NULL);
	lv_dropdown_open(ListStation);
	// lv_group_add_obj(main_g, ListStation);
	lv_group_focus_obj(ListStation);
	liststation = true;
	if (si != NULL)
	{
		free(si);
	}
}
//
static void event_btn_play_list(lv_event_t *e)
{
	lv_event_code_t code = lv_event_get_code(e);

	if (code == LV_EVENT_CLICKED)
	{
		list_station();
	}
}
//
	// button "PLAYLIST"
	lv_obj_t *btn_play_list = lv_btn_create(container_buttons);
	// lv_group_add_obj(main_g, btn_prev);
	clear_scrollable(btn_play_list);
	lv_obj_set_style_bg_color(btn_play_list, lv_palette_main(LV_PALETTE_DEEP_ORANGE), LV_STATE_FOCUSED);
	lv_obj_set_size(btn_play_list, 43, 26);
	lv_obj_add_event_cb(btn_play_list, event_btn_play_list, LV_EVENT_ALL, NULL);
	lv_obj_align(btn_play_list, LV_ALIGN_BOTTOM_LEFT, 5, -5);
	lv_obj_t *label_btn_play_list = lv_label_create(btn_play_list);
	lv_label_set_text(label_btn_play_list, LV_SYMBOL_LIST);
	lv_obj_center(label_btn_play_list);
	lv_group_remove_obj(btn_play_list);
//
/* Register an input device */
void check_click(lv_indev_drv_t *indev_drv, uint8_t code)
{
	// uint32_t event_key = 0;
	// lv_event_code_t code = lv_event_get_code(event);
	// ESP_LOGE(TAG, "Event main page:\n");
	uint32_t key = ir_indev->proc.types.keypad.last_key;
	switch (code)
	{
	case LV_EVENT_PRESSED:
		// ESP_LOGW("CHECK", "Pressed\n");
		break;

	case LV_EVENT_SHORT_CLICKED:
		// ESP_LOGW("CHECK", "Short clicked\n");
		break;

	case LV_EVENT_CLICKED:
		// ESP_LOGW("CHECK", "Clicked\n");
		break;

	case LV_EVENT_LONG_PRESSED:
		// ESP_LOGW("CHECK", "Long press\n");
		break;

	case LV_EVENT_LONG_PRESSED_REPEAT:
		// ESP_LOGW("CHECK", "Long press repeat\n");
		break;

	case LV_EVENT_RELEASED:
		// ESP_LOGW("CHECK", "Released\n");
		break;
	case LV_EVENT_KEY:
		// ESP_LOGW("CHECK", "Key pressed");
		switch (key)
		{
		case LV_KEY_UP:
			// ESP_LOGI("CHECK", "KEY_UP");
			break;
		case LV_KEY_DOWN:
			// ESP_LOGI("CHECK", "KEY_DOWN");
			break;
		case LV_KEY_RIGHT:
			// ESP_LOGI("CHECK", "KEY_RIGHT");
			if (!liststation)
				evtSetVolume(+10);
			break;
		case LV_KEY_LEFT:
			// ESP_LOGI("CHECK", "KEY_LEFT");
			if (!liststation)
				evtSetVolume(-10);
			break;
		case LV_KEY_ESC:
			// ESP_LOGI("CHECK", "KEY_ESC");
			break;
		case LV_KEY_DEL:
			// ESP_LOGI("CHECK", "KEY_DEL");
			break;
		case LV_KEY_BACKSPACE:
			// ESP_LOGI("CHECK", "KEY_BACKSPACE");
			break;
		case LV_KEY_ENTER:
			// ESP_LOGI("CHECK", "KEY_ENTER");
			// evtKeyOK();
			break;
		case LV_KEY_NEXT:
			// ESP_LOGI("CHECK", "KEY_NEXT");
			break;
		case LV_KEY_PREV:
			// ESP_LOGI("CHECK", "KEY_PREV");
			break;
		case '*':
			// ESP_LOGI("CHECK", "KEY *");
			list_station();
			break;
		case '#':
			toggletime();
			break;
		case '0':
			nbStation('0');
			break;
		case '1':
			nbStation('1');
			break;
		case '2':
			nbStation('2');
			break;
		case '3':
			nbStation('3');
			break;
		case '4':
			nbStation('4');
			break;
		case '5':
			nbStation('5');
			break;
		case '6':
			nbStation('6');
			break;
		case '7':
			nbStation('7');
			break;
		case '8':
			nbStation('8');
			break;
		case '9':
			nbStation('9');
			break;
		default:
			break;
		}
		break;
	default:
		break;
	}
}
//
void init_indev_ir(void)
{
	static lv_indev_drv_t ir_drv;
	lv_indev_drv_init(&ir_drv);
	ir_drv.type = LV_INDEV_TYPE_KEYPAD;
	ir_drv.read_cb = ir_read;
	ir_drv.feedback_cb = check_click;
	ir_indev = lv_indev_drv_register(&ir_drv);
	lv_indev_set_group(ir_indev, main_g);
	if (!get_touch_on())
		lv_group_set_default(main_g);
}
```
The first click on the button creates the list. A second click deletes the list.
But when using the IR remote control, the iddilia is broken.
When press the button for the first time, which calls the same function of creating / deleting a list of radio stations on the IR remote control, the list is created. But when press the button again, the ESP32-S2 halted.
```
Guru Meditation Error: Core  0 panic'ed (Cache error). 
MMU entry fault error occurred while accessing the address 0x403d386c (invalid mmu entry)

Core  0 register dump:
PC      : 0x40024340  PS      : 0x00060b34  A0      : 0x8009ce02  A1      : 0x3fffd7d0
A2      : 0x3ffd386c  A3      : 0x3fffd810  A4      : 0x00060720  A5      : 0x000000a5  
A6      : 0x3ffcc094  A7      : 0x00000001  A8      : 0x8017b314  A9      : 0x3fffd7a0
A10     : 0x3ffd3d8c  A11     : 0x3fffd810  A12     : 0x00060520  A13     : 0x3ffcbfe0  
A14     : 0x3ffcc094  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x00000007
EXCVADDR: 0x3ffd386c  LBEG    : 0x00060520  LEND    : 0x3ffcbfe0  LCOUNT  : 0x400271f8  

Backtrace:0x4002433d:0x3fffd7d0 0x4009cdff:0x3fffd7f0 0x4009ceba:0x3fffd810 0x4009d32e:0x3fffd850 0x4009d854:0x3fffd880 0x4009e3ba:0x3fffd8a0 0x400b0aa5:0x3fffd8e0 0x400b0b56:0x3fffd900 0x4008a326:0x3fffd920 0x40031e95:0x3fffd950
  #0  0x4002433d:0x3fffd7d0 in _KernelExceptionVector at ??:?
  #1  0x4009cdff:0x3fffd7f0 in event_send_core at components\lvgl\src\core/lv_event.c:413
  #2  0x4009ceba:0x3fffd810 in lv_event_send at components\lvgl\src\core/lv_event.c:73
  #3  0x4009d32e:0x3fffd850 in lv_group_send_data at components\lvgl\src\core/lv_group.c:273
  #4  0x4009d854:0x3fffd880 in indev_keypad_proc at components\lvgl\src\core/lv_indev.c:444
  #5  0x4009e3ba:0x3fffd8a0 in lv_indev_read_timer_cb at components\lvgl\src\core/lv_indev.c:103
  #6  0x400b0aa5:0x3fffd8e0 in lv_timer_exec at components\lvgl\src\misc/lv_timer.c:313 (discriminator 2)
  #7  0x400b0b56:0x3fffd900 in lv_timer_handler at components\lvgl\src\misc/lv_timer.c:109
  #8  0x4008a326:0x3fffd920 in lv_task_handler at components\lvgl/src/lv_api_map.h:37
      (inlined by) task_lcd at main/services.c:725
  #9  0x40031e95:0x3fffd950 in vPortTaskWrapper at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:168

ELF file SHA256: 8ee1068bc053b4b3

CPU halted.
```
Сode snippet `main/services.c: 725`
```c
		if (xSemaphoreTake(xGuiSemaphore, portMAX_DELAY))
		{

			lv_task_handler(); // <-- 725 line
			xSemaphoreGive(xGuiSemaphore);
		}
```
If call up the list of radio stations by clicking on the touchscreen, when press the button on the IR remote control the list of radio stations is deleted normally, the ESP32-S2 not halted.
Please help me figure it out. Maybe this is a bug, but most likely I'm missing something (as always).
The sequence of the code is not followed and is provided for information.
Sorry for the long code snippets.

kisvegabor, 2021-12-16:
Hi,

Exactly which LVGL version do you use? (commit hash) It's important to see what is at `lvgl\src\core/lv_event.c:413`.

SinglWolf, 2021-12-16:
> Exactly which LVGL version do you use? (commit hash) It's important to see what is at `lvgl\src\core/lv_event.c:413`.

Unfortunately, I cannot see what hash of the used version. I updated my local lvgl repo an hour ago, but haven't migrated to my project yet.

```c
static lv_res_t event_send_core(lv_event_t * e)
{
    EVENT_TRACE("Sending event %d to %p with %p param", e->code, (void *)e->current_target, e->param);

    /*Call the input device's feedback callback if set*/
    lv_indev_t * indev_act = lv_indev_get_act();
    if(indev_act) {
        if(indev_act->driver->feedback_cb) indev_act->driver->feedback_cb(indev_act->driver, e->code);
    }

    lv_res_t res = LV_RES_OK;
    res = lv_obj_event_base(NULL, e); // <-- line 413

    lv_event_dsc_t * event_dsc = res == LV_RES_INV ? NULL : lv_obj_get_event_dsc(e->current_target, 0);

    uint32_t i = 0;
    while(event_dsc && res == LV_RES_OK) {
        if(event_dsc->cb && (event_dsc->filter == LV_EVENT_ALL || event_dsc->filter == e->code)) {
            e->user_data = event_dsc->user_data;
            event_dsc->cb(e);

            /*Stop if the object is deleted*/
            if(e->deleted) return LV_RES_INV;
        }

        i++;
        event_dsc = lv_obj_get_event_dsc(e->current_target, i);
    }

    if(res == LV_RES_OK && e->current_target->parent && event_is_bubbled(e)) {
        e->current_target = e->current_target->parent;
        res = event_send_core(e);
        if(res != LV_RES_OK) return LV_RES_INV;

    }

    return res;
}
```

kisvegabor, 2021-12-20:
Maybe it's a stack overflow. Can you increase the task size in which `lv_taick_handler` is running?

SinglWolf, 2021-12-20:
> Maybe it's a stack overflow. Can you increase the task size in which `lv_taick_handler` is running?

The size of the task in which `lv_task_handler` is running is `4*1024` bytes.
Increased to `6*1024` bytes. The result is the same. But now there is no detailed information about the failure.

```
Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3f8f5270  PS      : 0x00060030  A0      : 0x8009d266  A1      : 0x3fff25b0
A2      : 0x3f8f5270  A3      : 0x3fff25f0  A4      : 0x00060820  A5      : 0x00000000  
A6      : 0x3ffcc094  A7      : 0x00000001  A8      : 0x8015ac40  A9      : 0x3fff2580
A10     : 0x3f870014  A11     : 0x3fff25f0  A12     : 0x00060820  A13     : 0x3ffcbfe0  
A14     : 0x3ffcc094  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x00000002
EXCVADDR: 0x3f8f5270  LBEG    : 0x00060820  LEND    : 0x3ffcbfe0  LCOUNT  : 0x400277f5  

Backtrace:0x3f8f526d:0x3fff25b0 |<-CORRUPTED
  #0  0x3f8f526d:0x3fff25b0 in ?? ??:0

ELF file SHA256: 761eadb5eabd6a75

CPU halted.
```

kisvegabor, 2021-12-21:
I have no idea. :slightly_smiling_face: 
It's very unlikely that it's an lvgl issue as it this case it caused errors on other platforms too.

SinglWolf, 2021-12-23:
> I have no idea. 🙁

I also have no idea.
For now, instead of deleting the widget, I apply hiding.
There another ​major issue with this widget.
If there are more than ten items in the list, widget navigation becomes very slow! This is especially evident on the display with the `ILI9488` controller.
Is it possible to speed up the navigation somehow? Is it possible to paginate the contents of a list?

SinglWolf, 2021-12-27:
Here is the code snippet that crashes if `open_equalizer()` is called repeatedly by the remote control button. Calling `open_equalizer()` to delete the `container_equalizer` object via the touchscreen does not fail. If the object is not in focus (if you tap on the button with focus), when you call `open_equalizer()` to delete the `container_equalizer` object with the button on the remote control, no crash occurs. Forced defocusing `lv_group_focus_obj(btn_play_stop)` does not help.
What should I do?
```c
static lv_obj_t *container_equalizer = NULL;					// контейнер для эквалайзера
void open_equalizer(void)
{
	if (equalizer)
	{
		if (container_equalizer)
		{
			lv_group_focus_obj(btn_play_stop);
			vTaskDelay(10);
			lv_group_remove_obj(container_equalizer);
			lv_obj_del(container_equalizer);
			container_equalizer = NULL;
		}
		change_volume = true;
		equalizer = false;
		return;
	}
	change_volume = false;
	/*Create a slider in the center of the display*/
	container_equalizer = lv_obj_create(main_screen);
	clear_scrollable(container_equalizer);
	lv_obj_set_size(container_equalizer, LV_HOR_RES - 50, LV_VER_RES - 50);
	lv_obj_align(container_equalizer, LV_ALIGN_CENTER, 0, 0);
	lv_obj_t *slider = lv_slider_create(container_equalizer);
	lv_obj_center(slider);
	lv_obj_add_event_cb(slider, slider_event_cb, LV_EVENT_VALUE_CHANGED, NULL);

	/*Create a label below the slider*/
	slider_label = lv_label_create(container_equalizer);
	lv_label_set_text(slider_label, "0%");

	lv_obj_align_to(slider_label, slider, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
	lv_group_focus_obj(container_equalizer);
	equalizer = true;
}

static void slider_event_cb(lv_event_t *e)
{
	lv_obj_t *slider = lv_event_get_target(e);
	char buf[8];
	lv_snprintf(buf, sizeof(buf), "%d%%", (int)lv_slider_get_value(slider));
	lv_label_set_text(slider_label, buf);
	lv_obj_align_to(slider_label, slider, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
}
```

kisvegabor, 2022-01-06:
There was a bugfix in LVGL in past few weeks that might be related.  Please, try with that latest LVGL from the `master` branch.

C47D, 2022-01-06:
Sounds a bit like #2972

SinglWolf, 2022-01-07:
> There was a bugfix in LVGL in past few weeks that might be related. Please, try with that latest LVGL from the `master` branch.

Updated `LVGL` to current version in a `master`. Alas...

```
0x40027791: _xt_user_exc at C:\Users\Alexander\.platformio\packages\framework-espidf\components\freertos\port\xtensa/xtensa_vectors.S:627
```
Could there be a problem with the `check_click` function (I showed it in the first post)? This function is registered as `feedback_cb`.
```c
void init_indev_ir(void)
{
	static lv_indev_drv_t ir_drv;
	lv_indev_drv_init(&ir_drv);
	ir_drv.type = LV_INDEV_TYPE_KEYPAD;
	ir_drv.read_cb = ir_read;
	ir_drv.feedback_cb = check_click;
	ir_indev = lv_indev_drv_register(&ir_drv);
	lv_indev_set_group(ir_indev, main_g);
	if (!get_touch_on())
		lv_group_set_default(main_g);
}
```
I need this `check_click` function to regardless of all objects control some functions, for example, volume control. I also want to use this function in the future to get the raw coordinates of the touchscreen when calibrating it.
Maybe need to generate some kind of event after list_station();?
```c
		case 'L':
			// ESP_LOGI("CHECK", "KEY *");
			list_station();
			// <<< EVENT? >>>
			break;
```

> Sounds a bit like #2972, do you have the watchdog enabled? What ESP-IDF version are you using?

Naturally, the `watchdog` is on. The trigger period is maximum. `ESP-IDF` version `4.3.1`.
In my opinion, nothing like that. But since we're talking about the watchdog timer, I'll tell you about the problem in my project associated with it.
There is a registered timer, `time_update_timer`, which, in the absence of any output to stdout for several tens of minutes (did not measure exactly), triggers the watchdog timer. I found a solution like this.
```c
lv_timer_create(time_update_timer, 1000, NULL);
```
```c
void time_update_timer(lv_timer_t *timer)
{
	timein++;

	if (timein >= TO_TIME_SCREEN) // TO_TIME_SCREEN = 30
	{
		timein = 0;
		ESP_LOGD("TO_TIME_SCREEN", "status_play: %s", status_play ? "PLAY" : "STOPPED"); // <-- Without this line, the watchdog timer is triggered.
		if (!status_play)
		{
			statusPLAY(WAIT);
			if (WiFiConfig->current_ap != APMODE)
				itAskStime = true;
		}
		else
		{
			itAskStime = false;
		}

	} // start the time display when paused

	timerScreen++;
}
```
I really do not like this solution. Maybe you know a more elegant solution to this problem? Thank you.

> There another ​major issue with this widget.
> If there are more than ten items in the list, widget navigation becomes very slow! This is especially evident on the display with the `ILI9488` [...]

Thanks for taking part in my problem. I apologize for having collected several unrelated problems in one topic. There is also this problem. Is there a way to speed up this widget?

kisvegabor, 2022-01-07:
I have an idea what could be the problem. It seems you are deleting and creating objects in the `feedback_cb`, in `list_station()`. 
It can be problem, as LVGL doesn't check (I don't know how could it know) if the currently active object is deleted in the `feedback_cb`. So LVGL continues to use a deleted object. 

You can avoid it by using `lv_obj_del_async(ListStation)` in `del_list_station`. It makes LVGL make the deletion later, in a dedicated  `lv_timer`.

SinglWolf, 2022-01-08:
> I have an idea what could be the problem. It seems you are deleting and creating objects in the `feedback_cb`, in `list_station()`. It can be problem, as LVGL doesn't check (I don't know how could it know) if the currently active object is deleted in the `feedback_cb`. So LVGL continues to use a deleted object.
> 
> You can avoid it by using `lv_obj_del_async(ListStation)` in `del_list_station`. It makes LVGL make the deletion later, in a dedicated `lv_timer`.
> 
> Does it make any difference?

You got ahead of me. I have already applied `lv_obj_del_async(ListStation);` and was so happy that I forgot to write about it here. Today I check my mail and see exactly the same solution from You.
In order to avoid a similar situation in the future for other users, this needs to be somehow documented, probably?
It remains to decide how to speed up navigation through the widget.

kisvegabor, 2022-01-08:
Glad to hear it works now! :slightly_smiling_face: 

Meanwhile I got an idea how to fix this. Could try this modification in `lv_event.c` and use normal `lv_obj_del()`?
```c
static lv_res_t event_send_core(lv_event_t * e)
{
    EVENT_TRACE("Sending event %d to %p with %p param", e->code, (void *)e->current_target, e->param);

    /*Call the input device's feedback callback if set*/
    lv_indev_t * indev_act = lv_indev_get_act();
    if(indev_act) {
        if(indev_act->driver->feedback_cb) indev_act->driver->feedback_cb(indev_act->driver, e->code);

        //ADD THIS
        if(e->deleted) return LV_RES_INV;   
    }

   ...
```

> It remains to decide how to speed up navigation through the widget.

Please open a new issue for it and:
- describe the interface type (SPI I suppose) and clock speed
- add a video
- add a simple code snippet to see how the UI is built

SinglWolf, 2022-01-08:
> Meanwhile I got an idea how to fix this. Could try this modification in `lv_event.c` and use normal `lv_obj_del()`?

Your modification works great!

> Please open a new issue for it and:
> 
> * describe the interface type (SPI I suppose) and clock speed
> * add a video
> * add a simple code snippet to see how the UI is built

I think need to wait for the 18-bit color support to come out. Hope it's not long to wait? Anyway, I am interested in making the widget navigation faster.

kisvegabor, 2022-01-08:
> Your modification works great!

Cool, I've just pushed it.

> I think need to wait for the 18-bit color support to come out. Hope it's not long to wait? 

It's on the roadmap, but can promise and exact date now. :slightly_frowning_face:

closed 2022-01-08 by kisvegabor

C47D, 2022-01-08:
@kisvegabor I think you mean, you can't promise an exact date

kisvegabor, 2022-01-09:
@C47D Yes, thanks. Updated. :slightly_smiling_face:

commit bfc8edf8 (2022-01-08): fix(event): handle object deletion in indev->fedback_cb

fixes #2906