lvgl/lvgl #2908: image cache optimization

FASTSHIFT, 2021-12-16:
### Introduce the problem
The current image cache improves the access speed by caching the entire image in the memory, but if you add a zoom or rotation effect to `lv_img`, lvgl needs to recalculate the transformed effect every time it renders. When the number of pictures is large, it will significantly reduce the performance of UI sliding.

### Examples and cases

### Suggested solution

Is it possible to add a cache to the zoomed or rotated image to improve the sliding performance. This function is configurable, and the behavior of the cache allows users to configure it according to hardware performance.
I thought about three situations:
* 1. Cache the original resolution pictures, recalculate each rendering (the method currently used by LVGL).
* 2. Only cache the zoomed picture, and read the original file again when the zoom and rotation are reset. This is very useful in the case of images that are limited by hardware memory and do not often zoom and rotate.
* 3. At the same time cache the original picture and the scaled picture, this kind of performance is the best, but the memory overhead will increase, how to efficiently manage the memory of the scaled picture in the cache is also a problem, because the size may change frequently.

FASTSHIFT, 2021-12-16:
Also, is it possible to limit the total memory occupied by the image cache to prevent the cache from reaching images that exceed the memory size. Similar to FreeType's `FTC_Manager`.

kisvegabor, 2021-12-16:
Interesting idea! 

I'd solve it in a more generic way. We could cache a whole widget too with or without it's children. It can be configured by the user which widget to cache or let LVGL decide using some heuristic. I see 2 advantages of it:
- elements which are slow to render can be buffered (transformation, large shadow, etc)
- all elements can be transformed as images (e.g. rotate labels, rectangles, etc).

What do you think?

> Also, is it possible to limit the total memory occupied by the image cache to prevent the cache from reaching images that exceed the memory size. Similar to FreeType's FTC_Manager.

It'd be also great. It'd require the image decoders to guess how much memory will they use before actually decoding the image. 

@X-Ryl669 has made a [great update](https://github.com/lvgl/lvgl/pull/2904) on the image caching and he might have some ideas for this too.

FASTSHIFT, 2021-12-17:
@kisvegabor 
It seems to be a perfect solution to the entire widget buffer, I will continue to pay attention to this.

kisvegabor, 2022-01-03:
As object buffering to image is planned for v8.2 and it will solve this issue too, I close this one.

closed 2022-01-03 by kisvegabor