lvgl/lvgl #2909: roller display abnormal

FarawayRay, 2021-12-17:
Set the cycle mode when using roller. If there are too many rollers, abnormal display will occur. It looks like a y coordinate problem

kisvegabor, 2021-12-17:
Please send a minimal code snippet to reproduce and a video about the issue.

FarawayRay, 2021-12-20:
`static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_VALUE_CHANGED) {
        char buf[32];
        lv_roller_get_selected_str(obj, buf, sizeof(buf));
        LV_LOG_USER("Selected month: %s\n", buf);
    }
}

**#define TIME_MINUTE                "00\n01\n02\n03\n04\n05\n06\n07\n08\n09\n"\
                                   "10\n11\n12\n13\n14\n15\n16\n17\n18\n19\n"\
                                   "20\n21\n22\n23\n24\n25\n26\n27\n28\n29\n"\
                                   "30\n31\n32\n33\n34\n35\n36\n37\n38\n39\n"\
                                   "40\n41\n42\n43\n44\n45\n46\n47\n48\n49\n"\
                                   "50\n51\n52\n53\n54\n55\n56\n57\n58\n59"**

/**
 * An infinite roller with the name of the months
 */
void lv_example_roller_1(void)
{
    lv_obj_t *roller1 = lv_roller_create(lv_scr_act());
    **lv_roller_set_options(roller1,
        TIME_MINUTE,
                        LV_ROLLER_MODE_INFINITE);**
    lv_obj_set_size(roller1, 280, 456);
    lv_roller_set_visible_row_count(roller1, 3);
    lv_obj_center(roller1);
    lv_obj_add_event_cb(roller1, event_handler, LV_EVENT_ALL, NULL);

    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_48, LV_PART_MAIN);
    lv_obj_set_style_text_color(roller1, lv_color_white(), LV_PART_MAIN);
    lv_obj_set_style_text_opa(roller1, LV_OPA_60, LV_PART_MAIN);
    lv_obj_set_style_text_font(roller1, &lv_font_montserrat_48, LV_PART_SELECTED);

}`

roller cycle pattern to **LV_ROLLER_MODE_INFINITE**
options contains **0~59**
font using **48** 
from lv_example_roller1 in demo

This is easy to reproduce

FarawayRay, 2021-12-20:
Thanks for answering this question

kisvegabor, 2021-12-20:
Probably you need `LV_USE_LARGE_COORD  1` (in `lv_conf.h`).

closed 2021-12-21 by FarawayRay

X-Ryl669, 2022-01-10:
What is the reason here? I don't get it.
60 items that are 48px high ~= 3000px means it should fit a int16 nicely.

kisvegabor, 2022-01-11:
The min/max values are `1<<13` (8k) because we store e.g. % values on the upper part.
The line height of `lv_font_montserrat_48` is 52px, plus some line space, let's say 8px, one line is 60px. 

Besides the infinite mode works with adding the options 3 more times below and above the normal options to make an infinite-like list. When the roller is released the the "pages" are moved to the center page in the background.

So 60px x 60 options x 7 pages = 25k

X-Ryl669, 2022-01-11:
> Besides the infinite mode works with adding the options 3 more times below and above the normal options to make an infinite-like list.

OMG !?! Why this ? Wouldn't adding the options near the current visible row enough ?
That is, if you have 3 visible rows in the roller, add the 3 last items before the first one, and the 3 first items after the last one ?

It's a pain to increase the size of the whole LVGL's lv_coord_t just because of this.

kisvegabor, 2022-01-11:
It really could be adaptive. E.g. add only +1 pages for 60 options but +3 for 5 options.