lvgl/lvgl #2915: Arduino Esp32 LVGL_Arduino.ino  lv_demo_widgets() err.

fsyzty, 2021-12-18:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

#### lv_demo_widgets();====>The screen flickers and no image is displayed.
log:
.......
C:\Users\User\Documents\Arduino\libraries\lvgl\src\core\lv_obj.c: In function 'lv_init':
C:\Users\User\Documents\Arduino\libraries\lvgl\src\core\lv_obj.c:146:18: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     char * txt = "Á";
                  ^~~~
.........
"C:\\Users\\User\\Documents\\Arduino\\libraries\\lvgl\\src\\extra\\widgets\\tabview\\lv_tabview.c" -o "C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_34223\\libraries\\lvgl\\extra\\widgets\\tabview\\lv_tabview.c.o"
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\widgets\tabview\lv_tabview.c: In function 'lv_tabview_add_tab':
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\widgets\tabview\lv_tabview.c:89:25: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         new_map[tab_id] = "";
                         ^
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\widgets\tabview\lv_tabview.c:98:24: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             new_map[1] = "";
                        ^
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\widgets\tabview\lv_tabview.c:101:37: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             new_map[tab_id * 2 - 3] = "\n";
                                     ^
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\widgets\tabview\lv_tabview.c:103:37: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
             new_map[tab_id * 2 - 1] = "";
                                     ^
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\widgets\tabview\lv_tabview.c: In function 'lv_tabview_constructor':
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\widgets\tabview\lv_tabview.c:206:21: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     tabview->map[0] = "";
                     ^
.......
C:\Users\User\Documents\Arduino\libraries\lvgl\src\widgets\lv_checkbox.c: In function 'lv_checkbox_constructor':
C:\Users\User\Documents\Arduino\libraries\lvgl\src\widgets\lv_checkbox.c:125:13: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     cb->txt = "Check box";
             ^
......
Using library lvgl at version 8.1.1-dev in folder: C:\Users\User\Documents\Arduino\libraries\lvgl 
Using library TFT_eSPI at version 2.3.70 in folder: C:\Users\User\Documents\Arduino\libraries\TFT_eSPI 
Using library SPI at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SPI 
Using library FS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\FS 
Using library SPIFFS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SPIFFS 
Using library lv_examples at version 8.1.1-dev in folder: C:\Users\User\Documents\Arduino\libraries\lv_examples 

#### lv_demo_benchmark();===>OK
#### lv_demo_music(); ====>OK
#### lv_conf.h Set #define LV_USE_LOG 1 

C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_239442\LVGL_Arduino.ino: In function 'void setup()':
LVGL_Arduino:83:31: error: invalid conversion from 'void (*)(lv_log_level_t, const char*, uint32_t, const char*, const char*)' {aka 'void (*)(signed char, const char*, unsigned int, const char*, const char*)'} to 'lv_log_print_g_cb_t' {aka 'void (*)(const char*)'} [-fpermissive]
     lv_log_register_print_cb( my_print ); /* register print function for debugging */
                               ^~~~~~~~
In file included from c:\users\user\documents\arduino\libraries\lvgl\lvgl.h:25,
                 from C:\Users\User\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,
                 from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_239442\LVGL_Arduino.ino:1:
c:\users\user\documents\arduino\libraries\lvgl\src/misc/lv_log.h:64:51: note:   initializing argument 1 of 'void lv_log_register_print_cb(lv_log_print_g_cb_t)'
 void lv_log_register_print_cb(lv_log_print_g_cb_t print_cb);
                               ~~~~~~~~~~~~~~~~~~~~^~~~~~~~
Using library lvgl at version 8.1.1-dev in folder: C:\Users\User\Documents\Arduino\libraries\lvgl 
Using library TFT_eSPI at version 2.3.70 in folder: C:\Users\User\Documents\Arduino\libraries\TFT_eSPI 
Using library SPI at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SPI 
Using library FS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\FS 
Using library SPIFFS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SPIFFS 
Using library lv_examples at version 8.1.1-dev in folder: C:\Users\User\Documents\Arduino\libraries\lv_examples 
exit status 1
invalid conversion from 'void (*)(lv_log_level_t, const char*, uint32_t, const char*, const char*)' {aka 'void (*)(signed char, const char*, unsigned int, const char*, const char*)'} to 'lv_log_print_g_cb_t' {aka 'void (*)(const char*)'} [-fpermissive]

### To Reproduce

fsyzty, 2021-12-18:
lv_conf.h Set #define LV_USE_FS_FATFS 'S'        /*Uses f_open, f_read, etc*/
err log:
......
In file included from C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:12:
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c: In function 'fs_seek':
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/fatfs/src/ff.h:320:25: warning: dereferencing 'void *' pointer
 #define f_tell(fp) ((fp)->fptr)
                         ^~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:187:29: note: in expansion of macro 'f_tell'
             f_lseek(file_p, f_tell(file_p) + pos);
                             ^~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/fatfs/src/ff.h:320:25: error: request for member 'fptr' in something not a structure or union
 #define f_tell(fp) ((fp)->fptr)
                         ^~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:187:29: note: in expansion of macro 'f_tell'
             f_lseek(file_p, f_tell(file_p) + pos);
                             ^~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/fatfs/src/ff.h:321:25: warning: dereferencing 'void *' pointer
 #define f_size(fp) ((fp)->obj.objsize)
                         ^~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:190:29: note: in expansion of macro 'f_size'
             f_lseek(file_p, f_size(file_p) + pos);
                             ^~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/fatfs/src/ff.h:321:25: error: request for member 'obj' in something not a structure or union
 #define f_size(fp) ((fp)->obj.objsize)
                         ^~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:190:29: note: in expansion of macro 'f_size'
             f_lseek(file_p, f_size(file_p) + pos);
                             ^~~~~~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c: In function 'fs_tell':
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/fatfs/src/ff.h:320:25: warning: dereferencing 'void *' pointer
 #define f_tell(fp) ((fp)->fptr)
                         ^~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:209:14: note: in expansion of macro 'f_tell'
     *pos_p = f_tell(file_p);
              ^~~~~~
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/fatfs/src/ff.h:320:25: error: request for member 'fptr' in something not a structure or union
 #define f_tell(fp) ((fp)->fptr)
                         ^~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:209:14: note: in expansion of macro 'f_tell'
     *pos_p = f_tell(file_p);
              ^~~~~~
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c: In function 'fs_dir_open':
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:222:5: error: unknown type name 'DIR'; did you mean 'DDR'?
     DIR * d = lv_mem_alloc(sizeof(DIR));
     ^~~
     DDR
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:222:35: error: 'DIR' undeclared (first use in this function); did you mean 'DDR'?
     DIR * d = lv_mem_alloc(sizeof(DIR));
                                   ^~~
                                   DDR
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:222:35: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:225:29: warning: passing argument 1 of 'f_opendir' from incompatible pointer type [-Wincompatible-pointer-types]
     FRESULT res = f_opendir(d, path);
                             ^
In file included from C:\Users\User\Documents\Arduino\libraries\lvgl\src\extra\libs\fsdrv\lv_fs_fatfs.c:12:
C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32/include/fatfs/src/ff.h:290:28: note: expected 'FF_DIR *' {aka 'struct <anonymous> *'} but argument is of type 'int *'
 FRESULT f_opendir (FF_DIR* dp, const TCHAR* path);      /* Open a directory */
                    ~~~~~~~~^~
Using library lvgl at version 8.1.1-dev in folder: C:\Users\User\Documents\Arduino\libraries\lvgl 
Using library TFT_eSPI at version 2.3.70 in folder: C:\Users\User\Documents\Arduino\libraries\TFT_eSPI 
Using library SPI at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SPI 
Using library FS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\FS 
Using library SPIFFS at version 2.0.0 in folder: C:\Users\User\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\libraries\SPIFFS 
Using library lv_examples at version 8.1.1-dev in folder: C:\Users\User\Documents\Arduino\libraries\lv_examples 
exit status 1
Error compiling for board ESP32 Dev Module.

kisvegabor, 2021-12-20:
> lv_demo_widgets();====>The screen flickers and no image is displayed.

Maybe an out of memory issue? Try with `LV_MEM_SIZE 48*1024`.

fsyzty, 2021-12-21:
Thank you very much .
lv_demo_widgets(); OK.

kisvegabor, 2021-12-21:
Great! The logging issue should be fixed like this: https://github.com/lvgl/lvgl/blob/master/examples/arduino/LVGL_Arduino/LVGL_Arduino.ino

fsyzty, 2021-12-22:
v8.1.0  log  Serial printf is ok.
Thank you.

kisvegabor, 2021-12-22:
Great! Is there any issue left?

fsyzty, 2021-12-25:
Does #define LV_USE_FS_STDIO'S' or #define LV_USE_FS_POSIX 'P' also copy lv_port_fs_template.c to lv_port_fs.c and complete the functions inside?

fsyzty, 2021-12-27:
1. lv_conf.h
#define LV_USE_FS_STDIO 'A'         /*Uses fopen, fread, etc*/
#define LV_FS_STDIO_PATH "/LFS/"    /*Set the working directory. If commented it will be "./" */
2.ino
......
if(!LittleFS.begin(FORMAT_LITTLEFS_IF_FAILED,"/LFS")){
        Serial.println("LittleFS Mount Failed");
        return;
    }
......
    lv_obj_t * img1;
    img1 = lv_img_create(lv_scr_act());
    /* Assuming a File system is attached to letter 'A'
     * E.g. set LV_USE_FS_STDIO 'A' in lv_conf.h */
    lv_img_set_src(img1, "A:wink.png");
    lv_obj_align(img1, LV_ALIGN_RIGHT_MID, -20, 0);
......
is OK.

closed 2021-12-27 by fsyzty