lvgl/lvgl #2923: LVGL touch event "LV_EVENT_LONG_PRESSED" can not be generated if I press the screen without lift up my finger

glin00, 2021-12-20:
Describe the bug
I am using lvgl as GUI library with zephyr. The display which is an OLED display can be displayed ok using lvgl. The OLED display touch function is working since I touch the display, the correct display x,y can be reported. If I press the screen and lift up my finger, "LV_EVENT_CLICKED" can be generated within lvgl. But if I press the screen without lift up my finger ,"LV_EVENT_LONG_PRESSED" can not be generated.

When I trace the code using west debug, I found that indev_proc_press can be entered but line 978 which will generate LV_EVENT_LONG_PRESSED can not be entered(see attached).

Can any one help me how to solve this problem?

To Reproduce
The following is the part of prj.conf related to lvgl.
#LVGL
CONFIG_MAIN_STACK_SIZE=16384
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_HEAP_MEM_POOL_SIZE=16384
CONFIG_LVGL_MEM_POOL_HEAP_KERNEL=y

CONFIG_LVGL=y
CONFIG_LVGL_HOR_RES_MAX=300
CONFIG_LVGL_VER_RES_MAX=340
CONFIG_LVGL_VDB_SIZE=100
CONFIG_LVGL_BITS_PER_PIXEL=16
CONFIG_LVGL_DPI=130

#LVGL color
CONFIG_LVGL_COLOR_DEPTH_16=y

#LVGL display
CONFIG_DISPLAY=y
CONFIG_LVGL_DISPLAY_DEV_NAME="SmallOLED"
CONFIG_LVGL_DISP_DEF_REFR_PERIOD=30

#LVGL INDEV
CONFIG_LVGL_POINTER_KSCAN=y
CONFIG_LVGL_POINTER_KSCAN_DEV_NAME="OLEDTouch"
CONFIG_LVGL_INDEV_DEF_READ_PERIOD=50
CONFIG_LVGL_INDEV_DEF_DRAG_LIMIT=10
CONFIG_LVGL_INDEV_DEF_DRAG_THROW=20
CONFIG_LVGL_INDEV_DEF_LONG_PRESS_TIME=400
CONFIG_LVGL_INDEV_DEF_LONG_PRESS_REP_TIME=100

#LVGL buffer
CONFIG_LVGL_BUFFER_ALLOC_STATIC=y

#LVGL USE options
CONFIG_LVGL_USE_BTN=y
CONFIG_LVGL_USE_CONT=y
CONFIG_LVGL_USE_LABEL=y

Impact
This project I am doing now requires lvgl to generate "LV_EVENT_LONG_PRESSED" when user touches the screen's button long enough without lift up finger.

Environment (please complete the following information):
Hardware:STM32L496AGI6P
Display: OLED display with 4 wire SPI interface.
Linux Ubuntu 20.04.3 LTS
Zephyr SDK,-0.13.0
Zephyr SHA1 ID:bb85a55
LVGL:7.6.1

glin00, 2021-12-20:
After I trace code, I found that ,within indev_proc_press function (line 975), proc->pr_timestamp (press screen time stamp) always equal to sys_time. This causes lv_tick_elaps return a value not greater than indev_act->driver.long_press_time.
As a result, LV_SIGNAL_LONG_PRESS can not be triggered.
Can any one help why proc->pr_timestamp is always equal to sys_time?

kisvegabor, 2021-12-20:
Are you using LVGL v7, rigtht?

It can can imagine 2 reasons
1. your driver continuously generates press/releases while you press it
2. `lv_tick_inc()` is not called. You can try it by printing `lv_tick_get()`. It should be increment as milliseconds  (not sure with 1 ms resolution).

knifter, 2022-01-04:
The issue is most likely in your indev callback function and related to how your touchscreen(driver) works, as kisvegabor suggested. Try posting your implementation of that function here.

glin00, 2022-01-07:
Sorry for the late reply.
@kisvegabor , for point 1, yes you are right, my touch driver continuously generates press/releases while I press the touch screen. So I am trying to fix it in this direction.
point2. lv_tick_inc() does get called periodically about 10ms interval.

glin00, 2022-01-07:
As this issue is very related to the problem of the touch screen driver. I will close this issue for now.

closed 2022-01-07 by glin00