lvgl/lvgl #2926: Incorrect code in img_get_srctype

X-Ryl669, 2021-12-20:
### Perform all steps below and tick them with [x]
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Currently, when setting a src to a `lv_img_t`, there are multiple places where the source type is queried to figure out if it's a file, a symbol or the data itself.

The issue comes from the fact that, in order to assert what kind of src it is, the "parser" estimate that ASCII char in range 32-127 is a file, 0 to 31 is the data and above 127 is a symbol IIUC.

I think using the same information for storing 3 different things is prone to error anyway.
Also, by not storing the actual source type explicitly leads to a performance degradation, since every place where the src_type is requested, the heuristics have to be run again (and this often implies hitting the file system or at least doing a string compare which is slow on a long string). 

### To Reproduce

### Expected behavior
Instead of having guess code going on here, we should have a structure describing the type of the data to use, with an enum for the supported types (the enum is the actual src_type) and the sub-type (what kind of decoder is required to decoder it (since we'll have them in the decoder's context soon).
This would avoid the `strlen(filename) + strcmp(filename, extension` which are both O(N) and O(N*M) each time the src is parsed (that is, for each frame of an animation).

I think the `lv_img_set_src` should specify the type of the source to use. In my partial_rlottie PR #2904, I've added a `lv_img_set_src_ex` function, and I think we could add the src type as an argument here so that:
1. No misdetection can happen (upon multiple file format supporting like json for rlottie / rive / whatever) 
2. No need for IMG_DECLARE macro anymore (this is clunky anyway, IMHO, since this object is only required for bin format, all other formats can find their image description by themselves). This would save precious flash space by not repeating the information. We can pass the image header as an optional argument too to the builtin decoder context. 
3. No query of the src type anymore in all functions later on the chain.

Typically, `lv_img_set_src` would do (for backward compatibility):
```cpp
void lv_img_set_src(obj, src) {
   src_type_t src_type = lv_get_src_type(src);
   dec_type_t dec_type = LV_BUILTIN_ID;
   if (src_type == SRC_FILE) dec_type = find_decoder_for_path(src);
   lv_img_set_src_ex(obj, src, src_type, dec_type, NULL);
}
```

Ideally, the linked list for the decoders could then be changed from a linked list to a map (decoder id=>decoder instance). That way, instead of doing a O(N) search in the decoder for a given type, use a O(1) map for a decoder pointer directly. Or, at least, have a decoder index in the list stored in the img_t so the O(N) does not imply calling N-1 decoder's `get_info` function that are very expensive.

What do you think?

X-Ryl669, 2021-12-20:
In the possible new processing flow, it would look like this:
1. if the decoder type is not set, then call `lv_img_decoder_get_info()` as usual
2. This will call each decoder in turns to know if one accept the input type
3. NEW: Once one decoder fits the format, it allocates its decoder context (optimization) 
4. Because it's returning success, img_dec_get_info store the dec_type in the lv_img_t object for future use
5. Later on, the image needs to be drawn
6. img_decoder_get_info is called again, and this time since dec_type is set, there is no need to check each decoder so directly seek to the right decoder
7. Since the context is already allocated, this is used to return the information directly, without parsing the img header again
8. Later on, decoder_open is called, since the context is already allocated, again, parsing time + allocation time is saved
9. and so on

If the decoder type is known (set in set_src_ex), then it's like step 6 above, with many shortcuts.

Please notice that if `lv_img_decoder_getinfo` is called without a following `open` (like in set_src), to avoid a memory leak, we need to close the decoder context, so an additional method need to be added (or we can reuse `decoder_close` function for this).

Right now, the current data flow is mainly:
(Dec.Get info => open the file => parse the file => parse the image format => extract w & h => close the file) x2
then (Dec.Open => open the file => parse the file =>  parse the image format => allocate buffer => close the file/or not) 
then (Dec.Close => free buffer => close the file) (if no image caching set up)

We see that there is numerous resources wasted here in doing the same thing again and again

kisvegabor, 2021-12-21:
> I think using the same information for storing 3 different things is prone to error anyway.

It could be but basically you can't pass any argument that results in unexpected behavior. The only issue I could recall was that the bytes in the `lv_img_dsc_t` needed to be swapped with `LV_BIG_ENDIAN_SYSTEM 1`.  

>  lv_img_set_src_ex(obj, src, src_type, dec_type, NULL);

I general I like the idea, and I agree its it's wasteful to find the right decoder every time. However, I think `  lv_img_set_src_ex` has  a quite complicated API especially for a novice user. Setting an image shouldn't be so complicated. So I agree to cache the decoder type in `lv_img_set_src` but IMO `lv_img_set_src_ex` has more cons than pros. It's unlikely that finding the decoder once will be a performance bottleneck.

X-Ryl669, 2021-12-21:
> It could be but basically you can't pass any argument that results in unexpected behavior. 

Well, that's because you're hiding away the src in a macro that's wrapping the data in some header structure. But that header structure is redundant, it's wasting flash space. Ideally, we should be able to say:
```
const uint8 * somePic[] = { 0x.... };
lv_img_set_src(img, somePic)
```
It should work for any picture format, since the decoder will find out the width & height & color format anyway.
But because of builtin bin format, we have to create a new object to store the *redundant* header and deal with it for all other format. The builtin format generation tool can as well store the header in the byte stream and we'd be done with it without the need for LV_IMG_DECLARE clunky macro.

IMHO, it would be better if we had a clear source type instead, like this:
```
lv_img_set_url(img, "/some/url/or/file/path");
lv_img_set_data(img, somePic);
```
a bit like CSS with its `url()` or `data-uri()` mode.

kisvegabor, 2021-12-21:
> lv_img_set_url(img, "/some/url/or/file/path");
lv_img_set_data(img, somePic);

I see, and it makes sense. In later release we should rework `lv_img_dsc_t` anyway because now it supports max 2048x2048 images which (in some scenarios) seemed to be small.

So let's postpone it a little, but in general I like to find a more generic solution.

X-Ryl669, 2021-12-21:
> It's unlikely that finding the decoder once will be a performance bottleneck.

You are right, of course. The main issue I've found while writing rlottiedec, is that the decoders are searched each time the obj is invalidated, so mainly for each frame of the animation. And since decoder_get_info does not have a context, it implies reparsing the animation's file again (with the high heap & cpu performance penalty) just to extract the width & height of the animation. Then hopefully, it's cached  in decoder_open, so at least we don't have to re-allocated the picture's buffer each frame. 

I'd like to remove this scheme that's not very smart, IMHO. Maybe we could have a "decoder_accept" function instead (that's used in img_set_src) to find if a decoder can accept the given URI/src, and if it does, simply store the decoder index in the list.
Then remove the decoder_info, since it's wasting resources. If we need to extract picture information, let's open the decoder and keep the decoder context allocated so we never have to reallocate it again until the picture is no more used.

kisvegabor, 2021-12-21:
Can't we simply store a pointer to the decoder in the `ctx`? It seems the most transparent than using IDs. It can replace `magic_id` too. If the user sets the decoder, LVGL will use only that decoder with that `ctx` so there is no change for failure. If the decoder is not set, `decoder_open` will set it to the found decoder. 

> Then remove the decoder_info, since it's wasting resources. If we need to extract picture information, let's open the decoder and keep the decoder context allocated so we never have to reallocate it again until the picture is no more used.

Let's consider an real life example. `lv_meter` can have needles that can be images. Now only the `src`s are stored, and when needed they are drawn with `lv_draw_img()`. Let's say the user uses `my_needle.png`. With your suggestion the following will happen:
1. Meter stores `lv_img_decoder_ctx_t`s instead of plain `src`s. 
2. When the needle is set, the meter opens the image to initialize the context. It also decode the needle into a large image buffer.
3. Meter saves that initialized context, and when the needles needs to be drawn, it uses the context to position the needles and to draw them
4.  Meanwhile, if too many images are opened,  the `my_needle` might be closed, and during next drawing it needs to be rendered again. 

The only issue I see is that if the cache is small (e.g. only 1 image can be kept open) it's wasteful to render the image when setting a new src, because it cannot be used later as the cache can hold only one from the many image.

So IMO there should be a method the get all info from the image but do not render anything.

---

One more thing that is not clear me. Let's say the image cache size is 3 but there are 10 image on a screen. So 10 `ctx`s are "floating" attached to various objects. `decoder_open` is used with a `ctx` and it considers a cache hit if the `ctx` stored in the decoder entry and the `ctx` passed as argument are the same. 
Let's say the cache manager closes an image whose `ctx` we store somewhere. If there is dynamically allocated data in `ctx` the decoder needs to free it as well and maybe the the whole `ctx` needs to be initialized to a default value. So the `ctx` in a widget can become uninitialized "randomly" and the developer needs to take care to reinitialize it before it's used. 
Really is this the case or am I missing something?

X-Ryl669, 2021-12-21:
> Can't we simply store a pointer to the decoder in the ctx?

Yes we can. Simple and perfect. In the final scheme discussed in the PR, the user don't need to deal with decoder context at all, since everything would be accessible via lv_img API. So that's great.

As for the rest of your example, you are misunderstanding the decoder context and decoder descriptor. The context is a small structure so it's quite ok to keep it around (at worst, it'd be 16 bytes: 4 bytes for the picture dimensions, 4 bytes for the user data/private pointer for extended context, 2 bytes for the destination frame index, 1 byte for the caps, 1 byte for the magic id/should_free) + (NEW) 4 bytes for the pointer to the decoder's vtable.

The context is the information used to initialize the decoder (not the same thing at all). It does not contain any picture and it not cleaned by the cache if the user provided it.
So to refer to your example:
1. The meter will store a pointer to src (option 1) or better, a lv_img_t (option 2)
2. When the needle's src is set, the meter call lv_img_set_src for the needle's object, and in set_src, the source type is deduced (either like now, or like I mentioned with a URI/data). The decoders are searched by calling decoder_accept(src). 
3. If no decoder can decode the picture, well, bad luck, as usual. It returns null. Else if one decoder accept the source, then it create its decoder context if required: that is, if the user did not provide it (option 1) or fill it (if the user gave one in option 2) and returns it.
4. lv_img_decoder_get_info then set the lv_img_t.dec_ctx to this context.
5. When drawing the needle is required, lv_img_t's draw function will use the pointer in the decoder's context to call the cache with (like in the current code). The cache will search for some available decoder's descriptor and if one is found for this decoder, simply call its decoder_open with that descriptor. Else, call decoder_open with a null descriptor and the decoder will actually allocate it, using the decoder context to initialize itself. The decoder descriptor is where the rendered buffer is stored, like the current code.
6. If too many descriptors are used, the cache will call decoder_close as usual, requesting the decoder to free its buffers (but the context is not freed, unless it was allocated by the accept function).

Checking who is freeing the decoder context looks like it's complex, and we can decide to only have the lv_img_t free it when it's destructed, so we'll remove the `should_free` member.

X-Ryl669, 2021-12-21:
> So IMO there should be a method the get all info from the image but do not render anything.

Yes. If the image format is simple (like PNG, BMP, GIF, Builtin) where fetching the information only requires parsing a few byte header, then the accept method can fill it in the decoder context/image header (like now). 

For more complex format (like rlottie, ffmpeg) where fetching the information requires parsing the whole file to get it, then it's absolutely required not to do that more than once. The strategy I've implemented in rlottiedec is to parse the whole file in get_info (and in the future _accept) but keep the allocated rlottie's context in the decoder context's user data. 
This does not contain the rendered picture and no rendering is done yet.

That way, when calling decoder_open, if the context contains the rlottie's context, it's reusing this, bypassing parsing the file again. If not, it's allocated as usual and stored in the user_data. Then the rendering buffer are allocated and stored in the descriptor.

When decoder_close is called, this context is freed (in addition to the descriptor). So for very high memory pressure, it'll still work as before. But if you have a cache, then we skip this useless creation/allocation most of the time.

kisvegabor, 2021-12-22:
Okay, got it. Mostly :slightly_smiling_face:  Only 2 things are not clear:
- If `lv_img_t` stores a `ctx` which has `user_data` how the image know how to free that user data as user data can be a complex structure with dynamically allocated fields. E.g.
```c
struct {
   const char * name; //malloc(name_len)
   lv_color_t * palette; //malloc(sizeof(lv_color_t) * palatte_size))
}my_user_data;
```

- Basically what is the difference between `_accept` and `_get_info`? Can't `get_info` set the accepted decoder too in the `ctx`?

X-Ryl669, 2021-12-23:
Currently, I have a field called `should_free` in the context and if it's true, then it's freed in decoder_close.
When the user set up a decoder context by herself (like in rlottie format), then `should_free` is 0. It's up to the rlottie_destructor to free it.
If the context was null and allocated by decoder_open, then `should_free` is 1 and it's freed in decoder_close.

I agree it's not that great. I couldn't think of a better scheme. If you have suggestions, feel free to expose them.

Ideally, `accept` should only say "ok this data/URI can be decoded by this decoder".
`_get_info` should say "here's the width & height & colorformat & caps of this picture". 
Since in all decoders, `get_info` does `decoder_open` minus the decoding, it's a waste of resource since most of the time, `get_info` will be followed by `decoder_open`. Resulting in opening the file/decoding the header twice.

I understand the need to extract the image's metadata (width/height/etc...) but I think we should rework it so that the decoder's descriptor is saved between those 2 calls since they will follow most of the time.
One way to do that is to remove `get_info` and have a flag in the `decoder_open` to tell to skip decoding. So the scheme will be:

1. set_src need to find which decoder is useful
2. It calls all decoders' _accept until one returns LV_RES_OK
3. If the size is required (not set by the user beforehand), it calls `decoder_open` with flag "SKIP_DECODING"
4. This allocate the decoder descriptor from the context and fills the required metadata
5. Later on, the image is drawn so the draw_img is called. It'll call `decoder_open` (via the cache) again but without the SKIP_DECODING flag. This times, the decoder will re-use its descriptor to allocate the picture's buffer and decoder, skipping the parsing phase since it was already done.

kisvegabor, 2022-01-04:
Sorry for the late reply, I was on Christmas holiday and just got back to work this week.

> Currently, I have a field called should_free in the context and if it's true, then it's freed in decoder_close.
When the user set up a decoder context by herself (like in rlottie format), then should_free is 0. It's up to the rlottie_destructor to free it.

I still not fully understand. If the `ctx` is stored in `lv_img` and the `ctx`'s `user_data` is allocated by `decoder_open` `lv_img_destructor` can't free it as it doesn't know what is behind the `void * user_data` field.

---

> One way to do that is to remove get_info and have a flag in the decoder_open to tell to skip decoding. So the scheme will be:

Sounds good. Having an option to pass flags for `decoder_open` could be for other purposes too.

X-Ryl669, 2022-01-04:
We have 2 actors that are possibly filling the context: The decoder (in get_info/open whatever) or the user code. So we must keep track of our own chicks here. If the decoder allocated and filled the context, they it must free it. If it's the user code, the decoder shouldn't free it, but the user's code should. 

In `lv_rlottie`, since the context is required to know the size beforehand and to keep the rlottie's instance opened (too long to parse the file for each animation frame), the context is managed by the *user* code, so it's freed in `lv_rlottie_destructor`. 

I'm pretty sure ffmpeg should work the same, since you don't want to parse a movie file for each frame.

kisvegabor, 2022-01-04:
Maybe I miss something but it still doesn't adds up. 

I fill some data in the `ctx`, such as the target width/height and let the decoder the manage the rest, including the `user_data`. I don't know which decoder will be able this image so I can't set anything specific.  So the `ctx` "object" is controlled by me (i.e. the image widgets) but the `user_data` is controlled by the decoder. Who will free the `user_data`?

X-Ryl669, 2022-01-04:
The decoder. The "user" in my answer is not you, it's the user code (so for example in lv_rlottie, it's lv_rlottie_constructor/destructor). In lv_rlottie, the code is dealing with its user_data by itself, so I needed a way to tell the decoder not to touch it, since I don't want it to free the rlottie's animator each time it's open & closed.

kisvegabor, 2022-01-06:
Okay, I think I finally got it. :slightly_smiling_face: 
If you initialize the `user_data`, because you know what you re doing (e.g. in case of rlottie) you need to free the user data as you wish.
If you leave the magic to the decoder, the decoder will close it too. 

The missing piece fro me was that you (the programmer) can actually initialize the use_data too.

---

So what's next? How these change affect the other open PRs?

X-Ryl669, 2022-01-07:
I've worked on dithering PR and likely finished it, please review. I'm going to resume the work on the partial rlottie support that'll integrate the changes discussed above. 

Just a remark: There is no reason why an image format couldn't have multiple (virtual) frames, since we can store each frame vertically. So adding a `.abin` support for animated button on stored binary image is very possible.

kisvegabor, 2022-01-07:
> I've worked on dithering PR and likely finished it, please review. I'm going to resume the work on the partial rlottie support that'll integrate the changes discussed above

It will be a large PR, but okay.

> Just a remark: There is no reason why an image format couldn't have multiple (virtual) frames, since we can store each frame vertically. So adding a .abin support for animated button on stored binary image is very possible.

True! Basically LVGL could have support for a simple animated image format that simply places multiple images with the same size after each other. It'd be easily seekable too. This way, it makes even more sense to make animbtn non-rlottie specific but simply make it to get an img src argument.

X-Ryl669, 2022-01-11:
Ok, not finished yet, but please have a look to #2904 for the proposed changes. 
It's a lot of small changes but impossible to split in multiple PR without breaking everything, since the logic of the image processing code is modified. 

Basically, the new code is a lot faster because it's skipping the decoder_open/close where it can. Background image tiling is also a lot faster and so for object manipulating images. 

Please let me know if you follow the new logic and if you have some criticism to the proposed implementation or idea of improvement. 

If ok, I'll port the required changes to the other image decoder too.

kisvegabor, 2022-01-11:
Thank you very much! :tada 
I'll start with this tomorrow.

kisvegabor, 2022-01-24:
I close it as the discussion continues in #2904.

closed 2022-01-24 by kisvegabor