lvgl/lvgl #2928: switch lv_obj_has_state and lv_obj_clear_state causes runtime error

elgerg, 2021-12-20:
### Perform all steps below and tick them with [x]
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I have a gobal variable for my switch and create it in the usual manor.

Since 8.1 uses lv_obj_clear_state not lv_switch_off I changed from:
lv_switch_off(hwsw, LV_ANIM_ON);
to
lv_obj_clear_state(hwsw, LV_STATE_CHECKED);

However I now get:
../lvgl/src/core/lv_obj.c:366:28: runtime error: member access within misaligned address 0x40a1eb8500000055 for type 'const struct lv_obj_t', which requires 8 byte alignment
0x40a1eb8500000055: note: pointer points here
<memory cannot be printed>
AddressSanitizer:DEADLYSIGNAL
=================================================================
==2278==ERROR: AddressSanitizer: SEGV on unknown address 0x143d80a000000a (pc 0x0055832354e4 bp 0x007ff46460f0 sp 0x007ff46460f0 T0)
==2278==The signal is caused by a READ memory access.
    #0 0x55832354e3 in lv_obj_has_class ../lvgl/src/core/lv_obj.c:366
    #1 0x5583234a0b in lv_obj_has_state ../lvgl/src/core/lv_obj.c:323
    #2 0x558329aeff in update_hw ../App/Tabs/Heating.c:441
    #3 0x55832a8cbf in handleHeating ../App/MQTT_Callbacks/Callbacks.c:418
    #4 0x55832ac513 in my_message_callback ../App/MQTT/MQTT.c:95
    #5 0x7fb627159f  (/lib/aarch64-linux-gnu/libmosquitto.so.1+0x559f)
    #6 0x7fb627537b  (/lib/aarch64-linux-gnu/libmosquitto.so.1+0x937b)
    #7 0x7fb627220f in mosquitto_loop_read (/lib/aarch64-linux-gnu/libmosquitto.so.1+0x620f)
    #8 0x7fb62727a3 in mosquitto_loop (/lib/aarch64-linux-gnu/libmosquitto.so.1+0x67a3)
    #9 0x558329628b in tsk_runmosq ../App/Tasks/Tasks.c:14
    #10 0x55830a29f3 in lv_timer_exec ../lvgl/src/misc/lv_timer.c:313
    #11 0x55830a16cb in lv_timer_handler ../lvgl/src/misc/lv_timer.c:109
    #12 0x55832b0a9f in main ../main.c:106
    #13 0x7fb5356d23 in __libc_start_main ../csu/libc-start.c:308
    #14 0x5582fc2c03  (/home/pi/eclipse-workspace/lv_sim_eclipse_sdl/Debug/lv_sim_eclipse_sdl+0x342c03)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV ../lvgl/src/core/lv_obj.c:366 in lv_obj_has_class
==2278==ABORTING

### To Reproduce
```c
		if(lv_obj_has_state(hwsw, LV_STATE_CHECKED))
		{
			lv_obj_clear_state(hwsw, LV_STATE_CHECKED);
		}
```

### Expected behavior
switch will town off or state will be set 

### Screenshots or video

elgerg, 2021-12-20:
Update. The first call or 2 works as expected, once it (the app) has been running a few minutes it fails. This is within a callback from an MQTT broker.

elgerg, 2021-12-20:
If I comment out the switch state change code the other updates to the UI work fine, label change etc..

kisvegabor, 2021-12-21:
Are the any preemptive tasks in your systems? If so take care to use a mutex as [described here](https://docs.lvgl.io/master/porting/os.html).

elgerg, 2021-12-21:
No, I dont think so. I'm calling the MQTT poll from within a LVGL timer so it should be ok?

IE:

void tsk_runmosq(lv_task_t * task)
{
    mosquitto_loop(mosq, 10, 1);
}

This didnt seem to be an issue when I was using 7.5:
lv_switch_off(hwsw, LV_ANIM_ON);

I am not sure why changing to 8.1 would cause this.

Thanks!
Alex

elgerg, 2021-12-21:
If the MqTT stuff was using preemptive tasks I would expect all of the sliders and labels to have the same issue.

kisvegabor, 2021-12-22:
Do you delete and create this switch dynamically in run time? If so maybe a variable points to an already freed widget.

elgerg, 2021-12-22:
No, its a statically created switch. It doesnt get removed.

elgerg, 2021-12-22:
Heating.h:

```c
/*
 * Heating.h
 *
 *  Created on: 9 Oct 2020
 *      Author: alex
 */

#ifndef TABS_HEATING_H_
#define TABS_HEATING_H_

#include <App/App.h>
#include <mosquitto.h>

lv_obj_t * hwsw;
lv_obj_t * htsw;
lv_obj_t * lblHWstatus;
lv_obj_t * lblHTstatus;

void create_tabHeating(lv_obj_t * parent);
void update_hw(bool publishUpdate);
void update_heating(bool publishUpdate);

void hw_event_cb(lv_event_t * event);
void btn_hw_boost15_cb(lv_event_t * event);
void btn_hw_boost60_cb(lv_event_t * event);
void btn_ht_boost15_cb(lv_event_t * event);
void btn_ht_boost60_cb(lv_event_t * event);
void heating_ddl_cb(lv_event_t * event);
void heating_event_cb(lv_event_t * event);

#endif /* TABS_HEATING_H_ */
```

Heating.c
```c
/*
 * Heating.c
 *
 *  Created on: 9 Oct 2020
 *      Author: alex
 */

#define _XOPEN_SOURCE 500

#include <App/Tabs/Heating.h>

/*
*******************************************
*
*   Heating
*
*******************************************
*/

void create_tabHeating(lv_obj_t * parent)
{

  int widthfromcenter = 200;
  int heightFromCenter = 150;

  /*
     Hot water
  */

  hwsw = lv_switch_create(parent);
  lv_obj_align(hwsw, LV_ALIGN_CENTER, -widthfromcenter, -heightFromCenter);

[... 1524 lines cut ...]
	json_object_put(rangeData_struct);
	json_object_put(payload);

}

void handleRegistrationItem(lv_obj_t * range, lv_obj_t * label, struct json_object *rangeData_struct, char* rangeName)
{

	struct json_object *payload;
	json_object_object_get_ex(rangeData_struct, rangeName, &payload);
    double rangeValue = json_object_get_double(payload);
	if(debugLevel >= 1)
	{
		printDateTime();
	    printf("%s %d\n", rangeName, json_object_get_int(payload));
	}

    if(rangeValue > 0)
    {
    	setSlider(range, label, rangeValue);
    }

	json_object_put(payload);
}
```

elgerg, 2021-12-24:
Hi, I think (think) I might have found the problem. I believe it may have been caused by the duplicate definition of the switch between the heating.h and app.h files.

Now I've removed one of them it seems to be more stable.

I'll close this off and open a new issue should it not be resolved.

Thanks!
Alex

closed 2021-12-24 by elgerg