lvgl/lvgl #2930: adding flex config to const style will cause unknown background color adding to children

lee88688, 2021-12-21:
currently I'm working on combining lvgl with react framework on [emscripten](https://github.com/lvgl/lv_sim_emscripten). since lvgl style is very close to css, I write a parser for converting css to lvgl style object which use const styles for my purpose.
when I add flex config to const style, I find a black background in some child element, my css style like this
```css
// container
.s1 {
  left: 100px;
  top: 0px;
  width: 500px;
  height: 300px;
  display: flex;
  flex-direction: row;
}
// children
.s3 {
  width: 50px;
  height: 50px;
}
```
and element structure like this
```html
<obj className={styles.s1}>
  <obj className={styles.s3}></obj>
</obj>
```
this will result to a black bround to my child
[image]

### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version, I'm using latest v8.1 branch
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
when I replay it using c code, I find it is relate to flex display config. I create a config close to before.
```c
const char *eval(const char *str) {
    lv_style_const_prop_t tmp1[] = {
        LV_STYLE_CONST_X(100),
        // LV_STYLE_CONST_Y(0), // uncomment this will not get wrong
        LV_STYLE_CONST_WIDTH(500),
        LV_STYLE_CONST_HEIGHT(300),
        LV_STYLE_CONST_LAYOUT(LV_LAYOUT_FLEX),
        { .prop=LV_STYLE_FLEX_FLOW, .value={ .num=LV_FLEX_FLOW_ROW } },
        { .prop=LV_STYLE_FLEX_MAIN_PLACE, .value={ .num=LV_FLEX_ALIGN_START } },
        { .prop=LV_STYLE_FLEX_CROSS_PLACE, .value={ .num=LV_FLEX_ALIGN_START } },
        { .prop=LV_STYLE_FLEX_TRACK_PLACE, .value={ .num=LV_FLEX_ALIGN_START } },
    };

    lv_style_const_prop_t *container_style_prop_value = malloc(sizeof(tmp1));
    memcpy(container_style_prop_value, tmp1, sizeof(tmp1));
    lv_style_t *container_style = malloc(sizeof(lv_style_t));
    *container_style = (lv_style_t) { .v_p = { .const_props = container_style_prop_value }, .has_group = 0xFF, .is_const = 1, .prop_cnt=countof(tmp1) };

    lv_style_const_prop_t tmp2[] = {
        LV_STYLE_CONST_WIDTH(50),
        LV_STYLE_CONST_HEIGHT(50),
    };
    lv_style_const_prop_t *child_style_prop_value = malloc(sizeof(tmp2));
    memcpy(child_style_prop_value, tmp2, sizeof(tmp2));
    lv_style_t *child_style = malloc(sizeof(lv_style_t));
    *child_style = (lv_style_t) { .v_p = { .const_props = child_style_prop_value }, .has_group = 0xFF, .is_const = 1, .prop_cnt=countof(tmp2) };

    lv_obj_t *container = lv_obj_create(lv_scr_act());
    lv_obj_add_style(container, container_style, 0);
    lv_obj_t *c = lv_obj_create(container);
    lv_obj_add_style(c, child_style, 0);

    ret = get_screen_obj_info();
    return NULL;
}
```
and this code will generate screen below on emscripten
[image]
if I uncomment `LV_STYLE_CONST_Y(0)` this line, it will be right. so I don't know if this were caused by flex config in const styles.

### Expected behavior
when I use flex config in const styles, it will get right background.

kisvegabor, 2021-12-22:
Const styles doesn't use `.prop_cnt` but waits for a special `LV_STYLE_PROP_INV` prop as last property. It's because it's not trivial to get the property count in compile time. 

See here:
https://github.com/lvgl/lvgl/blob/935caa64e4c33d63ec937a323d93d53730ec487e/src/misc/lv_style.h#L360-L366

There should be a paragraph about const style in the docs... :slightly_frowning_face:

lee88688, 2021-12-23:
I'm very sorry for that. and I don't find the docs neither. can you indicate it for me? thank you.

kisvegabor, 2022-01-03:
I added a paragraph at the end of [this section](https://docs.lvgl.io/master/overview/style.html#initialize-styles-and-set-get-properties).

I close this issue, if it's still not working feel free to comment here.

closed 2022-01-03 by kisvegabor