lvgl/lvgl #2932: v8 performance minotor bug

xennex22, 2021-12-21:
There is a bug in commit https://github.com/lvgl/lvgl/commit/6e00724797caf63deff0a46abeb5506650f88afe for issue https://github.com/lvgl/lvgl/pull/2910

`perf_monitor.perf_label` is never going to created because a local variable is assigned the value and the static struct never updated.
https://github.com/lvgl/lvgl/blob/28d06b03543a9c809d22784f9648607a2f369585/src/core/lv_refr.c#L281-L283
I've fixed this on my forked copy but can't do a PR while there is already an outstanding PR.

xennex22, 2021-12-21:
Also I see the same problem for the mem_monitor
https://github.com/lvgl/lvgl/blob/28d06b03543a9c809d22784f9648607a2f369585/src/core/lv_refr.c#L330-L332

mysterywolf, 2021-12-21:
https://github.com/lvgl/lvgl/pull/2910#issuecomment-998056654
https://github.com/lvgl/lvgl/pull/2910#issuecomment-998508245

kisvegabor, 2021-12-21:
Thanks, I've pushed a fix.

commit 0788d918 (2021-12-21): fix(mem/perf monitor): fix issue introduced in #2910

fixes: #2932
closed 2021-12-21 by kisvegabor