lvgl/lvgl #2936: roller obj only uses vertical scrolling, but it also occupies horizontal scrolling

guoweilkd, 2021-12-22:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

roller obj only uses vertical scrolling, but it also occupies horizontal scrolling.

### To Reproduce
I want to swipe to the right, but roller absorbs horizontal scrolling and cannot do it.

```c
    lv_obj_t *par = lv_obj_create(lv_scr_act());
    lv_obj_center(par);
    lv_obj_set_size(par, 400, 400);
    lv_obj_set_style_border_width(par, 1, 0);
    lv_obj_set_style_border_color(par, lv_palette_main(LV_PALETTE_ORANGE), 0);
    for (int col = 0; col < 5; col++) {
        if (col == 0) {
            lv_obj_t *roller = lv_roller_create(par);
            lv_obj_set_size(roller, 380, 380);
            lv_obj_center(par);
        } else {
            lv_obj_t *child = lv_obj_create(par);
            lv_obj_set_pos(child, col * 385, 0);
            lv_obj_set_size(child, 380, 380);
        }
    }
```

kisvegabor, 2021-12-22:
It's because the roller has 
```c
lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLL_CHAIN);
```

So you can:
```c
lv_obj_add_flag(roller, LV_OBJ_FLAG_SCROLL_CHAIN);
```
But be sure the there are no vertically scrollable parents.

When I tested you code, I saw some simplification options. Maybe you find something useful in it:
```c
     lv_obj_t *par = lv_obj_create(lv_scr_act());
     lv_obj_center(par);
     lv_obj_set_size(par, 400, 400);
     lv_obj_set_style_border_width(par, 1, 0);
     lv_obj_set_style_border_color(par, lv_palette_main(LV_PALETTE_ORANGE), 0);
     lv_obj_set_scroll_snap_x(par, LV_SCROLL_SNAP_CENTER);
     lv_obj_set_flex_flow(par, LV_FLEX_FLOW_ROW);
     for (int col = 0; col < 5; col++) {
         if (col == 0) {
             lv_obj_t *roller = lv_roller_create(par);
             lv_obj_set_size(roller, lv_pct(100), lv_pct(100));
             lv_obj_add_flag(roller, LV_OBJ_FLAG_SCROLL_CHAIN);
         } else {
             lv_obj_t *child = lv_obj_create(par);
             lv_obj_set_size(child, lv_pct(100), lv_pct(100));
         }
     }
```

guoweilkd, 2021-12-23:
As you said, if the parent has vertical scrolling, there will be some interaction problems.Can we solve this problem from a more general level？
At present, our scrolling is for the size of obj, without considering the scrolling inside obj.we can subdivide LV_OBJ_FLAG_SCROLL_CHAIN into LV_OBJ_FLAG_SCROLL_CHAIN_HOR and LV_OBJ_FLAG_SCROLL_CHAIN_VER.  If vertical scrolling is used inside obj, then it can only enable horizontal scrolling. Such as the following code, I used LV_OBJ_FLAG_USER_1 to indicate LV_OBJ_FLAG_SCROLL_CHAIN_VER to do a simple test：

```c
static lv_obj_t * find_scroll_obj(_lv_indev_proc_t * proc)
{
    lv_obj_t * obj_candidate = NULL;
    lv_dir_t dir_candidate = LV_DIR_NONE;
    ..........
    lv_obj_t * obj_act = proc->types.pointer.act_obj;
    while(obj_act) {
        /*Decide if it's a horizontal or vertical scroll*/
        bool hor_en = false;
        bool ver_en = false;
        if(LV_ABS(proc->types.pointer.scroll_sum.x) > LV_ABS(proc->types.pointer.scroll_sum.y)) {
            hor_en = true;
        }
        else {
            ver_en = true;
        }

        if(lv_obj_has_flag(obj_act, LV_OBJ_FLAG_SCROLLABLE) == false) {
            /*If this object don't want to chain the scroll ot the parent stop searching*/
            if(lv_obj_has_flag(obj_act, LV_OBJ_FLAG_SCROLL_CHAIN) == false) break;

            if (lv_obj_has_flag(obj_act, LV_OBJ_FLAG_USER_1) == true &&  ver_en) {
                break;
            }
            obj_act = lv_obj_get_parent(obj_act);
            continue;
        }
```

kisvegabor, 2022-01-04:
I was thinking about the same and it's more general. Actually, it it really makes sens to enable/disable scroll chain in an directions. 

If we add `LV_OBJ_FLAG_SCROLL_CHAIN_HOR/VER` we can also add
```c
#define LV_OBJ_FLAG_SCROLL_CHAIN (LV_OBJ_FLAG_SCROLL_CHAIN_HOR | LV_OBJ_FLAG_SCROLL_CHAIN_VER)
``` 
for backward compatibility.

If you have time could you send a PR?

kisvegabor, 2022-01-07:
#2961 is merged so it can be closed.

Thank you for the PR.

closed 2022-01-07 by kisvegabor