lvgl/lvgl #2937: Problem with text rendering when switching to LVGL's own Kconfig

brgl, 2021-12-22:
I'm working on bringing lvgl v8.1.0 support to Zephyr (which currently uses v7). During the review of my PRs it was suggested that we should switch to using LVGL's own Kconfig to allow for easier upgrades in the future.

I have a version of Zephyr using lvgl v8.1.0 and its own Kconfig + custom lv_conf.h and a second one which tries to use the lvgl supplied Kconfig. The code can be inspected [here](https://github.com/brgl/zephyr/commit/db77dba98ed3c3f4cc34bd8c4e2b44ecf95e25b7).

There's a problem in my changeset: it causes the text to be blury/lowers the resolution/turns off antialiasing. I'm not sure which one but it looks like this:

Good one:
[image]

Bad one:
[image]

Any hints as to which config options should I look at? The color depth and DPI are the same on both. So far I failed to identify the one that causes this and the patch itself is pretty big so I thought I'd ask here before going line by line.

kisvegabor, 2021-12-22:
Hi,

I suspect that it's because somehow `disp_drv.antialiasing` is set to `0`. Could it be the case?

brgl, 2021-12-23:
@kisvegabor I tracked my disp_drv.antialiasing with gdb and past the initial value of 1 set in lv_disp_drv_init (because the color depth is 32) it never gets modified again.

kisvegabor, 2021-12-23:
Please attach the `autoconf.h` (the output of Kconfig, I'm not sure if it is the filename).

Is there a `disp_drv.set_px_cb` set? If so, please copy that function here too.

brgl, 2021-12-23:
Here's the `.config` output by Kconfig:

```
# CONFIG_GPIO is not set
# CONFIG_SPI is not set
# CONFIG_I2C is not set
CONFIG_LVGL_VDB_SIZE=10
CONFIG_LVGL_BITS_PER_PIXEL=32
# CONFIG_KSCAN is not set
# CONFIG_WIFI is not set
# CONFIG_MODEM is not set
# CONFIG_BUILD_OUTPUT_BIN is not set
CONFIG_BUILD_OUTPUT_EXE=y
# CONFIG_OUTPUT_PRINT_MEMORY_USAGE is not set
CONFIG_BOARD="native_posix_64"
CONFIG_LOG_BACKEND_NATIVE_POSIX=y
CONFIG_NATIVE_POSIX_CONSOLE=y
CONFIG_SDL_DISPLAY=y
# CONFIG_EMUL is not set
CONFIG_SOC="inf_clock"

#
# Modules
#

#
# Available modules.
#

#
# CANopenNode (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/lib/canopennode)
#
CONFIG_ZEPHYR_CANOPENNODE_MODULE=y
# end of CANopenNode (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/lib/canopennode)

#
# hal_gigadevice (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/hal/gigadevice)
#
CONFIG_ZEPHYR_HAL_GIGADEVICE_MODULE=y
# end of hal_gigadevice (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/hal/gigadevice)

#
# hal_nordic (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/hal/nordic)
#
CONFIG_ZEPHYR_HAL_NORDIC_MODULE=y
# end of hal_nordic (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/hal/nordic)

#
# hal_nxp (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/hal/nxp)
#
CONFIG_ZEPHYR_HAL_NXP_MODULE=y
# end of hal_nxp (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/hal/nxp)

#
# loramac-node (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/lib/loramac-node)
#
CONFIG_ZEPHYR_LORAMAC_NODE_MODULE=y
# CONFIG_HAS_SEMTECH_RADIO_DRIVERS is not set
# end of loramac-node (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/lib/loramac-node)

#
# lz4 (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/lib/lz4)
#
CONFIG_ZEPHYR_LZ4_MODULE=y
# CONFIG_LZ4 is not set
# end of lz4 (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/lib/lz4)

#
# mbedtls (/home/brgl/workspace/ostc/lvgl_v8/zephyrproject/modules/crypto/mbedtls)
#
[... 1255 lines cut ...]
#define CONFIG_LINKER_SORT_BY_ALIGNMENT 1
#define CONFIG_SRAM_OFFSET 0x0
#define CONFIG_LINKER_GENERIC_SECTIONS_PRESENT_AT_BOOT 1
#define CONFIG_NATIVE_APPLICATION 1
#define CONFIG_SIZE_OPTIMIZATIONS 1
#define CONFIG_COMPILER_COLOR_DIAGNOSTICS 1
#define CONFIG_COMPILER_OPT ""
#define CONFIG_RUNTIME_ERROR_CHECKS 1
#define CONFIG_KERNEL_BIN_NAME "zephyr"
#define CONFIG_OUTPUT_STAT 1
#define CONFIG_OUTPUT_DISASSEMBLY 1
#define CONFIG_COMPAT_INCLUDES 1
```

And the set_px_cb callback:

```
void lvgl_set_px_cb_32bit(lv_disp_drv_t *disp_drv,
		uint8_t *buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y,
		lv_color_t color, lv_opa_t opa)
{
	uint32_t *buf_xy = (uint32_t *)(buf + x * 4U + y * 4U * buf_w);
	*buf_xy = lv_color_to32(color);
}
```

kisvegabor, 2021-12-23:
If `LV_COLOR_DEPTH` is 32 and the display's color depth is also 32 you don't need a `set_px_cb`. `set_px_cb` is used to allow writing pixels into buffers with non-RGB format (typically monochrome and grayscale displays)

You should simply use `disp_drv.set_px_cb = NULL` or not set is at all as  `NULL` is the default.

Regardless to all these it should work with `set_px_cb` too but there is a minor bug in `lvgl_set_px_cb_32bit`. It doesn't take `opa` into account. This one should work:
```c
void lvgl_set_px_cb_32bit(lv_disp_drv_t *disp_drv,
		uint8_t *buf, lv_coord_t buf_w, lv_coord_t x, lv_coord_t y,
		lv_color_t color, lv_opa_t opa)
{
	uint32_t *buf_xy = (uint32_t *)(buf + x * 4U + y * 4U * buf_w);
	if(opa == LV_OPA_COVER) {  //Do not mix if not required
              *buf_xy = lv_color_to32(color);
        } else {
           lv_color_t bg_color = *((lv_color_t*) buf_xy); 
           *buf_xy = lv_color_to32( lv_color_mix(color, bg_color, opa));
       }
}
```

brgl, 2022-01-03:
Thanks @kisvegabor this worked fine. I'll integrate this into my Zephyr PR and give you a shoutout.

kisvegabor, 2022-01-03:
Great! I close this issue then.

closed 2022-01-03 by kisvegabor