lvgl/lvgl #2942: Some problems using lvgl 8.1 3rd image Libraries

xiewensir, 2021-12-24:
Hello, Mr/Ms. I met two problems. Could you please do me a favor? One is that when I call  lv_img_set_src(img, "e32.bmp") and show this e32.bmp(which is 32 bit) on the window, there are many random vertical lines on the bmp picture in version 8.1. And if I move my cursor on the bmp picture,the random vertical lines are erased. The previous version 8.0 doesn't have this problem. Secondly, in Version 8.1, when I load a .jpg picture,which size is 25.6KB, I need to malloc 6400KB memory to store this .jpg picture, or it shows [Error] (4.035, +0)      lv_mem_alloc: used:  13136 (  3 %), frag:   1 %, biggest free: 642112  (in lv_mem.c line #143) with mallocing 640 KB memory. Look forward to your reply.Thanks a lot.

kisvegabor, 2022-01-04:
Hi,

>  One is that when I call lv_img_set_src(img, "e32.bmp") and show this e32.bmp(which is 32 bit) on the window, there are many random vertical lines on the bmp picture in version 8.1. And if I move my cursor on the bmp picture,the random vertical lines are erased. 

Please attache the image file and a minimal code snippet to reproduce the issue.

> Secondly, in Version 8.1, when I load a .jpg picture,which size is 25.6KB, I need to malloc 6400KB memory to store this .jpg picture

We support a custom format called sjpg. It's like JPG but can be decompressed with much less RAM.
See https://docs.lvgl.io/master/libs/sjpg.html

Regarding

closed 2022-02-11 by lvgl-bot

adokitkat, 2023-04-26:
I also seem to have problem with vertical lines when using 32 bit BMP pictures. 16 bit BMP works just fine. I use esp32-wrover-kit and I am reading the files from SD card.  Could you help me please?

[image]

```c

#define MOUNT_POINT "/sdcard"
#define BMP_16BIT "/esp16.bmp"
#define BMP_32BIT "/esp24.bmp"

// ...

static lv_obj_t *pic;

void example_lvgl_bmp(lv_obj_t *scr) {
    pic = lv_img_create(scr);
#ifdef CONFIG_LV_COLOR_DEPTH_16
    lv_img_set_src(pic, "A:"MOUNT_POINT BMP_16BIT);
#elif CONFIG_LV_COLOR_DEPTH_32
    lv_img_set_src(pic, "A:"MOUNT_POINT BMP_32BIT);
#else
    #error "Unsupported LVGL color depth"
#endif
    lv_obj_center(pic);
}
```

kisvegabor, 2023-04-26:
Hi,

LVGL doesn't convert the color depth of BMP images. So the image's color depth should match `LV_COLOR_DEPTH`.

adokitkat, 2023-04-26:
I have 2 different pictures with different bit depths and I set which to show with which bit color option before compiling, it's all set correctly. However I found out esp32 -wrover-kit may not be able to correctly handle 32 bit image depth anyway. Max support seems to be 18 bits so 16 bit option should be used.