lvgl/lvgl #2948: lv_img_create() and lv_img_set_src() not working with commit 'allow replacing the draw engine'

NeroBurner, 2021-12-30:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version (yes done with git bisect)
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)
  - yes, doing it porting Infinitime to LVGL8 and using a pc-simulator based on https://github.com/lvgl/lv_sim_eclipse_sdl

### Describe the bug

`lv_img` is broken (random noise instead of image) on PineTime and Simulator starting with db53ea925c9502b20f38db0fc30c4ef599bdfc33

The commit before is fine

### To Reproduce

```c
const lv_img_dsc_t disc = /* picture-array-definition */;
lv_obj_t* imgDisc = lv_img_create(lv_scr_act());
lv_img_set_src_arr(imgDisc, &disc);
```

### Expected behavior
image should be shown as before

### Screenshots or video

screenshot with 0831b6ba761bf68e535a158ff93ef3551d2bd108

[image]

screenshot with db53ea925c9502b20f38db0fc30c4ef599bdfc33

[image]

kisvegabor, 2022-01-03:
Hi,

`lv_img_set_src_arr` is not a lvgl function. Did you mean `lv_img_set_src`?

Please send the converted image too that you want to set.

NeroBurner, 2022-01-03:
yes, sorry, the `_arr` function was an internal wrapper for the `lv_img_set_src` function

The original png file:

[image]

cpp-file can be found at https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/displayapp/icons/music/disc.cpp and also uploaded here and extended with `.txt` (GitHub has a file extension filter). 
[disc.cpp.txt](https://github.com/lvgl/lvgl/files/7802444/disc.cpp.txt)

Should I convert the declaration to a pure C file or is the C++ file OK?

kisvegabor, 2022-01-03:
Thanks, I've fixed it in 109255077 and 5a0dbccf8.

Please, test it and let me know if it works on your end too.

NeroBurner, 2022-01-03:
works like a charm, thank you very much! :)

kisvegabor, 2022-01-03:
Great, thank for the feedback! :slightly_smiling_face:

closed 2022-01-03 by kisvegabor