lvgl/lvgl #2956: lv_demo_widgets(); hangs immediately

epikao, 2022-01-03:
Hello

I'm using platformIO with following setup:
```
[env:disco_f746ng]
platform = ststm32
board = disco_f746ng
framework = arduino
lib_deps = lvgl/lvgl@^8.1.0
```

I'm using lvgl version with merged demos. My demo folder is src/demos (in the main-file I use #include <demos/lv_demos.h>). 
All demos works, except lv_demo_widgets();
This demo hangs immediately, because I've no debug output, no LED blinking (see loop code below).

What could that be? I've no idea...

Thank you

```
void loop(void)
{
  lv_timer_handler(); /* let the GUI do its work */
  delay( 5 );

  if( (millis()-lastDelayMillis) > 500){
    digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
    lastDelayMillis = millis();
  }
}
```

kisvegabor, 2022-01-03:
I've just fixed a bug a  few minutes ago: https://github.com/lvgl/lvgl/issues/2943

epikao, 2022-01-03:
> I've just fixed a bug a few minutes ago: #2943

I've checked file lv_demo_widgets.c, there is no function with "lv_draw_polygon..", no query with "code == LV_EVENT_DRAW_POST" and "lv_obj_set_grid_cell..." looks correct, so how could that relate to this problem?

kisvegabor, 2022-01-03:
Oh sorry, I misread the title :slightly_smiling_face: 
Does it work if you increase `LV_MEM_SIZE`?

epikao, 2022-01-03:
> Oh sorry, I misread the title 🙂 Does it work if you increase `LV_MEM_SIZE`?

nop, already tried with: 
`    #define LV_MEM_SIZE (32U * 2048)          /*[bytes]*/`

then the screen changes to black...

kisvegabor, 2022-01-03:
Then please [enable logging](https://docs.lvgl.io/master/porting/log.html) to see when could be the problem.

epikao, 2022-01-03:
> Then please [enable logging](https://docs.lvgl.io/master/porting/log.html) to see when could be the problem.

since the program hangs already at startup, the debug output does not help either. The program never comes into the loop.

ahmetalperenbulut, 2022-01-03:
Hi, 
Could you change the LV_MEM_CUSTOM  value to 1 ?
Today i faced a same problem.

epikao, 2022-01-04:
> Hi,
> Could you change the LV_MEM_CUSTOM value to 1 ?
> Today i faced a same problem.

Yes, now it works, interesting.... but I don't quite understand the context - but never mind :-D

ahmetalperenbulut, 2022-01-04:
@kisvegabor 
It looks like a bug.

kisvegabor, 2022-01-04:
Hm, I'm pretty sure that LVGL's malloc works well. If it wasn't it would cause a lot of exotic issues everywhere. 

I'd be happy to help to track it, but need more info. Exactly where it halts? Does a debugger show something interesting?

closed 2022-01-26 by lvgl-bot