lvgl/lvgl #2957: Release 8.1.0 not available in Arduino

patricklaf, 2022-01-03:
The latest available version in Arduino Library Manager is 8.0.2.

kisvegabor, 2022-01-05:
It should be fixed as described [here](https://github.com/lvgl/lvgl/commit/c60ed68e94c7ab08a8286ee8415d43f5cfa1e7f9#r62838038) by @giulcioffi.

giulcioffi, 2022-01-05:
Hi @kisvegabor, thanks for fixing it!
In order to correctly publish the library, you would need to make a new release with a new tag. If you don't release this last commit, the job that checks the library will not fetch it and will keep failing.

kisvegabor, 2022-01-05:
Oh, I see. Shall it be a semver like tag, e.g. `v8.2.0`?

giulcioffi, 2022-01-05:
Yes, it should be okay. Just remember to update the version also in the `library.properties` file.

kisvegabor, 2022-01-05:
I see, thanks. 

We weren't planning a release, but due to [other things](https://github.com/lvgl/lvgl/issues/2790#issuecomment-1005809513) too maybe we should have one soon.

giulcioffi, 2022-01-05:
Okay, then once you make the release you can check if the release process succeeded just looking at the latest available library version on the Arduino Library Manager. Thanks :)

kisvegabor, 2022-01-20:
@giulcioffi Is there a way to test if the current version of lvgl is compatible with Arduino's standard?

giulcioffi, 2022-01-20:
Hi @kisvegabor, in general you can refer to the Library Manager registry documentation: https://github.com/arduino/library-registry/blob/main/FAQ.md#what-are-the-requirements-for-publishing-new-releases-of-libraries-already-in-the-library-manager-list.
In addition to this, you could run a check with Arduino Lint. This can be done via GitHub Actions using a workflow similar to the one we use in all of our libraries:
https://github.com/arduino-libraries/ArduinoIoTCloud/blob/master/.github/workflows/check-arduino.yml .

kisvegabor, 2022-01-20:
Thanks, I've just added the action and it haven't found error or warnings.
https://github.com/lvgl/lvgl/runs/4881972210?check_suite_focus=true

PierreRambaud, 2022-01-25:
For people like me who want to use a recent version of this awesome library.
I clone the repository inside `~/Arduino/libraries/lvgl` and I'm now able to use the v8.1.0 or even the master branch, it's maybe not the best approach, but at least, I'm now able to use recent features and maybe one day, I'll be enough skilled to contribute :sweat_smile:

Xtreemtec, 2022-01-25:
Trying to setup a LVGL under Arduino 1.8.16   for ESP32  
Installed lvgl version 8.0.2   and lv_examples 8.1.0    These are the latest available in the arduino library manager at this moment.

Following the Getting Started https://docs.lvgl.io/master/get-started/arduino.html    But it's really confusing.  Did the 2 conf files to the root of libraries and configed them as mentioned. 
But then "Use examples and Demo's"   I did this and the Arduino system keeps trowing errors that lvgl/lvgl.h is not found. 
The instruction also mentions to move lvgl/demos > lvgl/src/demos    But there is no demos folder in the lvgl that came with the library manager from Arduino...  

It's not really clear what is going wrong here..  Looking at the github the folder lvgl/demos does exist.   So looks like the Arduino installer misses things..  

Going to try to rebuild the folders manually now from this github.  But the Arduino instructions are not too clear for users that have not worked with Lvgl before.   ;)   Might be that the info needs an update or at least a better explanation what needs to be done would be appreciated.

Xtreemtec, 2022-01-25:
Oke placed demos folder from the github in lvgl folder.  
Trowing this error that lv_demo wrong lvgl version..    
So in the end i removed the full lvgl folder, pulled a new direct from Github.  
Re-did the config files in the Library Root.  
Recompile, working out serveral compiler issues due to code that isn't compatible with the board i'm trying to make compatible ( Wireless Tag  WT32-SC01  which doesn't support touchscreen from the TFT_eSPI library. 

Then hit a snag due to illegal conversion errors within the lvgl code..   

Any change somebody else could hit compile inside Arduino with ESP32 Wrover selected and find what is going wrong?
Because i'm not sure if it my Arduino setup, my failure in following setup instructions,  maybe i missed something or Arduino isnt able to handle it...   (These errors i got with compiling the lvgl Github  25-1-22 21:50 GMT.  ) 

Some copy paste from the a few of dozen conversion errors:

C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c: In function '_lv_demo_music_main_create':
C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:111:19: error: 'lv_font_montserrat_12' undeclared (first use in this function); did you mean 'lv_font_montserrat_14'?
     font_small = &lv_font_montserrat_12;
                   ^~~~~~~~~~~~~~~~~~~~~
                   lv_font_montserrat_14
C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:111:19: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:112:19: error: 'lv_font_montserrat_16' undeclared (first use in this function); did you mean 'lv_font_montserrat_14'?
     font_large = &lv_font_montserrat_16;
                   ^~~~~~~~~~~~~~~~~~~~~
                   lv_font_montserrat_14
C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c: In function 'spectrum_draw_event_cb':
C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:820:29: warning: passing argument 1 of 'lv_draw_polygon' from incompatible pointer type [-Wincompatible-pointer-types]
             lv_draw_polygon(poly, 4, clip_area, &draw_dsc);
                             ^~~~
In file included from c:\users\bmcreations\documents\arduino\libraries\lvgl\src\draw\lv_draw.h:27,
                 from c:\users\bmcreations\documents\arduino\libraries\lvgl\src/hal/lv_hal_disp.h:21,
                 from c:\users\bmcreations\documents\arduino\libraries\lvgl\src/hal/lv_hal.h:16,
                 from c:\users\bmcreations\documents\arduino\libraries\lvgl\lvgl.h:33,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,
                 from c:\users\bmcreations\documents\arduino\libraries\lv_examples\lv_demo.h:35,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music.h:16,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.h:16,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:9:
c:\users\bmcreations\documents\arduino\libraries\lvgl\src\draw\lv_draw_triangle.h:30:46: note: expected 'struct _lv_draw_ctx_t *' but argument is of type 'lv_point_t *' {aka 'struct <anonymous> *'}
 void lv_draw_polygon(struct _lv_draw_ctx_t * draw_ctx, const lv_draw_rect_dsc_t * draw_dsc, const lv_point_t points[],
                      ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:820:35: warning: passing argument 2 of 'lv_draw_polygon' makes pointer from integer without a cast [-Wint-conversion]
             lv_draw_polygon(poly, 4, clip_area, &draw_dsc);
                                   ^
In file included from c:\users\bmcreations\documents\arduino\libraries\lvgl\src\draw\lv_draw.h:27,
                 from c:\users\bmcreations\documents\arduino\libraries\lvgl\src/hal/lv_hal_disp.h:21,
                 from c:\users\bmcreations\documents\arduino\libraries\lvgl\src/hal/lv_hal.h:16,
                 from c:\users\bmcreations\documents\arduino\libraries\lvgl\lvgl.h:33,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lvgl\src/lvgl.h:17,
                 from c:\users\bmcreations\documents\arduino\libraries\lv_examples\lv_demo.h:35,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music.h:16,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.h:16,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:9:
c:\users\bmcreations\documents\arduino\libraries\lvgl\src\draw\lv_draw_triangle.h:30:83: note: expected 'const lv_draw_rect_dsc_t *' {aka 'const struct <anonymous> *'} but argument is of type 'int'
 void lv_draw_polygon(struct _lv_draw_ctx_t * draw_ctx, const lv_draw_rect_dsc_t * draw_dsc, const lv_point_t points[],
                                                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
[... cut ...]
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.h:16,
                 from C:\Users\BMCreations\Documents\Arduino\libraries\lv_examples\src\lv_demo_music\lv_demo_music_main.c:9:
c:\users\bmcreations\documents\arduino\libraries\lvgl\src\draw\lv_draw_triangle.h:31:31: note: expected 'uint16_t' {aka 'short unsigned int'} but argument is of type 'lv_draw_rect_dsc_t *' {aka 'struct <anonymous> *'}
                      uint16_t point_cnt);
                      ~~~~~~~~~^~~~~~~~~

embeddedt, 2022-01-25:
I don't have the Arduino IDE installed, but on #3048 it was mentioned that disabling the music demo fixes that error.

Xtreemtec, 2022-01-25:
> I don't have the Arduino IDE installed, but on #3048 it was mentioned that disabling the music demo fixes that error.

Thanks!! That fixed a lot of issues indeed.  This should be mentioned in the Documentation of the Arduino setup.  More people will fail on this.   
Been able to compile the LVGL_Arduino.ino now with a white screen and Hello Arduino text.     But still cant get the Demo widgets to compile.   But at least seen some result.  Will play with it some more to see where that fails.. 

So to sum up:  
- Arduino Library doesnt give you all maps.  ( Demos is missing for sure )
- Have to be really carefull how your folder structure is set up.  Instructions could be more clear. See image.
- Music demo must be set to 0 in config.  Else it trows mayor errors
- Oh and slightly confusing to have a lv_demos.h and lv_demo.h   

[image]

kisvegabor, 2022-01-27:
> Been able to compile the LVGL_Arduino.ino now with a white screen and Hello Arduino text. But still cant get the Demo widgets to compile. But at least seen some result. Will play with it some more to see where that fails..

> Arduino Library doesnt give you all maps. ( Demos is missing for sure )

> Have to be really carefull how your folder structure is set up. Instructions could be more clear. See image.

Take a look at this part of the [docs](https://docs.lvgl.io/master/get-started/arduino.html#use-the-examples-and-demos).

> Music demo must be set to 0 in config. Else it trows mayor errors

All demos are disabled by default in the master.

> Oh and slightly confusing to have a lv_demos.h and lv_demo.h

Where is `lv_demo.h`?

Xtreemtec, 2022-01-27:
> All demos are disabled by default in the master.
Pretty sure i did not enable it myself.   I will try to trace it down in what i downloaded.. Maybe that might give some insight why..

> Where is lv_demo.h?
Arduino library > lv_examples(version 8.1.0) > lv_demo.h