lvgl/lvgl #2962: Build failed

xaowang96, 2022-01-05:
When i set LV_COLOR_SCREEN_TRANSP enabled, i got a compile error, like this

[image]

kisvegabor, 2022-01-05:
It should be fixed in the latest master version.

xaowang96, 2022-01-05:
> It should be fixed in the latest master version.

Thanks for it. By the way, about what time is it?

kisvegabor, 2022-01-05:
In db53ea925c

xaowang96, 2022-01-05:
> In [db53ea9](https://github.com/lvgl/lvgl/commit/db53ea925c9502b20f38db0fc30c4ef599bdfc33) [here](https://github.com/lvgl/lvgl/commit/db53ea925c9502b20f38db0fc30c4ef599bdfc33#diff-f48aa9733ddf89ef69050e3d54cb14b09a8fcf19c6a949b3d35d24621c335175R42-R44).

Ok.

kisvegabor, 2022-01-05:
Please test if it really works on your end too so that we can close this issue.

xaowang96, 2022-01-06:
> Please test if it really works on your end too so that we can close this issue.

Sorry for keeping you waiting, i tested everything ok.

embeddedt, 2022-01-06:
Great!

closed 2022-01-06 by embeddedt