lvgl/lvgl #2963: group_del does not clear indev->group

knifter, 2022-01-05:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
indev->group is kept when group is deleted.

When I delete a group (group_del()), that group could still be assigned to an indev (lv_indev_set_group()). 
As the objects in the group are 'told' the group is deleted, I would assume the indev is as well. But it is not. I had to make a workaround for this as I'm deleting groups, setting lv_indev_set_group(indev, nullptr) to remove it.
Not sure if this is intended behaviour but since these kind of things seem to be taken care of in the rest of LVGL I would think this is a bug.

### To Reproduce
```
v_group_t* g = lv_group_create();
lv_indev_set_group(g);
lv_group_del(g);
have some fun with your indev->group
```

### Expected behavior
I think group_del shoudl iterate the indev's to see if anyone has its group set to the one going to be deleted.

### Screenshots or video

knifter, 2022-01-05:
I can make a pull request if you like

knifter, 2022-01-05:
The PR fixes the problem for me

knifter, 2022-01-06:
PR was merged into master

embeddedt, 2022-01-06:
Not yet; I approved it but didn't merge it quite yet as I wanted @kisvegabor to confirm.

kisvegabor, 2022-01-06:
Merged now, thank you! :slightly_smiling_face:

closed 2022-01-06 by kisvegabor