lvgl/lvgl #2967: lv_tableThere is a memory leak

Feng573795502, 2022-01-06:
static void lv_table_destructor(const lv_obj_class_t * class_p, lv_obj_t * obj)
{
    LV_UNUSED(class_p);
    lv_table_t * table = (lv_table_t *)obj;
    /*Free the cell texts*/
    uint16_t i;

    for(i = 0; i < table->col_cnt * table->row_cnt; i++) {
        if(table->cell_data[i]) {
            lv_mem_free(table->cell_data[i]);
            table->cell_data[i] = NULL;
        }
    }
		
    if(table->cell_data) lv_mem_free(table->cell_data);
    if(table->row_h) lv_mem_free(table->row_h);
}

Need to add if(table->col_w) lv_mem_free(table->col_w);

X-Ryl669, 2022-01-06:
Thanks. Can you submit a pull request ?

ChinaKServer, 2022-01-14:
I also encountered this problem. I thought the previous version had fixed this problem. After looking for it for a long time, the result was that the table component leaked. Each release would leak about 52 bytes

kisvegabor, 2022-01-15:
I added the changes as suggested by @Feng573795502

commit 8d52de14 (2022-01-15): fix(table, chart): fix memory leaks

fixes #2967
closed 2022-01-15 by kisvegabor