lvgl/lvgl #2972: Watch Dog Timeout in lv_task_handle for ESP IDF  v4.3.0 onwards

quaizar-chefman, 2022-01-06:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Have been using LVGL v7.7.1 with ESP IDF v4.1.1 successfully for many past releases.  However when we updated our ESP32 IDF to v4.3.1 the application code will randomly (and frequently) hang in lv_task_handle followed by a wdt timeout.  Backtrace from the hang leads right back to:  lv_refr_vdb_flush().  Updated LVGL to v7.11.0, and problem still persist.  

Full Backtrace log is as follows:
 Backtrace:0x4008DBDD:0x3FFBF550 0x400844F9:0x3FFBF570 0x40167B8C:0x3FFF0EE0 0x401680CB:0x3FFF0F50 0x4016867D:0x3FFF0FE0 0x401713DD:0x3FFF1030 0x401714E0:0x3FFF1050 0x400F7C36:0x3FFF1080 0x400E8887:0x3FFF10A0 0x400E3570:0x3FFF1160 0x400DF636:0x3FFF1270 0x400DF723:0x3FFF1290 
0x4008dbdd: esp_crosscore_isr at /home/murtuza/esp/esp-idf/components/esp32/crosscore_int.c:77

0x400844f9: _xt_lowint1 at /home/murtuza/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40167b8c: lv_refr_vdb_flush at /home/murtuza/esp/rj40-esp32-w2/rj40-esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:941

0x401680cb: lv_refr_area_part at /home/murtuza/esp/rj40-esp32-w2/rj40-esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:576

0x4016867d: lv_refr_area at /home/murtuza/esp/rj40-esp32-w2/rj40-esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:478
 (inlined by) lv_refr_areas at /home/murtuza/esp/rj40-esp32-w2/rj40-esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:400
 (inlined by) _lv_disp_refr_task at /home/murtuza/esp/rj40-esp32-w2/rj40-esp32/build/../components/lvgl/lvgl/src/lv_core/lv_refr.c:199

0x401713dd: lv_task_exec at /home/murtuza/esp/rj40-esp32-w2/rj40-esp32/build/../components/lvgl/lvgl/src/lv_misc/lv_task.c:386 (discriminator 1)

0x401714e0: lv_task_handler at /home/murtuza/esp/rj40-esp32-w2/rj40-esp32/build/../components/lvgl/lvgl/src/lv_misc/lv_task.c:134

### To Reproduce

### Expected behavior

I expected the WDT to not timeout.  But here we are.  Bumped up the WDT to max in menuconfig, and yet it still hangs.  I understand this is a platform related problem to ESP32, so any help or useful information that i can provide on the ESP32 form would be helpful as well.    

### Screenshots or video

chegewara, 2022-01-09:
Hi, 
after spending few hours with espressif example on S2 kaluga board i finally found the solution, just dont know if it is good solution, and then i found this issue.

To me problem is solved when i add `draw_buf->flushing = 0;` in this line:
https://github.com/lvgl/lvgl/blob/v8.0.2/src/core/lv_refr.c#L481

chegewara, 2022-01-09:
Hi,
after spending few hours with espressif example on S2 kaluga board i finally found the solution, just dont know if it is good solution, and then i found this issue.

To me problem is solved when i add `draw_buf->flushing = 0;` in this line:
https://github.com/lvgl/lvgl/blob/v8.0.2/src/core/lv_refr.c#L919

embeddedt, 2022-01-09:
That suggests `lv_disp_flush_ready` is not being called at the correct time, or there is memory corruption somewhere, as that function should be doing exactly that.

chegewara, 2022-01-09:
Thanks @embeddedt 
Looks like espressif example is missing call `lv_disp_flush_ready(drv);` in flush callback for ST7789. That solved my issue.

C47D, 2022-01-09:
Were you using the `lv_port_esp32` example project?

quaizar-chefman, 2022-01-10:
I appreciate the team coming together to answer this question.  This graphical project was updated to lvgl 7.7.1 on esp idf v4.1 around 13 months ago.  At the time i did use `lv_port_esp32` example project  as a starting point.  

As per @chegewara comment about espressif example missing call to `lv_disp_flush_ready(drv)` in the flush callback for ST7789.  He is correct, if he is referencing this: https://github.com/lvgl/lvgl_esp32_drivers/blob/9fed1cc47b5a45fec6bae08b55d2147d3b50260c/lvgl_tft/st7789.c#L198.  However the  `lv_disp_flush_ready(drv)` is found here: https://github.com/lvgl/lvgl_esp32_drivers/blob/9fed1cc47b5a45fec6bae08b55d2147d3b50260c/lvgl_tft/disp_spi.c#L313 on the spi callback.    

Digging deeper, i see a comment in `lv_hal_disp.h`  about the `flush_cb` that says:

>     /** MANDATORY: Write the internal buffer (VDB) to the display. 'lv_disp_flush_ready()' has to be
     * called when finished */
    void (*flush_cb)(struct _disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p);

Again this is working on IDF 4.1 and 4.2, but somehow not as well on 4.3. Perhaps ESP made some change to the spi callback, and therefore not reliably calling `lv_disp_flush_ready()` as it should?

So for clarity purposes @chegewara  when you said you got it working, did you remove the  'lv_disp_flush_ready()' from the spi callback, and add it to the `st7789_flush` routine? 

And to@C47D,  was `lv_port_esp32` tested for ESP IDF v4.3.0 or v4.3.1?

C47D, 2022-01-10:
> Again this is working on IDF 4.1 and 4.2, but somehow not as well on 4.3. Perhaps ESP made some change to the spi callback, and therefore not reliably calling lv_disp_flush_ready() as it should?

I will look into that, I'm currently using that display driver and ESP-IDF v4.3 for the develop branch.

chegewara, 2022-01-11:
> Were you using the `lv_port_esp32` example project?

Nope, i am using this example, because `lv_port_esp32` is not compatible with idf v5.0:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/lvgl

As you can see we have flush in this callback:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/lvgl/main/lvgl_example_main.c#L55
which is fine for I80 driver but its missing in SPI driver callback:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/lcd/lvgl/main/lvgl_example_main.c#L59-L68

But i think this is not the place to discuss about espressif examples and it should be continues in espressif github.

quaizar-chefman, 2022-01-11:
> * If you do hit the assert then the issue is inside one of the #if blocks.

Instead of an assert, i placed a counter when the flag is set with `DISP_SPI_SIGNAL_FLUSH`, and when the flag is not set.  And in a separate timer task i printed out the counter value every 1 sec.  What i learned is the following:
Before the watch dog happens, the `spi_ready` is called about 120 times per second.  Of which about 20 times the flag is set with `DISP_SPI_SIGNAL_FLUSH`.  But leading up to watch dog trigger, i get 0 calls to  `spi_ready`.  And about 5 seconds latter of that i watch dog.

X-Ryl669, 2022-01-23:
So it's an ESP-IDF issue then (not calling the callback as documented) ?

mquaizar, 2022-01-23:
> So it's an ESP-IDF issue then (not calling the callback as documented) ?

I have a strong suspicious that this is the case.  However waiting for final conclusion from @C47D , and also working on a simplified project to highlight this problem on an official ESP demo board so that the their and confirm and fix.

C47D, 2022-01-24:
Hi, do you think enabling logging in LVGL can help us giving more clues? I've been looking in the ESP-IDF repo for similar issues but I haven't been able to find anything similar (only this one https://github.com/espressif/esp-idf/issues/6624)

X-Ryl669, 2022-01-24:
Did you try this patch ?
https://github.com/espressif/esp-idf/pull/7874/files

If you have multiple device on your SPI bus, it seems there's some change recently on the SPI DMA's burst length and this seems to break people's code. Not sure why, it changes the sequence of event and some interrupts are lost.

mquaizar, 2022-02-11:
Thanks for the suggestion @X-Ryl669, but the patch does not resolve the SPI Callback from locking up. 

@C47D Here is a log i collected from LVGL right before the SPI Locks up:
W (23800) vTask_Timer: 230 / 1160
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: container created 	(lv_cont.c #118 lv_cont_create())
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: container created 	(lv_cont.c #118 lv_cont_create())
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: container created 	(lv_cont.c #118 lv_cont_create())
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: image created 	(lv_img.c #127 lv_img_create())
Info: lv_img_set_src:  `LV_IMG_SRC_VARIABLE` type found 	(lv_img.c #180 lv_img_set_src())
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: container created 	(lv_cont.c #118 lv_cont_create())
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: label created 	(lv_label.c #166 lv_label_create())
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: container created 	(lv_cont.c #118 lv_cont_create())
Info: Object create ready 	(lv_obj.c #470 lv_obj_create())
Info: label created 	(lv_label.c #166 lv_label_create())
Info: image draw: cache miss, close and reuse an entry 	(lv_img_cache.c #119 _lv_img_cache_open())

W (24800) vTask_Timer: 232 / 1175

The counter before the log is 230 / 1160 and  after the log it is 232/1175  which means that the SPI callback happened 15 times between these two time periods, of which 2 times the DISP_SPI_SIGNAL_FLUSH flag was true.

After this moment the SPI callback permanently freezes up.

mquaizar, 2022-02-11:
It is interesting that this issue is not widely replicated.  i will do my best to tear down my application code so that all that is left is the lvgl code and the symptom on the esp32 platform.  wish me luck.

C47D, 2022-02-14:
I got an watchdog timeout when adding another task to try change the display rotation at runtime, after that I removed the other task and tried to change the rotation in the same task as the demo, and it seems to me that `lv_task_handle` is not returning.

```c
    create_demo_application();

    uint32_t counter = 0;
    lv_disp_rot_t rotation = LV_DISP_ROT_NONE;

    while (1) {
        counter++;

        ESP_LOGI("g", "counter: %d", counter);
        if (counter > 100) {
            counter = 0;
            rotation++;

            if (LV_DISP_ROT_270 > rotation) {
                rotation = LV_DISP_ROT_NONE;
            }

            ESP_LOGI("rot", "Setting new rotation %d", rotation);
            lv_disp_set_rotation((lv_disp_t *) &disp_drv, rotation);
        }
        /* Try to take the semaphore, call lvgl related function on success */
        lv_task_handler();
        /* Delay 1 tick (assumes FreeRTOS tick is 10ms */
        // vTaskDelay(pdMS_TO_TICKS(10));

        ESP_LOGI("rot", "here");
    }
```

The monitor output:
[image]

I will try to find out where's it is getting stuck, I guess I will arrive to the same place you did @mquaizar , I don't have a debugger tho

embeddedt, 2022-02-14:
With v8, I think you have to call `lv_disp_set_rotation` on the return value of `lv_disp_drv_register`, as that is a proper `lv_disp_t *`.

C47D, 2022-02-15:
Sorry, I forgot to mention I'm still using v8

embeddedt, 2022-02-15:
I think the same is true in v7.

C47D, 2022-02-16:
@embeddedt how often should `lv_task_handler` should be called? Increasing the SPI clock allow it to return, before the change didn't allowed it to return.

kisvegabor, 2022-02-16:
> how often should lv_task_handler should be called?

Even 30-50 ms (max 20-33 FPS) can be fine.

C47D, 2022-02-16:
@mquaizar how often are you calling `lv_task_handler`? I managed to get rid of the watchdog timeout by calling it more often (by increasing the SPI clock, thus calling `lv_disp_flush_ready` 'faster'.

mquaizar, 2022-02-16:
@C47D I call `lv_task_handler` every 100ms.  But there are times when it can can take longer, like when updating images.

C47D, 2022-02-16:
Is it possible to increase the SPI clock? Just to test if that solution also applies on your use case.

C47D, 2022-02-16:
I'm just guessing, but maybe there's a lot mora data being sent on those flushes. If you have a logic analyzer we might be able to know the transaction time, or maybe just logging the size of the buffers being sent in the flush callback.

I haven't seen any other watchdog timeout since the SPI clock got increased from 2MHz to 20MHz.

mquaizar, 2022-02-17:
That is a great point.  i will investigate this.

closed 2022-03-12 by lvgl-bot

murtuzaq, 2022-04-21:
Guys i solved my issue with Watch Dog Time out.   I went back to review the lv_port_esp32 project and found a discrepancy  between how it allocated buffer memory.  
In my source file, i had allocated buf1 and buf2 to external memory like this:
` lv_color_t* buf1 = malloc(DISP_BUF_SIZE * 2)`
where in menuconfig malloc was to use external memory for malloc's greater then 2k.  

And in lv_port_esp32 example, buf1 and buf2 memory was allocated as:
` lv_color_t* buf1 = heap_caps_malloc(DISP_BUF_SIZE * sizeof(lv_color_t), MALLOC_CAP_DMA);`
which used internal memory.

Since that change, i have not wdt since.

Thank you everybody on this ticket for your considerable help and feedback.  And a special Thank you to @C47D for the example port code that you have given to this community.

C47D, 2022-04-22:
@murtuzaq Thanks for reporting back your finding and the solution. I'm glad the example port helped you!

kisvegabor, 2022-04-26:
Amazing news! Thanks for sharing it!

X-Ryl669, 2022-04-26:
I'm amazed it even worked with DMA between the PSRAM and the screen. That's another reason why we need some heap manager in ESP32 that's linked with LVGL.