lvgl/lvgl #2974: the OLED display is displayed with some broken lines while lvgl is displaying dynamically

glin00, 2022-01-07:
Describe the issue:
I am using zephyr's lvgl as GUI library with zephyr. The display which is an OLED display can be displayed ok while lvgl is displaying any static pictures(attahced good). But the OLED display is displayed with some broken lines(attached brokens) while lvgl is displaying dynamically.

If I remove  lvgl and make firmware just display dynamic pictures(ex. a full screen red , a full screen blue, then a full screen green repeatedly) the broken lines problem will not appear. In this way, I prove that the hardware is not the root cause of this problem.  

Can any one help me how to solve this problem?

To Reproduce
The following is the part of prj.conf related to lvgl.
#LVGL
CONFIG_MAIN_STACK_SIZE=16384
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
CONFIG_HEAP_MEM_POOL_SIZE=16384
CONFIG_LVGL_MEM_POOL_HEAP_KERNEL=y

CONFIG_LVGL=y
CONFIG_LVGL_HOR_RES_MAX=300
CONFIG_LVGL_VER_RES_MAX=340
CONFIG_LVGL_VDB_SIZE=100
CONFIG_LVGL_BITS_PER_PIXEL=16
CONFIG_LVGL_DPI=130

#LVGL color
CONFIG_LVGL_COLOR_DEPTH_16=y

#LVGL display
CONFIG_DISPLAY=y
CONFIG_LVGL_DISPLAY_DEV_NAME="SmallOLED"
CONFIG_LVGL_DISP_DEF_REFR_PERIOD=30

#LVGL INDEV
CONFIG_LVGL_POINTER_KSCAN=y
CONFIG_LVGL_POINTER_KSCAN_DEV_NAME="OLEDTouch"
CONFIG_LVGL_INDEV_DEF_READ_PERIOD=50
CONFIG_LVGL_INDEV_DEF_DRAG_LIMIT=10
CONFIG_LVGL_INDEV_DEF_DRAG_THROW=20
CONFIG_LVGL_INDEV_DEF_LONG_PRESS_TIME=400
CONFIG_LVGL_INDEV_DEF_LONG_PRESS_REP_TIME=100

#LVGL buffer
CONFIG_LVGL_BUFFER_ALLOC_STATIC=y

#LVGL USE options
CONFIG_LVGL_USE_BTN=y
CONFIG_LVGL_USE_CONT=y
CONFIG_LVGL_USE_LABEL=y

Environment (please complete the following information):
Hardware:STM32L496AGI6P
Display: OLED display with 4 wire SPI interface.
Linux Ubuntu 20.04.3 LTS
Zephyr SDK,-0.13.0
Zephyr SHA1 ID:bb85a55
LVGL:7.6.1

kisvegabor, 2022-01-07:
Hi,

Painting the whole screen to a color is special (the simplest) case. From my experience it will be a display driver issue. (If it were an LVGL issue it'd be wrong for everyone.)

I suggest trying to draw smaller colorful rectangles on the screen with random size and position.

glin00, 2022-01-13:
I do find one hardware issue or limitation here. As the OLED display's datasheet points out, if any pixel data is to written to display, the data's x0 and y0 and x_width and y_width all need to be even numbers, can not be odd numbers. (note:x0, y0 are the coordinates of the upper left corner of pixels data)
For a test, I made two versions of firmware.  
(A)One that rejects the pixels data to be written to display when  x0 or y0 or x_width or y_width is odd.
[image]

(B)The other that accepts any pixels data to be written to display without checking  x0 or y0 or x_width or y_width is even or odd. 
[image]

The display result is significantly better using the one (A) that rejects pixel data with odd address or width.  
Question: Since it is the OLED display's limitation which the vendor said it can not be changed, how can I modify LVGL codes so that the pixels data written to display will have  x0 and y0 and x_width and y_width are even numbers?

kisvegabor, 2022-01-13:
It's great that found this limitation. Fortunately, it can be solved in LVGL. 
All you need to do is registering a [rounder_cb](https://docs.lvgl.io/master/porting/display.html#optional-fields).

glin00, 2022-01-17:
it is easy to register a rounder_cb function in lvgl 7.6.1 without zephyr. But using zephyr's lvgl  7.6.1, I am still finding a way to do it. Do you have any hints how to do it using  zephyr's lvgl  7.6.1?

glin00, 2022-01-18:
@kisvegabor , can you please give some advise on the issue I posted yesderday? I know what the code to be written in rounder_cb, but I just don't know where to register rounder_cb using zephyr's lvgl.

kisvegabor, 2022-01-18:
I think you can do it here: 
https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/gui/lvgl/lvgl_display.c#L9

LianKaFoHaLi, 2022-01-20:
> I think you can do it here: https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/gui/lvgl/lvgl_display.c#L9

I have meet the same problem in my project, Accoring to the description of rounder-cb, I define my_rounder_cb() function and  register it in lv_port_disp_init(),but the problem still exist.
[image]
I print the refresh area in disp_flush(),according to the log,my_rounder_cb() function works.

[image]
[image]
my lvgl version is 8.1.0,the oled driver ic is same as ”glin00“

LianKaFoHaLi, 2022-01-20:
> @kisvegabor , can you please give some advise on the issue I posted yesderday? I know what the code to be written in rounder_cb, but I just don't know where to register rounder_cb using zephyr's lvgl.

Hi,bro,have you solved the problem yet? I have meet the same problem.The oled driver ic is RM69310.

LianKaFoHaLi, 2022-01-21:
> > I think you can do it here: https://github.com/zephyrproject-rtos/zephyr/blob/main/lib/gui/lvgl/lvgl_display.c#L9
> 
> @kisvegabor I add a rule to print data that didn't meet requirement,according to the log,my_rounder_cb() does not seem to work in some cases [image] [image]

[image]

LianKaFoHaLi, 2022-01-21:
[image]
[image]
The start row pos and end row pos is also limited.

kisvegabor, 2022-01-24:
@LianKaFoHaLi 

How did you initialize the display driver? Please copy/paste including the display buffer size an initialization.

(Please send the code not an image to make it easy to copy into my project.)

glin00, 2022-01-25:
@kisvegabor 
I used the rounder_cb to make x0 and y0 and x_width and y_width even numbers.  Then this problem is solved. 
Thank you for your support.

closed 2022-01-25 by glin00

LianKaFoHaLi, 2022-01-25:
Thank you for your reply,the following is lv_port_disp.c file and description.  

To solve this problem ,I define another full  screen size buffer in oled driver,when refresh area don't meet the requirement,I called 'oled_write_buf()' function that write data into another screen size buffer and complement data to fit the requirement,then use spi dma send data to screen.if refresh area fit the requirement,called 'oled_write_buf_refresh()' function that write data into another screen size buffer,meanwhile use spi dma send data to screen. 
because of 'my_rounder_cb' function added,above  90% refresh events are fit the requirement. But it's cost too many ram resources to define another full screen size buffer.
I still want to do some measure that make the lvgl refresh area fit the oled requirement  in 'disp_flush()' function.
`/**
 * @file lv_port_disp_templ.c
 *
 */

 /*Copy this file as "lv_port_disp.c" and set this value to "1" to enable content*/
#if 1

/*********************
 *      INCLUDES
 *********************/
#include "lv_port_disp.h"
#include "lvgl.h"
// #include "userdef.h"
/*********************
 *      DEFINES
 *********************/
#define MY_DISP_HOR_RES 126
#define MY_DISP_VER_RES 294  

/**********************
 *      TYPEDEFS
 **********************/

/**********************
 *  STATIC PROTOTYPES
 **********************/
static void disp_init(void);

static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p);
//static void gpu_fill(lv_disp_drv_t * disp_drv, lv_color_t * dest_buf, lv_coord_t dest_width,
//        const lv_area_t * fill_area, lv_color_t color);

/**********************
 *  STATIC VARIABLES
 **********************/

/**********************
 *      MACROS
 **********************/

/**********************
 *   GLOBAL FUNCTIONS
 **********************/

void lv_port_disp_init(void)
{
    /*-------------------------
     * Initialize your display
     * -----------------------*/
    disp_init();

    /*-----------------------------
     * Create a buffer for drawing
     *----------------------------*/

    /**
     * LVGL requires a buffer where it internally draws the widgets.
     * Later this buffer will passed to your display driver's `flush_cb` to copy its content to your display.
     * The buffer has to be greater than 1 display row
     *
     * There are 3 buffering configurations:
     * 1. Create ONE buffer:
     *      LVGL will draw the display's content here and writes it to your display
     *
[... 112 lines cut ...]
//{
//    /*It's an example code which should be done by your GPU*/
//    int32_t x, y;
//    dest_buf += dest_width * fill_area->y1; /*Go to the first line*/
//
//    for(y = fill_area->y1; y <= fill_area->y2; y++) {
//        for(x = fill_area->x1; x <= fill_area->x2; x++) {
//            dest_buf[x] = color;
//        }
//        dest_buf+=dest_width;    /*Go to the next line*/
//    }
//}
void my_rounder_cb(struct _lv_disp_drv_t * disp_drv, lv_area_t * area){

    area->x1 = (area->x1/2)*2;
    area->y1 = (area->y1/2)*2;
    area->x2 = (area->x2/2)*2+1;
    area->y2 = (area->y2/2)*2+1;
}

#else /*Enable this file at the top*/

/*This dummy typedef exists purely to silence -Wpedantic.*/
typedef int keep_pedantic_happy;
#endif`

LianKaFoHaLi, 2022-01-25:
> @kisvegabor I used the rounder_cb to make x0 and y0 and x_width and y_width even numbers. Then this problem is solved. Thank you for your support.
Hi, I also meet the same problem,but in some case the refresh area still don't fit the requirement.