lvgl/lvgl #2975: Horizontal subpixel rendering font is not displaying properly

sam0910, 2022-01-07:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Horizontal subpixel rendering font is not displaying properly.
This happens only when I checked horizontal subpixel rendering option.
If I add a few labels, only some of them displays strange or disappear.

please note the screen shot. 

I am using,
* Micropython port
* Check the "Horizontal subpixel rendering" on font convertor.
* 4 bpp, size 40px
* LV_USE_FONT_SUBPX 1
* Font is Noto sans - extra bold
* Display drv is gc9a01

### To Reproduce
```c
disp = gc9a01(
    ...
    width=240,
    height=240,
    power_on=0,
    half_duplex=False,
    asynchronous=True,
    # rot=LANDSCAPE,
    hybrid=True,
    colormode=COLOR_MODE_BGR,
)

fonts = lv.font_36n
label1 = lv.label(lv.scr_act())
label1.set_text("456")
label1.set_style_text_align(lv.ALIGN.CENTER, 0)
label1.align(lv.ALIGN.CENTER, -40, 0)
label1.set_style_text_font(fonts, 0)

label2 = lv.label(lv.scr_act())
label2.set_text("456")
label2.set_style_text_align(lv.ALIGN.CENTER, 0)
label2.align(lv.ALIGN.CENTER, 40, 0)
label2.set_style_text_font(fonts, 0)
```

[image]

### Screenshots or video

kisvegabor, 2022-01-07:
Hi,

Can you reproduce it in a C simulator on desktop?

sam0910, 2022-01-08:
I am not familiar with C, but I've tried install both linux sim and platformio.
But I could not even execute it.

kisvegabor, 2022-01-12:
Take look at this list of the simulators: https://docs.lvgl.io/master/get-started/pc-simulator.html#select-an-ide
You can find one that is easy to use on your OS.

closed 2022-02-04 by lvgl-bot

geert-KLA-BE, 2022-04-20:
I have stumped into the same issue. It has very weird behavior. The things I noticed after a week searching for the cause:
 - I can only see it  only on my target with optimisation turned on (> -O0 )
 - I can hide the issue if I compile lv_font.c with -O0
 - Sometimes only part of a label is shown, sometimes it isn't shown, sometimes it works ok after refresh
 - Changing some not related code can make it appear and disappear
 - Enabling it on a specific font makes it appear only on labels using that font

I am using an STM32F4 with the latest official ARM gcc compiler

kisvegabor, 2022-04-20:
Which LVGL version do you use?

geert-KLA-BE, 2022-04-20:
I am using release version 8.2.

Met vriendelijke groeten / Cordialement / Kind regards,

Geert Van De Velde
Software Engineer
Research and Development
Kingspan Light & Air BE Sint-Niklaas
Anthonis De Jonghestraat 50|Sint-Niklaas|9100|Belgium
E: ***@***.***
www.kingspanlightandair.be
This email and its attachments may be confidential and are intended solely for the use of the individual to whom it is addressed. Any views or opinions expressed are solely those of the author and do not necessarily represent those of Kingspan Limited. If you are not the intended recipient of this email and its attachments, you must take no action based upon them, nor must you copy or show them to anyone. Please contact the sender if you believe you have received this email in error..v1037468

sam0910, 2022-04-20:
That is exactly same as me. 
Some case with different position it shows, some is not.

kisvegabor, 2022-05-01:
I have an STM32F412-Discovery board. If you also have this board, it'd be great if could send a project to reproduce the issue. If not, please send a `c` file and the converted font. Having these files I hope I can reproduce the issue.

kctan0805, 2022-05-27:
I cannot show any text on my platform. I reference draw_letter_normal() found that draw_letter_subpx() lacks adding
blend_dsc.mask_res = LV_DRAW_MASK_RES_CHANGED;
before calling lv_draw_sw_blend(...);
after adding it I fixed my problem.
Should I send a PR?

kisvegabor, 2022-05-27:
Great! Yes, please send a PR.