lvgl/lvgl #2977: Add assert to check when condition is evaluated to be true

C47D, 2022-01-07:
### Introduce the problem
There are currently multiple asserts, but none to check if a condition is evaluated to be true

```
/*Enable asserts if an operation is failed or an invalid data is found.
 *If LV_USE_LOG is enabled an error message will be printed on failure*/
#define LV_USE_ASSERT_NULL          1   /*Check if the parameter is NULL. (Very fast, recommended)*/
#define LV_USE_ASSERT_MALLOC        1   /*Checks is the memory is successfully allocated or no. (Very fast, recommended)*/
#define LV_USE_ASSERT_STYLE         0   /*Check if the styles are properly initialized. (Very fast, recommended)*/
#define LV_USE_ASSERT_MEM_INTEGRITY 0   /*Check the integrity of `lv_mem` after critical operations. (Slow)*/
#define LV_USE_ASSERT_OBJ           0   /*Check the object's type and existence (e.g. not deleted). (Slow)*/

/*Add a custom handler when assert happens e.g. to restart the MCU*/
#define LV_ASSERT_HANDLER_INCLUDE <stdint.h>
#define LV_ASSERT_HANDLER while(1);   /*Halt by default*/
```

### Examples and cases

`lv_bar_set_range` doesn't validate if `max` is greater than `min`

```c
void lv_bar_set_range(lv_obj_t * obj, int32_t min, int32_t max)
{
    LV_ASSERT_OBJ(obj, MY_CLASS);

    lv_bar_t * bar = (lv_bar_t *)obj;

    if(bar->min_value == min && bar->max_value == max) return;

    bar->max_value = max;
    bar->min_value = min;

    if(lv_bar_get_mode(obj) != LV_BAR_MODE_RANGE)
        bar->start_value = min;

    if(bar->cur_value > max) {
        bar->cur_value = max;
        lv_bar_set_value(obj, bar->cur_value, LV_ANIM_OFF);
    }
    if(bar->cur_value < min) {
        bar->cur_value = min;
        lv_bar_set_value(obj, bar->cur_value, LV_ANIM_OFF);
    }
    lv_obj_invalidate(obj);
}
```

A test in the simulator, min is greater than max, and the bar is empty even when we set the value, without any clue why is that.
[image]

### Suggested solution

Add `LV_ASSERT`-like symbol, if condition evaluates to false then execute `LV_ASSERT_HANDLER`. [SerenityOs](https://github.com/SerenityOS/serenity) has a assert named `VERIFY`, which I think, it's a better name when verifying parameters/preconditions.

```c
void lv_bar_set_range(lv_obj_t * obj, int32_t min, int32_t max)
{
    LV_ASSERT_OBJ(obj, MY_CLASS);
    LV_ASSERT(max >= min);

    lv_bar_t * bar = (lv_bar_t *)obj;

    if(bar->min_value == min && bar->max_value == max) return;

    bar->max_value = max;
    bar->min_value = min;

    if(lv_bar_get_mode(obj) != LV_BAR_MODE_RANGE)
        bar->start_value = min;

    if(bar->cur_value > max) {
        bar->cur_value = max;
        lv_bar_set_value(obj, bar->cur_value, LV_ANIM_OFF);
    }
    if(bar->cur_value < min) {
        bar->cur_value = min;
        lv_bar_set_value(obj, bar->cur_value, LV_ANIM_OFF);
    }
    lv_obj_invalidate(obj);
}
```

kisvegabor, 2022-01-08:
I see your point and I really like the name `LV_ASSERT_VERIFY`. However, I don't think it's be good practice to assert (i.e. halt the UI) on these "soft issues". The UI can happily run with min > max value, just the bar won't do anything. 

So far asserts are used for the cases when the UI is crashed or will crash if it continues to run. For these parameter issues we added `LV_LOG_WARN`s so far. 

This approach still makes sens to me, but of course we can reconsider it. 

I was thinking about displaying last few warnings as a label on the screen. We can even enable this feature by default. If there are no warnings nothing will be displayed and the last messages can disappear automatically after a few seconds.

It'd be great to display `LV_LOG_ERROR`s this way too but when LVGL is about to crash we can't set the text of the labels. Maybe we could have a fallback to write plain text on the screen using the flush callback directly.

C47D, 2022-01-09:
Thanks for the reply, maybe we could disable the assertion handler being executed by the verify check by default and let the user enable it when they need to.

I didn't thought of it as a soft issue, I saw it more like the user not using the API as documented. 

In the particular case of the example, the docs doesn't mention that the max value has to be greater than min, otherwise we don't get the expected result.

kisvegabor, 2022-01-10:
> Thanks for the reply, maybe we could disable the assertion handler being executed by the verify check by default and let the user enable it when they need to.

Can we find some other placed where we code use `LV_ASSERT_VERIFY`? Not in slider only but in LVGL in general.
I didn't get much idea immdiatelly.

> In the particular case of the example, the docs doesn't mention that the max value has to be greater than min, otherwise we don't get the expected result.

I thought it's obvious, but really could be some tricks (e.g. if max < min the slider is inverted somehow)

> Here's an interesting talk about the topic, maybe we can get some ideas from it https://youtu.be/1QhtXRMp3Hg

I started to watch, but it seems he get to the point a little bit slowly :slightly_smiling_face:

C47D, 2022-01-10:
> I started to watch, but it seems he get to the point a little bit slowly 🙂

They were 2 talks of the same length, this is only one 😄

C47D, 2022-01-11:
Another example of checking preconditions with asserts, here we would validate the bar mode.

https://github.com/lvgl/lvgl/pull/2845#issuecomment-1007094159

kisvegabor, 2022-01-11:
> We should consider when _lv_txt_ins` is receiving a NULL pointer, maybe it's a failure indicator and it currently only returns silently.

> Another example of checking preconditions with asserts, here we would validate the bar mode.

Both are good point! So in general we should increase the "assertion coverage" of LVGL, right? If you agree, let's open a PR where we try to add the missing asserts (including `LV_ASSERT_VERIFY`).

C47D, 2022-01-12:
If you agree I'm locating string handling and memory allocation functions and evaluating how to proceed.

e.g. in `lv_table_set_cell_value`:

- In the `LV_USE_ARABIC_PERSIAN_CHARS` branch we return early if `table->cell_data[cell] == NULL`, but we don't do the same in the else branch. Does `table->cell_data[cell] == NULL` means the allocation failed? If so, should we halt?
- The `strcpy` parameters are not checked, e.g. `dst` must be large enough to contain `src`. According to [this](https://en.cppreference.com/w/c/string/byte/strcpy) information UB will happen if that requirement is not met.

```c
void lv_table_set_cell_value(lv_obj_t * obj, uint16_t row, uint16_t col, const char * txt)
{
    LV_ASSERT_OBJ(obj, MY_CLASS);
    LV_ASSERT_NULL(txt);

    lv_table_t * table = (lv_table_t *)obj;

    /*Auto expand*/
    if(col >= table->col_cnt) lv_table_set_col_cnt(obj, col + 1);
    if(row >= table->row_cnt) lv_table_set_row_cnt(obj, row + 1);

    uint32_t cell = row * table->col_cnt + col;
    lv_table_cell_ctrl_t ctrl = 0;

    /*Save the control byte*/
    if(table->cell_data[cell]) ctrl = table->cell_data[cell][0];

#if LV_USE_ARABIC_PERSIAN_CHARS
    /*Get the size of the Arabic text and process it*/
    size_t len_ap = _lv_txt_ap_calc_bytes_cnt(txt);
    table->cell_data[cell] = lv_mem_realloc(table->cell_data[cell], len_ap + 1);
    LV_ASSERT_MALLOC(table->cell_data[cell]);
    if(table->cell_data[cell] == NULL) return;

    _lv_txt_ap_proc(txt, &table->cell_data[cell][1]);
#else
    table->cell_data[cell] = lv_mem_realloc(table->cell_data[cell], strlen(txt) + 2); /*+1: trailing '\0; +1: format byte*/
    LV_ASSERT_MALLOC(table->cell_data[cell]);

    strcpy(table->cell_data[cell] + 1, txt);  /*+1 to skip the format byte*/
#endif

    table->cell_data[cell][0] = ctrl;
    refr_size(obj, row);

    lv_obj_invalidate(obj);
}
```

kisvegabor, 2022-01-17:
> In the LV_USE_ARABIC_PERSIAN_CHARS branch we return early if table->cell_data[cell] == NULL, but we don't do the same in the else branch. Does

For consistency we should early return in both cases. However this leads very far to the topic of error handling. There is a discussion about it #2194. IMO introducing a strict error handling with return values in every function wouldn't be a good direction. It's because if the system is out of memory or an object gets corrupted it's game over anyway. At least in most of the cases. And I find it bad to sacrifice simplicity for the sake of a few special cases. 

Therefore, my view on error handling is that in case of LVGL is this:
- Use asserts and logs during development to verify the UI is working well. I.e. no memory leak, no corrupted objects, no infinite cycles, etc.
- Once the UI works well, verify the input data. E.g. do not create list with 10000 elements, or do not use not terminated strings, etc. 

With this way of thinking LVGL can be "error resistant" meaning do nothing on invalid data (except adding logs or assert if enabled)

C47D, 2022-01-25:
Sorry @kisvegabor just read your reply and also the replies on #2194, let me know how you want to proceed.

> IMO introducing a strict error handling with return values in every function wouldn't be a good direction.

Agree on adding return values in every function is not a good direction.

> Use asserts and logs during development to verify the UI is working well. I.e. no memory leak, no corrupted objects, no infinite cycles, etc.

> Once the UI works well, verify the input data. E.g. do not create list with 10000 elements, or do not use not terminated strings, etc.

This is more or less what I learned from the video I sent, use a level of assertions at development, where you have the ability to debug any assertion failure, once you're running the "production" code you can enable only some level of assertions.

kisvegabor, 2022-01-25:
> This is more or less what I learned from the video I sent, use a level of assertions at development, where you have the ability to debug any assertion failure, once you're running the "production" code you can enable only some level of assertions.

Its great that we are on the same page. :slightly_smiling_face: 

Let's discuss it in #2194 and hopefully we can close that issue too.

C47D, 2022-01-25:
> Let's discuss it in #2194 and hopefully we can close that issue too.

Sure, do you think this issue can be closed?

closed 2022-01-25 by C47D