lvgl/lvgl #2978: lv_obj_del participates in memory leaks.

Satnet, 2022-01-08:
### Introduce the problem
When deleting objects, for example on screen transition. There will be a memory leak if you have used lv_style_init, this is because LVGL does not clean/delete those style init's. Eventually you will have a crash. I learned this the hard way. Now I manually use lv_style_reset to solve this issue. However, I believe that lv_obj_del should take care of this as well. Call it a bug or a feature request. Not sure what to make of it.

### Examples and cases
When using lv_scr_load_anim with automatic object/screen deletion, memory leaks are introduced as the style elements keep adding up.

### Suggested solution
Add a parent/child loop to clear all styles associated with all the objects connected to the main object you want to delete.

kisvegabor, 2022-01-08:
Actually it's a feature. Sorry, that it caused you hard times. :slightly_frowning_face: 

Styles are not related to any objects. The styles can be used on any object, not only on the objects of a given screen. E.g. the `red_btn` can be used on all of the buttons. 

This issue can be solved by organizing your code slightly differently by having a `my_style_init()` function which is called only once.

If want to init/deinit the styles on screen creation/deletion to save memory or so, it's considered a special case and it requires special attention. Anyway, in the `LV_EVENT_DELETED` event of the screen you can make any clean up you need.

If you are familiar with HTML/CSS, LVGL follows the same way of thinking. You have the GUI in HTML files and set styles (classes) from the CSS file. If you remove a `div` with `red_btn` class, the class srill remains in the CSS file. 

BTW, if you enable `LV_USE_ASSERT_STYLE` and `LV_USE_LOG` LVGL will warn you if you re-initialize an already initialized style.

Satnet, 2022-01-08:
It's okay, I solved it in a few hours. You are right, it's better to init the styles and let them just hang round, then apply them where needed. I will try this approach. Thank you Kisvegabor!

kisvegabor, 2022-01-08:
You are welcome! :slightly_smiling_face: 

A "Best practices" section in the [docs](https://docs.lvgl.io/master/overview/style.html) seems helpful for cases like this. If you have time could you summarize your experience in a PR? 

As the original problem is solved, I close this issue.

closed 2022-01-08 by kisvegabor