lvgl/lvgl #2982: Lack description for which source files should be added to a project during porting

GorgonMeducer, 2022-01-10:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I read the documents and found that:
- In [poring](https://docs.lvgl.io/master/porting/index.html) there is no any description about how to add source code to a project. 
- I only found the following content which is insufficient for an arbitrary IDE, i.e. MDK:

> The graphics library itself is the lvgl directory which should be copied into your project.

Obviously, IDEs like MDK and IAR do not rely on cmake or makefile, hence the 'lv_xxx.mk' files inside the src (and its sub-folders) do not help. 

I don't think I am the first person who encountered this issue. 

Could you please update your documents and make it more clear for arbitrary IDE users?
Please NOTE that copying the **_lvgl_** folder to a project doesn't add the source files to a project AUTOMATICALLY. Instead, users have to add them manually (at least in MDK).

### To Reproduce

### Expected behavior

### Screenshots or video

embeddedt, 2022-01-10:
Yes, all the source files in the `src` folder should be added (and if you want the examples/demos, add those folders as well).

GorgonMeducer, 2022-01-10:
Thank you. @embeddedt .
Will there be a document update?

kisvegabor, 2022-01-10:
I've just updated the docs: https://github.com/lvgl/lvgl/commit/6ce73486d319bfdb1c379d090036a7eeaabf5b43

GorgonMeducer, 2022-01-10:
Wow! what a quick action.

closed 2022-01-10 by GorgonMeducer

kisvegabor, 2022-01-10:
Trying our best :slightly_smiling_face: