lvgl/lvgl #2983: Duplicated content in 'optional fields' section

GorgonMeducer, 2022-01-10:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
There is duplicated content in the ['optional fields' section](https://docs.lvgl.io/master/porting/display.html#optional-fields)

[image]

Here the 'user_data' line is duplicated. 

### To Reproduce

### Expected behavior

### Screenshots or video

GorgonMeducer, 2022-01-10:
https://github.com/lvgl/lvgl/pull/2984

kisvegabor, 2022-01-10:
Fixed by #2984.

closed 2022-01-10 by kisvegabor