lvgl/lvgl #2986: do you have plan to add cmsis-pack for lvgl

GorgonMeducer, 2022-01-10:
### Introduce the problem
It appears that currently there is no official cmsis-pack for lvgl. Do you have a plan to create one?
I can contribute one if you don't mind.

### Examples and cases

### Suggested solution

kisvegabor, 2022-01-11:
Hi,

I don't know `cmsis-pack` but from the docs it seems interesting. I've read the basics from [here](https://www.keil.com/pack/doc/CMSIS/Pack/html/index.html) but still have a two main questions:
- What projects/IDEs/SDKs are using `cmsis-pack`?
- How do we need to maintain the pack?  

> I can contribute one if you don't mind.

Thanks for offering your help!

GorgonMeducer, 2022-01-11:
For now, MDK and IAR are using cmsis-pack.
The way to maintain the pack is that:
a. every time you want to release an lvgl version, you might want to update the pack
b. you might put examples, code template and new platform support to cmsis-pack

Now cmsis-pack is donated to Linaro:

https://www.open-cmsis-pack.org/#:~:text=Overview,with%20Arm%2C%20NXP%20and%20ST.

>The Open-CMSIS-Pack project will deliver the infrastructure to integrate and manage software components and improve code reuse across embedded and IoT projects. The project is currently hosted and managed as an incubation project by Linaro in partnership with Arm, NXP and ST.

kisvegabor, 2022-01-11:
> every time you want to release an lvgl version, you might want to update the pack

Does it mean simple bumping the version number in the descriptor file or publishing by hand?

GorgonMeducer, 2022-01-11:
For most of the time, YES. Unless you have some major update that changes all file names (this happened before in LVGL), in that case, you have to update the script which we used to generate the cmsis-pack, it takes you 10mins.

kisvegabor, 2022-01-11:
Ah, great! If it's this simple I think it'd be great to support `cmsis-pack`.

GorgonMeducer, 2022-01-11:
Sure, I will do that. 
Here is an example, you can have a look. 

For a cmsis-pack, here are immediately some questions for you:

Please fill blanks in the following code (I put mine as example):

```
  <vendor>GorgonMeducer</vendor>
  <name>perf_counter</name>
  <description>A dedicated performance counter for the Cortex-M Systick. It shares the SysTick with users' original SysTick function without interfering with it. This library will bring new functionalities, such as performance counter, delay_us and clock() service defined in time.h</description>
  <url>https://raw.githubusercontent.com/GorgonMeducer/perf_counter/CMSIS-Pack/cmsis-pack/</url>
  <supportContact>https://github.com/GorgonMeducer/perf_counter/issues</supportContact>
  <license>lib/LICENSE</license>
```

GorgonMeducer, 2022-01-11:
One more question, I have read the 'how-to contribute', it says:
> Set the base branch. It means where you want to merge your update. In the lvgl repo fixes go to master, new features to dev branch.

But there is no such branch called 'dev':

[image]

Which branch should I use? 'master' ??

GorgonMeducer, 2022-01-11:
First cmsis-pack is ready to review:

[image]

[image]

[image]

GorgonMeducer, 2022-01-11:
https://github.com/GorgonMeducer/lvgl/tree/add-cmsis-pack

GorgonMeducer, 2022-01-11:
The support for extra themes, widgets and layouts has been added:

[image]

GorgonMeducer, 2022-01-11:
lv_conf_cmsis.h was a copy of lv_conf.h, but to modularize the lvgl in the configuration view, I have to modify the original content inside lv_conf_cmsis.h.

Let me show you an example:

[image]

As you can see, libs PNG is a separated module. You might immediately notice that, ideally, the macro **LV_USE_PNG** should be defined as '1' as long as the libs PNG is selected.  The problem of the original lv_conf.h is that there is not way to modify the macro automatically:

[image]

Hence I modify it inside lv_conf_cmsis.h as you can guess:

'''
/*PNG decoder library*/
#ifndef LV_USE_PNG
    #define LV_USE_PNG      0
#endif
'''

and inside the script:

'''
            <component Cgroup="lvgl" Csub="Libs PNG"  condition="LVGL-Essential">
              <description>Add PNG support</description>
              <files>
                
                <file category="sourceC"            name="src/extra/libs/png/lodepng.c" />
                <file category="sourceC"            name="src/extra/libs/png/lv_png.c" />
              </files>
              
              <RTE_Components_h>
              
/*! \brief enable PNG support */
#define LV_USE_PNG          1
              </RTE_Components_h>
              
            </component>
'''

Once people select the Lib PNG in the RTE window, the following code will be inserted to RTE_Components.h which is included by lv_conf_cmsis.h
'''
/*! \brief enable PNG support */
#define LV_USE_PNG          1
'''

[image]

GorgonMeducer, 2022-01-11:
Further more, MDK provides a simple syntax that turns any macro definitions into GUI-based configuration wizard. For example:

```
//-------- <<< Use Configuration Wizard in Context Menu >>> -----------------
//
// <h>Arm-2D General Configurations
// =======================
//
// <q>Enable Asynchronous Programmers' model support
// <i> Note that if you don't have any hardware accelerator, disable this feature can reduce code size and gain a small performance uplift.
// <i> This feature is enabled by default.
#ifndef __ARM_2D_HAS_ASYNC__
#   define __ARM_2D_HAS_ASYNC__                                     1
#endif

// <q>Enable anti-alias support for all rotation operations.
// <i> Note that enabling this feature suffers a non-negligible performance drop.
// <i> This feature is disabled by default.
#ifndef __ARM_2D_HAS_INTERPOLATION_ROTATION__
#   define __ARM_2D_HAS_INTERPOLATION_ROTATION__                    0
#endif

// <q>Enable support for accessing individual colour channels
// <i> Note that enabling this feature will add the support for a special colour type: ARM_2D_CHANNEL_8in32
// <i> This feature is disabled by default to save code size
#ifndef __ARM_2D_CFG_SUPPORT_COLOUR_CHANNEL_ACCESS__
#   define __ARM_2D_CFG_SUPPORT_COLOUR_CHANNEL_ACCESS__             0
#endif
// </h>

// <<< end of configuration section >>>
```

Will be shown as:
[image]

Details about this syntax can be found [here](https://arm-software.github.io/CMSIS_5/Pack/html/configWizard.html)

GorgonMeducer, 2022-01-11:
I believe there are other wizard-based solutions, such as the one used in menueconfig, the one used in the RT-Thread package and the one used in MDK.

An ultimate solution is:
- Write lv_conf_template.h in a machine-readable manner
- Use a script (.py?) generate lv_conf_cmsis.h and other required wizard description files.

What do you think?

GorgonMeducer, 2022-01-11:
CMSIS 0.7.0

Almost everything inside lvgl repository:

[image]

https://github.com/GorgonMeducer/lvgl/tree/add-cmsis-pack/cmsis-pack

kisvegabor, 2022-01-11:
> Hence I modify it inside lv_conf_cmsis.h as you can guess:
> /*PNG decoder library*/
#ifndef LV_USE_PNG
    #define LV_USE_PNG      0
#endif

In this case it's be better to simple remove `LV_USE_PNG` (and the other similar ones) as [lv_conf_internal.h](https://github.com/lvgl/lvgl/blob/master/src/lv_conf_internal.h#L1858-L1864) takes care of defining any undefined configs. 

It seems simple to write a script which copies `lv_conf_template.h` as `lv_conf_cmsis.h` and removes some config. This way we could easily and safely update `lv_conf_cmsis.h` later.

> Write lv_conf_template.h in a machine-readable manner
Use a script (.py?) generate lv_conf_cmsis.h and other required wizard description files.

Sounds good! For now let's find a workaround, but in the long term it'd be great. 

 > Make each macro switch override-able (so people can define a macro in a project configuration and it overrides the default value in lv_conf.h.

IMO It'd make `lv_conf.h` practically unreadable for the users :slightly_frowning_face: 

---

I'd have a few comments but I think it's cleaner to discuss them in a PR. So please create one. :slightly_smiling_face:

GorgonMeducer, 2022-01-11:
No problem, let me use the workaround methods and create a PR, i.e. 
> ...remove LV_USE_PNG (and the other similar ones) as lv_conf_internal.h takes care of defining any undefined configs...

GorgonMeducer, 2022-01-11:
https://github.com/lvgl/lvgl/pull/2993

closed 2022-01-11 by GorgonMeducer