lvgl/lvgl #2990: STM32 HW acceleration

ahmetalperenbulut, 2022-01-11:
Hi All, 

I'm using STM32H7 with hw acceleration and I've seen some problem in large changes. 
@kisvegabor  have you seen any problem using hw acceleration ?  
I have fixed hw acceleration in my side but i don't know it occurs only my side or general problem.

kisvegabor, 2022-01-11:
Hi,

There were a large rework on the draw architecture so it might be some regression. 

How did you fix it on your side?

ahmetalperenbulut, 2022-01-11:
I've used DMA2D with HAL functions and it fixed.  
I can create a pull request if you want

kisvegabor, 2022-01-11:
We moved to the register-level version to make LVGL work without HAL too.

I took a look and `lv_draw_stm32_dma2d_blend_fill` and `lv_draw_stm32_dma2d_blend_map` doesn't wait for he GPU to finish.

Can you try if adding `while(DMA2D->CR & DMA2D_CR_START_Msk);` at the beginning of these functions helps?

ahmetalperenbulut, 2022-01-12:
I'm working in v8.0.1 branch but I checked out master branch. there is no problem in master branch

closed 2022-01-12 by ahmetalperenbulut

kisvegabor, 2022-01-12:
Great! Mean while I fond that there is waiting in `lv_draw_sw_blend()`.