lvgl/lvgl #2995: Gesture events are not triggered when the object is scrolled

FASTSHIFT, 2022-01-12:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce
```c
static void event_cb(lv_event_t* e)
{
  lv_dir_t dir = lv_indev_get_gesture_dir(lv_indev_get_act());
  LV_LOG_USER("dir: %d", dir);
}

static void gesture_test()
{
  lv_obj_t* obj = lv_obj_create(lv_scr_act());
  lv_obj_clear_flag(obj, LV_OBJ_FLAG_GESTURE_BUBBLE);
  lv_obj_set_size(obj, 400, 400);

  lv_obj_add_event_cb(obj, event_cb, LV_EVENT_GESTURE, NULL);

  lv_obj_t* obj2 = lv_obj_create(obj);
  lv_obj_set_size(obj2, 400, 600);
}
```

### Expected behavior
Gesture events can still trigger when the object is scrollable.

### Screenshots or video

kisvegabor, 2022-01-12:
Gestures are intentionally not triggered while scrolling.

What could be use case for triggering gestures while scrolling?

FASTSHIFT, 2022-01-12:
I haven't thought of a suitable application scenario yet.