lvgl/lvgl #2997: The default background of the roller is filled with white, not transparent

alec-2018, 2022-01-12:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I am using the screen of the spi interface, I need to set LV_COLOR_DEPTH = 16, LV_COLOR_16_SWAP = 1, but when using the example "lv_example_roller_1" on my device, it is inconsistent with running on the simulator, what is the reason for it?
Use 8.1.1dev
### To Reproduce

```c

LV_COLOR_DEPTH = 16
LV_COLOR_16_SWAP = 1

static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_VALUE_CHANGED) {
        char buf[32];
        lv_roller_get_selected_str(obj, buf, sizeof(buf));
        LV_LOG_USER("Selected month: %s\n", buf);
    }
}

/**
 * An infinite roller with the name of the months
 */
void lv_example_roller_1(void)
{
    lv_obj_t *roller1 = lv_roller_create(lv_scr_act());
    lv_roller_set_options(roller1,
                        "January\n"
                        "February\n"
                        "March\n"
                        "April\n"
                        "May\n"
                        "June\n"
                        "July\n"
                        "August\n"
                        "September\n"
                        "October\n"
                        "November\n"
                        "December",
                        LV_ROLLER_MODE_INFINITE);

    lv_roller_set_visible_row_count(roller1, 4);
    lv_obj_center(roller1);
    lv_obj_add_event_cb(roller1, event_handler, LV_EVENT_ALL, NULL);
	printf("--------<%d.\n", lv_obj_get_style_bg_opa(roller1, 0));
	//lv_obj_set_style_bg_opa(roller1, LV_OPA_0, 0);
}
```
### Expected behavior

[image]

### Screenshots or video

[image]

kisvegabor, 2022-01-12:
Finally what is inconsistent?
Do you set `lv_obj_set_style_bg_opa(roller1, LV_OPA_0, 0);` but roller is still white?

alec-2018, 2022-01-12:
> 最后是什么不一致？ 你设置了，`lv_obj_set_style_bg_opa(roller1, LV_OPA_0, 0);`但滚筒还​​是白色的？

I set lv_obj_set_style_bg_opa(roller1, LV_OPA_0, 0); to achieve the expected effect, which means that every gadget needs to be set, the same code in the simulator and the result of the mcu are inconsistent, the screen above the mcu is the spi interface, The mcu is white, but the emulator is transparent

alec-2018, 2022-01-12:
```c
static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_VALUE_CHANGED) {
        char buf[32];
        lv_roller_get_selected_str(obj, buf, sizeof(buf));
        LV_LOG_USER("Selected month: %s\n", buf);
    }
}

/**
 * An infinite roller with the name of the months
 */
void lv_example_roller_1(void)
{
    lv_obj_t *roller1 = lv_roller_create(lv_scr_act());
    lv_roller_set_options(roller1,
                        "January\n"
                        "February\n"
                        "March\n"
                        "April\n"
                        "May\n"
                        "June\n"
                        "July\n"
                        "August\n"
                        "September\n"
                        "October\n"
                        "November\n"
                        "December",
                        LV_ROLLER_MODE_INFINITE);

    lv_roller_set_visible_row_count(roller1, 4);
    lv_obj_center(roller1);
    lv_obj_add_event_cb(roller1, event_handler, LV_EVENT_ALL, NULL);
}
```
The result of running this code on the simulator is: the background of the scroll wheel is transparent, and it is white when running on the mcu

alec-2018, 2022-01-12:
I think the problem is caused by the software renderer, because in the version that has not modified the software renderer (updated master on December 14th) I tested without problems

kisvegabor, 2022-01-12:
I can hardly follow. Just to be sure: you use tha same (latest master) in both the simulator and the device but got different result?

alec-2018, 2022-01-12:
> 我几乎跟不上。只是为了确定：您在模拟器和设备中都使用相同的（最新主）但得到不同的结果？

yes

alec-2018, 2022-01-13:
The default color of the latest master widget does not change according to the theme

alec-2018, 2022-01-13:
@kisvegabor #2872 solved this problem

X-Ryl669, 2022-01-13:
Should be fixed by #3002. Please confirm too.

alec-2018, 2022-01-13:
> 应该由#3002修复。也请确认。

Sorry for being late, this modification works fine

closed 2022-01-13 by kisvegabor

MauricioEmilio, 2024-02-10:
Hello, I'm using the last lvgl UI graphic designer (just upgraded three days ago), I'm using a roller with default colors aspects and Normal mode, works fine, but when I set a infinite behavior doesn't show the options, seems like typos let the same background color of the highlight selection, however the roller give the changed values as expected.

kisvegabor, 2024-02-11:
Please open a new issue and send a code snippet to reproduce the issue.

PR #3002 (2022-01-13): Fix missed bg_color renaming in the draw function
### Description of the feature or fix
Should fix the issue I've introduced in Gradient Dithering PR. 
Should fix #2997 too 

On PC simulator, bg_color and grad.stops[0].color points to the same value so it worked. On ESP32 it doesn't so I missed this line in my previous PR. Should be fixed now.

### Checkpoints
- [x] Follow the [styling guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md)
- [X] Run `code-format.py` from the `scripts` folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [X] Updat [... cut]