lvgl/lvgl #3009: lv_disp_flush_is_last() does not correctly indicate the last flush

FASTSHIFT, 2022-01-15:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I found `lv_disp_flush_is_last()` to be unreliable, being called **twice** in certain cases, which could lead to repeated buffer submissions to the display, resulting in erroneous behavior.
I tried inserting flag bits in lvgl's code to catch the unusual behavior. I found that if in the object's draw event (lv_refr_area is being called...), moving the coordinates of another non-adjacent object (making the calculated layout change again), this abnormal behavior will occur.
Here is my test code:

### To Reproduce
```c
static lv_obj_t* obj1;
static lv_obj_t* obj2;

static void draw_event_cb(lv_event_t* e)
{
    lv_obj_t* obj = lv_event_get_current_target(e);
    lv_obj_set_x(obj2, lv_obj_get_style_x(obj, 0));
}

static void timer_cb(lv_timer_t* timer)
{
    lv_obj_t* obj = timer->user_data;
    lv_coord_t x = lv_obj_get_x(obj);

    x += 10;

    if(x > 200)
    {
        x = 0;
    }

    lv_obj_set_x(obj, x);
}

static void move_test()
{
    obj1 = lv_obj_create(lv_scr_act());
    lv_obj_add_event_cb(obj1, draw_event_cb, LV_EVENT_DRAW_MAIN, NULL);

    obj2 = lv_obj_create(lv_scr_act());
    lv_obj_set_y(obj2, 200);

    lv_timer_t* timer = lv_timer_create(timer_cb, 20, obj1);
}
```

**lv_refr.c: lv_refr_areas()**
[image]

**lv_refr.c: _lv_inv_area()**
[image]

**sdl.c: sdl_display_flush()**
[image]

### Expected behavior
`lv_disp_flush_is_last()` can correctly indicate the last flush.

### Screenshots or video
[image]

kisvegabor, 2022-01-15:
It's really not stated clearly in the docs but you shouldn't move the object (in general shouldn't set any properties)  in a draw event. It can really mess up the "invalidated area" calculations.

Why do you need to move the object in a draw event?

FASTSHIFT, 2022-01-17:
@kisvegabor 
This bug was discovered by accident, since it is an incorrect usage, can the documentation be updated? Insert flags and logs in `lv_refr.c` to remind users.

kisvegabor, 2022-01-19:
What do you think about mentioning it [here](https://docs.lvgl.io/master/overview/event.html#drawing-events)?

FASTSHIFT, 2022-01-19:
> What do you think about mentioning it [here](https://docs.lvgl.io/master/overview/event.html#drawing-events)?

I agree with this approach.

kisvegabor, 2022-01-19:
Added.

closed 2022-01-19 by kisvegabor

FASTSHIFT, 2022-02-11:
@kisvegabor 
Can I submit pr this code that detects abnormal behavior?  This is an easily overlooked problem :)

kisvegabor, 2022-02-18:
> Can I submit pr this code that detects abnormal behavior?

Yes, please. I have no good idea how to detect it so I'm curious to see your idea :slightly_smiling_face:

FASTSHIFT, 2022-02-18:
@kisvegabor 
As shown in the figure above, I added the `cache_inv_area` flag in `lv_refr_areas()`, and checked the `cache_inv_area` flag in` _lv_inv_area()`. If the user is using it normally, the `cache_inv_area` in `_lv_inv_area()` is in the false state; if the user modifies the attributes of obj in the draw event, `_lv_inv_area()` will be called, and the `cache_inv_area` is in the true state.

kisvegabor, 2022-02-18:
> This way doesn't seem elegant, any better suggestion? :)

Actually I think it's a great solution! Just 2 minor suggestions:
- Move `catch_inv_area` into `lv_disp_t` as 1-bit flag and call it something like `rendering_in_progress`
- In `_lv_inv_area()` check this bit at the very beginning and besides logging an error  also `return` to not mess up the current invalidated areas which are being rendered.

Habbie, 2024-02-23:
(commenting on a closed ticket just to provide context for readers)

If you find your objects (sliders in my case) not rendering updates, and through some debugging process (in my case, bisecting), discover that #3107 has the commit that broke it, I ended up finding out I had not read what https://docs.lvgl.io/8.3/porting/os.html had to say about mutexes. Adding the mutex made my sliders update reliably after this PR.

commit 933d67fe (2022-01-19): docs(event): tell to not adjust widgets in draw events

fixes #3009