lvgl/lvgl #3010: The buffer will not automatically switch when double full-screen buffer + direct_mode

FASTSHIFT, 2022-01-15:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The buffer will not automatically switch when double full-screen buffer + direct_mode.

### To Reproduce
[image]

### Expected behavior
The buffer will automatically switch when double full-screen buffer+direct_mode.

### Screenshots or video
[image]

**Disable direct_mode**
[image]

FASTSHIFT, 2022-01-15:
The buffer switching logic in the `draw_buf_flush()` function may be incorrect, because `draw_buf->flushing_last` may be cleared in the user's `flush_cb` (calling `lv_disp_flush_ready()`)
[image]
[image]

FASTSHIFT, 2022-01-16:
Can  remove the line `disp_drv->draw_buf->flushing_last = 0;` in the `lv_disp_flush_ready()` function?

kisvegabor, 2022-01-18:
> The buffer switching logic in the draw_buf_flush() function may be incorrect, because draw_buf->flushing_last may be cleared in the user's flush_cb (calling lv_disp_flush_ready())

Finally, this the reason of the missed buffer switching?

> Can remove the line disp_drv->draw_buf->flushing_last = 0; in the lv_disp_flush_ready() function?

If so, I'd rather save the `flushing_last` in line 1004 and use the saved value in line 1015. What do you think?

FASTSHIFT, 2022-01-18:
> > The buffer switching logic in the draw_buf_flush() function may be incorrect, because draw_buf->flushing_last may be cleared in the user's flush_cb (calling lv_disp_flush_ready())
> 
> Finally, is this the reason of the missed buffer switching?
> 
> > Can remove the line disp_drv->draw_buf->flushing_last = 0; in the lv_disp_flush_ready() function?
> 
> If so, I'd rather save the `flushing_last` in line 1004 and use the saved value in line 1015. What do you think?

Yes, I tried to remove the line `disp_drv->draw_buf->flushing_last = 0l`, the buffer can be switched normally. 
I agree with the way to save the value of `flushing_last`.

FASTSHIFT, 2022-01-18:
I tested it and it seems to work fine.
[image]

kisvegabor, 2022-01-20:
Fixed, as discussed.

commit 731ef5a7 (2022-01-20): fix(refr): fix missed buffer switch in double full-screen buffer + direct_mode

fixes #3010
closed 2022-01-20 by kisvegabor