lvgl/lvgl #3011: lv_dropdown ghosting

bfjm, 2022-01-15:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Hello,All!
I have a lv_dropdown object, andI set the first tab of lv_tabview to the parent of lv_dropdown.
It's ghosting happend, when i clicked.CPU is ARM.

### To Reproduce

### Expected behavior

### Screenshots or video

like this.
[image]

kisvegabor, 2022-01-15:
Hi, 

Please send a video and a code snippet to reproduce.

Please also test it  a simulator.

bfjm, 2022-01-18:
[image]

like this!

kisvegabor, 2022-01-21:
> Please send a video and a code snippet to reproduce.

bfjm, 2022-01-28:
`
static lv_style_t style_box;  
lv_obj_t* dd = lv_dropdown_create(cont, NULL);
lv_style_init(&style_box);
lv_style_set_value_align(&style_box, LV_STATE_DEFAULT, LV_ALIGN_OUT_TOP_LEFT);
lv_style_set_value_ofs_y(&style_box, LV_STATE_DEFAULT, - LV_DPX(10));
lv_style_set_margin_top(&style_box, LV_STATE_DEFAULT, LV_DPX(30));
lv_obj_add_style(dd, LV_CONT_PART_MAIN, &style_box);
lv_obj_set_style_local_value_str(dd, LV_CONT_PART_MAIN, LV_STATE_DEFAULT, "Dropdown");
lv_obj_set_width(dd, 200);
lv_obj_set_pos(dd, 16, 175);
lv_dropdown_set_options(dd, "Alpha\nBravo\nCharlie\nDelta\nEcho");
`
**cont** is container on second. 

https://user-images.githubusercontent.com/38400203/151565705-6e148864-8808-4f6a-8551-df88a8277ab8.mp4

kisvegabor, 2022-01-31:
It seems you are using v7. Exactly which version?

The video is not playing. :slightly_frowning_face:

bfjm, 2022-02-07:
yep, v.7.11.1

bfjm, 2022-02-07:
[image]

kisvegabor, 2022-02-16:
I've tested with v7.11 and it looks well here:
[image]

closed 2022-03-11 by lvgl-bot