lvgl/lvgl #3013: lv_scr_anim_load with auto_del set to true causes crashes sometimes.

Satnet, 2022-01-16:
### Describe the bug
lv_scr_anim_load works perfectly for the most part, but it can on ocation cause LVGL to freeze, which in the end makes the MCU reboot (crash). It happens most likely because the object that's being deleted is still being "used" in the animation transition.

### To Reproduce
To reproduce this, simply enable auto_del. This is intended to delete the previous screen which it does, but it can sometimes freeze LVGL, aspecially for me who has a menu where I click around allot.

### Screenshots or video
Take my word for it, I have tested with disabling the auto delete and that causes the crashes to completely go away. As soon as I enable auto_del they come back at random points when clicking around (changing screens). But only sometimes! The crashes are random and happen say 20% of the time.

I also appreciate any workaround for this, I want previous screens to be deleted safely when leaving them.

Satnet, 2022-01-16:
I have found the issue to lie here, if I uncomment this line, the issue goes away, alltho this will ofc overload the memory eventually with to many screens. But that's not the point.

`static void scr_anim_ready(lv_anim_t * a)
{
    lv_disp_t * d = lv_obj_get_disp(a->var);

    lv_event_send(d->act_scr, LV_EVENT_SCREEN_LOADED, NULL);
    lv_event_send(d->prev_scr, LV_EVENT_SCREEN_UNLOADED, NULL);

    if(d->prev_scr && d->del_prev) lv_obj_del(d->prev_scr); // <- THIS SOMETIMES FAIL WHICH CAUSES FREEZE.
    d->prev_scr = NULL;
    d->scr_to_load = NULL;
    lv_obj_remove_local_style_prop(a->var, LV_STYLE_OPA, 0);
}
`

Satnet, 2022-01-17:
Further investigation proves my theory, crash always happens randomly when lv_obj_del is called, but it's not actually lv_obj_del that is the issue. My theory states that it has to do with lv_scr_act() which is used to tell which screen is active. When this function is unsure (bug) lv_obj_del causes a freeze.

```
[Trace] (33.885, +0)     lv_mem_realloc: using zero_mem         (in lv_mem.c line #186)
[Trace] (33.885, +0)     lv_mem_free: freeing 0x20017384        (in lv_mem.c line #161)
[Trace] (33.885, +0)     lv_mem_free: freeing 0x2001735c        (in lv_mem.c line #161)
[Trace] (33.885, +0)     lv_mem_free: freeing 0x20016eb8        (in lv_mem.c line #161)
[Trace] (33.886, +1)     lv_mem_free: freeing 0x20018348        (in lv_mem.c line #161)
[Trace] (33.886, +0)     lv_mem_realloc: reallocating 0x200182e0 with 32 size   (in lv_mem.c line #184)
[Trace] (33.886, +0)     lv_mem_realloc: allocated at 0x200182e0        (in lv_mem.c line #203)
[Trace] (33.886, +0)     lv_mem_realloc: reallocating 0x200182e0 with 24 size   (in lv_mem.c line #184)
[Trace] (33.886, +0)     lv_mem_realloc: allocated at 0x200182e0        (in lv_mem.c line #203)
[Trace] (33.886, +0)     lv_mem_realloc: reallocating 0x200182e0 with 16 size   (in lv_mem.c line #184)
[Trace] (33.886, +0)     lv_mem_realloc: allocated at 0x200182e0        (in lv_mem.c line #203)
[Trace] (33.886, +0)     lv_mem_realloc: reallocating 0x200182e0 with 8 size    (in lv_mem.c line #184)
[Trace] (33.886, +0)     lv_mem_realloc: allocated at 0x200182e0        (in lv_mem.c line #203)
[Trace] (33.886, +0)     lv_mem_realloc: reallocating 0x200182e0 with 0 size    (in lv_mem.c line #184)
[Trace] (33.886, +0)     lv_mem_realloc: using zero_mem         (in lv_mem.c line #186)
[Trace] (33.886, +0)     lv_mem_free: freeing 0x200182e0        (in lv_mem.c line #161)
[Trace] (33.886, +0)     lv_mem_free: freeing 0x2000d39c        (in lv_mem.c line #161)
[Trace] (33.886, +0)     lv_mem_free: freeing 0x20019364        (in lv_mem.c line #161)
[Trace] (33.886, +0)     lv_mem_realloc: reallocating 0x20016d38 with 16 size   (in lv_mem.c line #184)
[Trace] (33.886, +0)     lv_mem_realloc: allocated at 0x20016d38        (in lv_mem.c line #203)
[Trace] (33.886, +0)     lv_mem_free: freeing 0x20018994        (in lv_mem.c line #161)
[Trace] (33.886, +0)     lv_obj_del: finished (delete 0x20018994)       (in lv_obj_tree.c line #81)
```

The last line is where LVGL froze.

kisvegabor, 2022-01-19:
For me it worked in a reliable way.

Could you create a stress test (i.e. switch screen in a loop) and share it?

embeddedt, 2022-01-19:
Hmm, if it works for @kisvegabor and not you, do you possibly have an `LV_EVENT_DELETED` handler that does some type of invalid memory access?

If you make a stress test I can also try running it through Valgrind when I have a chance. It tends to be effective at pinpointing this type of issue.

Satnet, 2022-02-04:
Hello, I am sorry for the late reply, I rebuilt my office and took everything down. Finally up and running properly!

I just tested this with a brand new MCU and what do you know, it works fine. I should have tested this in an earlier stage, something I will keep in mind for the future!

kisvegabor, 2022-02-08:
Happy to here it's working. :slightly_smiling_face: 

I'm closing this issue.

closed 2022-02-08 by kisvegabor