lvgl/lvgl #3014: Compilation error: size of array 'static_assert298' is negative

nicklasb, 2022-01-16:
Hi,

I am trying to compile my project with:
* lvgl 8.1.0 (and 8.0.2) in PlatformIO (C 5.2.4, H 3.4.0, VSC 1.63.2)
* Both as the pio library and the cloned repo
* On macOS 12.1 

And I am getting ".pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:294:29: error: size of array 'static_assert298' is negative", here:

#define tlsf_static_assert(exp) \
    typedef char _tlsf_glue(static_assert, __LINE__) [(exp) ? 1 : -1]

https://github.com/lvgl/lvgl/blob/935caa64e4c33d63ec937a323d93d53730ec487e/src/misc/lv_tlsf.c#L294

I am not master of the language, and it seems I am the only one having this problem, so I guess there is some config I should've done since it seems to be some checks of sorts..?

nicklasb, 2022-01-16:
Not sure if that is relevatn, but here is my lv_conf (trying to get my SSD1306 to rock lvgl):
```
#define LV_MEM_SIZE      (2U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_HOR_RES          (128)
#define LV_VER_RES          (64)
#define LV_ANTIALIAS         0       /*1: Enable anti-aliasing*/

#define LV_COLOR_DEPTH      1                     /*Color depth: 1/8/16/32*/
#define USE_LV_SHADOW           0               /*1: Enable shadows*/
#define USE_LV_GROUP            0               /*1: Enable object groups (for keyboards)*/
#define USE_LV_GPU              0               /*1: Enable GPU interface*/
#define USE_LV_REAL_DRAW        0               /*1: Enable function which draw directly to the frame buffer instead of VDB (required if LV_VDB_SIZE = 0)*/
#define USE_LV_FILESYSTEM       0               /*1: Enable file system (might be required for images*/
#define USE_LV_LOG      0   /*Enable/disable the log module*/
// I turned off all the other themes
#define USE_LV_THEME_MONO       1       /*Mono color theme for monochrome displays*/
#define LV_MEM_SIZE      (2U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_HOR_RES          (128)
#define LV_VER_RES          (64)
#define LV_ANTIALIAS         0       /*1: Enable anti-aliasing*/

#define LV_COLOR_DEPTH      1                     /*Color depth: 1/8/16/32*/
#define USE_LV_SHADOW           0               /*1: Enable shadows*/
#define USE_LV_GROUP            0               /*1: Enable object groups (for keyboards)*/
#define USE_LV_GPU              0               /*1: Enable GPU interface*/
#define USE_LV_REAL_DRAW        0               /*1: Enable function which draw directly to the frame buffer instead of VDB (required if LV_VDB_SIZE = 0)*/
#define USE_LV_FILESYSTEM       0               /*1: Enable file system (might be required for images*/
#define USE_LV_LOG      0   /*Enable/disable the log module*/
// I turned off all the other themes
#define USE_LV_THEME_MONO       1       /*Mono color theme for monochrome displays*/
```

kisvegabor, 2022-01-17:
[This assert](https://github.com/lvgl/lvgl/blob/935caa64e4c33d63ec937a323d93d53730ec487e/src/misc/lv_tlsf.c#L298) can fail only if `CHAR_BIT == 0`.

Please try adding this [here](https://github.com/lvgl/lvgl/blob/935caa64e4c33d63ec937a323d93d53730ec487e/src/misc/lv_tlsf.c#L9):

```c
#if CHAR_BIT == 0
#error "CHAR_BIT==0"
#elif CHAR_BIT == 8
#error "CHAR_BIT==8"
#else
#error "CHAR_BIT not 0 and not 8"
#endif
```

nicklasb, 2022-01-17:
Thanks!

The snippet claims that CHAR_BIT is set to 8; "#error "CHAR_BIT==8", happens, if I add it. 

So it doesn't seem to be zero. As I understand more of the syntax and errors, it seems that they all fail.

````
Processing uno (platform: atmelavr; board: uno; framework: arduino)
----------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES: 
 - framework-arduino-avr 5.1.0 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.10.12
|   |-- <Adafruit BusIO> 1.10.1
|   |   |-- <Wire> 1.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit SSD1306> 2.5.1
|   |-- <Adafruit GFX Library> 1.10.12
|   |   |-- <Adafruit BusIO> 1.10.1
|   |   |   |-- <Wire> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <Adafruit BusIO> 1.10.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <lvgl> 8.1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in debug mode
Compiling .pio/build/uno/libbfa/lvgl/misc/lv_tlsf.c.o
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:13:6: error: #error "CHAR_BIT==8"
     #error "CHAR_BIT==8"
      ^~~~~
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:302:29: error: size of array 'static_assert305' is negative
     typedef char _tlsf_glue(static_assert, __LINE__) [(exp) ? 1 : -1]
                             ^
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:299:27: note: in definition of macro '_tlsf_glue2'
 #define _tlsf_glue2(x, y) x ## y
                           ^
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:302:18: note: in expansion of macro '_tlsf_glue'
     typedef char _tlsf_glue(static_assert, __LINE__) [(exp) ? 1 : -1]
                  ^~~~~~~~~~
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:305:1: note: in expansion of macro 'tlsf_static_assert'
 tlsf_static_assert(sizeof(int) * CHAR_BIT == 32);
 ^~~~~~~~~~~~~~~~~~
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:302:29: error: size of array 'static_assert306' is negative
     typedef char _tlsf_glue(static_assert, __LINE__) [(exp) ? 1 : -1]
                             ^
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:299:27: note: in definition of macro '_tlsf_glue2'
 #define _tlsf_glue2(x, y) x ## y
                           ^
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:302:18: note: in expansion of macro '_tlsf_glue'
     typedef char _tlsf_glue(static_assert, __LINE__) [(exp) ? 1 : -1]
                  ^~~~~~~~~~
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:306:1: note: in expansion of macro 'tlsf_static_assert'
 tlsf_static_assert(sizeof(size_t) * CHAR_BIT >= 32);
 ^~~~~~~~~~~~~~~~~~
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:302:29: error: size of array 'static_assert310' is negative
     typedef char _tlsf_glue(static_assert, __LINE__) [(exp) ? 1 : -1]
                             ^
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:299:27: note: in definition of macro '_tlsf_glue2'
 #define _tlsf_glue2(x, y) x ## y
                           ^
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:302:18: note: in expansion of macro '_tlsf_glue'
     typedef char _tlsf_glue(static_assert, __LINE__) [(exp) ? 1 : -1]
                  ^~~~~~~~~~
.pio/libdeps/uno/lvgl/src/misc/lv_tlsf.c:310:1: note: in expansion of macro 'tlsf_static_assert'
 tlsf_static_assert(sizeof(unsigned int) * CHAR_BIT >= SL_INDEX_COUNT);
 ^~~~~~~~~~~~~~~~~~
*** [.pio/build/uno/libbfa/lvgl/misc/lv_tlsf.c.o] Error 1
````

kisvegabor, 2022-01-19:
Do you use this [board](https://docs.platformio.org/en/latest/boards/atmelavr/uno.html)?

If so, it won't be enough for LVGL. :slightly_frowning_face: 
See the requirements [here](https://docs.lvgl.io/master/intro/index.html#requirements).

nicklasb, 2022-01-20:
Hi, 

Yes, I realized that I had misread that after getting it to compile (after removing those asserts).
Not a big thing though, I am getting an ESP32 shortly. 

However, I am not sure that this compilation error should occur anyway, on those grounds?
At least in this case, it made it take way longer for me to realize that the board wouldn't work.

nicklasb, 2022-01-20:
Either way, I am closing this issue as I'm satisfied with the response.
Thanks a bunch for your help @kisvegabor!

closed 2022-01-20 by nicklasb

kisvegabor, 2022-01-20:
I was checking a  few things but couldn't imagine a good way to test if it's a 8 bit architecture. 
If we can come up with something reliable, I'd be happy to add guards to detect issue like this.

nicklasb, 2022-01-20:
Aren't there any bit-shift-stuff or memory tricks that returns differently on different architectures?
Anyways; people have been thinking about it: 
https://stackoverflow.com/questions/152016/detecting-cpu-architecture-compile-time