lvgl/lvgl #3015: Performance monitor NULL pointer access

FASTSHIFT, 2022-01-17:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Decoupling the disp_refr timer: https://github.com/lvgl/lvgl/commit/85cc84ad947786bb3d4857290503047946a55c43. Since `disp_refr->refr_timer = NULL` will lead to null pointer access after the performance monitor is turned on. Is there any good solution?

### To Reproduce
`#define LV_USE_PERF_MONITOR 1` in `lv_conf.h`

```c
lv_disp_t *disp = lv_disp_get_default();
lv_timer_del(disp->refr_timer);
disp->refr_timer = NULL;

while (1)
{
  lv_timer_handler();
  _lv_disp_refr_timer(NULL);
  usleep(1000);
}
```

### Expected behavior
Can correctly count frame rate.

### Screenshots or video
[image]

embeddedt, 2022-01-17:
`refr_timer->period` is only used to determine the maximum possible FPS, so I think it would be safe to just set `fps_limit` to an arbitrarily high number in the case that there is no timer. I think the FPS will still display correctly since that value is just the maximum.

FASTSHIFT, 2022-01-17:
@embeddedt 
How about `LV_DISP_DEF_REFR_PERIOD` instead of `disp_refr-> timer->period`?

kisvegabor, 2022-01-19:
> How about LV_DISP_DEF_REFR_PERIOD instead of disp_refr-> timer->period?

I think it'd be a good fallback if `disp_refr-> timer->period` is not available.

embeddedt, 2022-01-19:
Hmm... doesn't that mean the reported frame time would be artificially capped by `LV_DISP_DEF_REFR_PERIOD` even if the display is being refreshed faster?

kisvegabor, 2022-01-19:
It does. But showing e.g. 1000 FPS on idle would look ugly too. `LV_DISP_DEF_REFR_PERIOD` at least provides some control over the max value.

FASTSHIFT, 2022-02-10:
> It does. But showing e.g. 1000 FPS on idle would look ugly too. `LV_DISP_DEF_REFR_PERIOD` at least provides some control over the max value.

[image]

This is indeed the case, I suggest that it is more appropriate to modify it like this:

[image]

FASTSHIFT, 2022-02-10:
@kisvegabor 
Need to reopen this issue :)

kisvegabor, 2022-02-18:
#3105 is merged so I close this issue.

Thank you fr the PR!

closed 2022-02-18 by kisvegabor