lvgl/lvgl #3018: LVGL Cortex-M55 (MPS3) Porting

GorgonMeducer, 2022-01-17:
Hi all,

In order to demonstrate the LVGL cmsis-pack, I have created a porting for Corstone-300 running on Arm MPS3 development board that hosts a Cortex-M55 example design. Arm also provides a free emulation tool called FVP that can be used together with MDK (so you don't need the real HW, i.e. MPS3).  

Could you please create a repo as lv_port_mps3_an547_cm55 ?

Please check https://github.com/GorgonMeducer/lv_port_mps3_an547_cm55 for details.

Thank you.

GorgonMeducer, 2022-01-19:
This porting is using the latest cmsis-pack v0.8.3 (https://github.com/lvgl/lvgl/pull/3021)

kisvegabor, 2022-01-19:
So Cortex-M55 is still not available in a real chip, right?

Is "MPS3 development board" for "causal developers"? It seems to me it's for beta testing new core architectures but correct me if I see it wrong.

GorgonMeducer, 2022-01-20:
@kisvegabor are there any suggestions?

kisvegabor, 2022-01-22:
Okay, I'm convinced and just created the repo: https://github.com/lvgl/lv_port_mps3_an547_cm55
I also sent and invitation to you with "Maintainer" role. 

---

Do you think we can use  "Corstone-300-FVP" to benchmark LVGL in reproducible way in real hardware-like environment? The goal would be to have a CI which measures the execution time (or number of clocks) for some specific operations (mainly rendering) and report if there is regression.

GorgonMeducer, 2022-01-22:
To be honest, Corstone-300-FVP is not designed for performance evaluation, and Arm always suggests that performance data should be derived from real hardware, i.e. here MPS3 + AN547 FPGA image. 

But if we don't pursue "absolute performance data", but want to observe "accumulated performance differences" in a relative manner (between different code versions), then this is possible. 

I have a dedicated project https://github.com/GorgonMeducer/perf_counter that can measure CPU cycles for all Cortex-M processors. It uses Systick without interfering with the existing SysTick Applications, e.g. period timer or RTOS tick timer etc. The problem for FVP is, its emulation of SysTick is not accurate, you might find its value is the times of 25000 cycles... That's why I say, FVP cannot be used for absolute performance evaluation.

2D operations usually take millions of cycles, hence using SysTick to observe a relative performance difference is viable. 

I hope this helps. @kisvegabor

GorgonMeducer, 2022-01-23:
Furthermore, the Corstone-300-FVP supports command line mode and built-in semihoting. That means **_in command-line mode_** (**NOT** inside MDK), you can use almost all stdio.h features to access host PC file system, i.e. fopen, fread, fwrite, fprintf etc without additional retargeting.  Hence it is suitable for CI system I believe.

kisvegabor, 2022-01-24:
Thanks. On a 100 MHz MCU it's 0.25 ms accuracy which is enough for us. We can even run the tests multiple times to get longer execution time and higher precision. 

Can you link some info about the command line mode?

```c
    __cycleof__("Draw strings on LCD") {
        __LL_LCD_PRINT_BANNER("Hello LVGL!!");
    }
```

from the README seems quite handy.

GorgonMeducer, 2022-01-24:
```
Usage: FVP_Corstone_SSE-300_Ethos-U55.exe [OPTION]... [APPLICATION]...

-L, --cadi-log                                      log all CADI calls into an XML logfile
-S, --cadi-server                                   start CADI server allowing debuggers to connect to targets in the simulation
-I, --iris-server                                   Start Iris server allowing debuggers to connect to targets in the simulation
-i, --iris-log                                      Iris log level (may be specified multiple times), example: -ii for level 2
-A, --iris-allow-remote                             Allow remote connections from other machine to Iris server (default: disabled)
    --cpulimit N                                    number of host active CPU seconds for the simulation to run, excluding simulation startup and shutdown.
                                                    Ignored when debug server started (default: unlimited)
    --cyclelimit N                                  number of cycles to run, ignored when debug server started (default: unlimited)
-h, --help                                          print this help message and exit
-K, --keep-console                                  keep console window open after completion (Windows only)
    --list-instances                                print list of target instances to standard output
-l, --list-params                                   print list of model parameters to standard output
    --dump-params                                   dump list of model parameters to JSON file
    --list-regs                                     print model register information to standard output
    --check-regs                                    same as --list-regs but do more consistency checks on the CADI register API
    --list-memory                                   print model memory information to standard output
-o, --output FILE                                   redirect parameters, memory and instance lists to output file FILE
-C, --parameter SPEC                                set parameter, format: -C INST.PARAM=VALUE
-p, --print-port-number                             print port number debug server is listening to. Use in combination with --cadi-server or --iris-server
-R, --run                                           run the simulation immediately at start of simulation when a debug server is started. Use in combination with --cadi-server or --iris-server
    --stat                                          print some run statistics on simulation exit
-T, --timelimit N                                   number of wall clock seconds for the simulator to run, excluding startup and shutdown.
                                                    Ignored when debug server started (default: unlimited)
-b, --break ADDRESS                                 set breakpoint, format: -b [INST=]ADDRESS (may be specified multiple times)
    --start ADDRESS                                 set initial PC to application start address, format: --start [INST=]ADDRESS
    --dump FILE@ADDRESS,SIZE                        dump section of memory into FILE, format: --dump [INST=]FILE@[MEMSPACE:]ADDRESS,SIZE (may be specified multiple times)
-q, --quiet                                         run quiet, suppress informational output
-f, --config-file FILE                              load model parameters from configuration file FILE
    --plugin FILE                                   load plugin FILE
    --trace-plugin FILE (DEPRECATED: use --plugin)  load trace plugin FILE
-a, --application FILE                              application to load, format: -a [INST=]FILE (use -a INST=FILE for a specific instance, use -a INST*=FILE to match multiple instances using wildcards e.g. for SMP cores)
    --fast-ram FILE                                 enable FastRAM and load the configurations of FastRAM from FILE
-s, --save DIR                                      Save checkpoint to DIR on simulation exit
-r, --restore DIR                                   Restore checkpoint from DIR on simulation startup
    --disable-analytics                             disable product analytics gathering for current run.
                                                    For persistent disabling, set ARM_DISABLE_ANALYTICS env variable.
    --simlimit N                                    Number of seconds to simulate, ignored when debug server started (default: unlimited)
    --iris-port-range MIN:MAX                       Set the range of ports to scan when starting an Iris server.
                                                    The first available port found will be used.
    --iris-port PORT                                Set a specific port to use for the Iris server.
    --data FILE@ADDRESS                             raw data to load at specified address, format: --data [INST=]FILE@[MEMSPACE:]ADDRESS (may be specified multiple times)
-D, --allow-debug-plugin                            allow a plugin to debug the simulation
    --cpi-file FILE                                 Use FILE to set Cycle Per Instruction class
-Q, --quantum N              Number of ticks to simulate per quantum (default to 10000)
-M, --min-sync-latency N     Number of ticks to simulate  before synchronizing. Events occurring
                             at a higher frequency than this are missed
                             (default to 100)
    --log FILE               Log all SystemC reports into given FILE
    --disable-model-exitcode Disables the retrieval of any exit code
                             potentially returned by the model or a plugin
    --version                Print version info

To stop running simulation press Ctrl+C
```

GorgonMeducer, 2022-01-24:
@kisvegabor  
```c
__cycleof__(__STRING) {
    /* Your code block here */
}
```

GorgonMeducer, 2022-01-24:
I have added an FVP configuration file for the command line mode, and the following option is used to disable popup window.

https://github.com/lvgl/lv_port_mps3_an547_cm55/blob/5f3ce4a69cc6d58db796099525a0df6664589e6b/fvp_cfg.txt#L1692

kisvegabor, 2022-01-27:
Awesome, thank you!

One more question: any chance to run it on Linux?

GorgonMeducer, 2022-01-27:
> Awesome, thank you!
> 
> One more question: any chance to run it on Linux?

Yes, there is a Linux version: https://developer.arm.com/-/media/Arm%20Developer%20Community/Downloads/OSS/FVP/Corstone-300/FVP_Corstone_SSE-300_11.16_26.tgz

kisvegabor, 2022-01-29:
Thanks, I've installed the Linux version and found `FVP_Corstone_SSE-300_Ethos-U55`. If I start it I get this window which seems good:
[image]

I still don't know how the create an app (I.e. build a LVGL UI?) without Keil. Sorry for these basic questions. If you can point to a get started tutorial, I can probably figure out these things on my own . :slightly_smiling_face:

GorgonMeducer, 2022-01-29:
If you have MDK and open the project, you should be able to find a very useful option “Create Batch File”

[image]

Once you select this option, after a successful compilation, a batch file will be generated. 

[image]

It contains all the information for compiling the project in command line mode. It is similar to cmake verbal output. 
With that information, it should be sufficient for you to create a makefile or cmake.

If you still have questions, please feel free to let me know.

GorgonMeducer, 2022-01-29:
Let me reopen this discussion.

kisvegabor, 2022-02-02:
I've just realized the Keil doesn't support Linux. 

As I'm using Linux I'll need some time to make experiments.

I'll post here if I could try out the mentioned things.

GorgonMeducer, 2022-02-02:
@kisvegabor don't worry, I will generate all related files and upload them to https://github.com/lvgl/lv_port_mps3_an547_cm55.

kisvegabor, 2022-02-24:
Sorry, I didn't have time to dive into this topic :slightly_frowning_face: 

@embeddedt pinned this issue so it will remain opened as a reminder.

GorgonMeducer, 2022-03-08:
Hi @kisvegabor,

Maybe you haven't noticed that Arm released a community license for MDK which allows users to compile and debug freely for non-commercial purposes only:

[image]

So please don't worry about the license issue.

kisvegabor, 2022-03-12:
Thanks for sharing this, @GorgonMeducer!

kisvegabor, 2022-05-28:
Let's warm up this topic! In summery it'd be great to  create a CI from Corstone-300-FVP to test the performance of some critical parts: 
- malloc
- layouts
- widgets creation
- drawing

The [README](https://github.com/GorgonMeducer/perf_counter) describes very well how to measure the cycle count. So the only question is how to compile a GUI/LVGL project on Linux? I've installed https://developer.arm.com/-/media/Arm%20Developer%20Community/Downloads/OSS/FVP/Corstone-300/FVP_Corstone_SSE-300_11.16_26.tgz

but can't see where to add and how t compile a project.

GorgonMeducer, 2022-05-28:
@kisvegabor FVP is used to run given elf / axf file. You can use cmake/makefile to compile a LVGL/GUI project.

kisvegabor, 2022-05-30:
I might have missed some relevant docs, but it's still not clear to me how to turn run an LVGL code with Corstohowne-300-FVP. 
E.g. 
- how to compile it?
- what driver to use?
-  how to load the elf?

GorgonMeducer, 2022-05-30:
For your questions:
> how to compile it?
Please use the normal toolchain to generate an elf file for Cortex-M55, i.e. gcc, llvm or armclang

> what driver to use?
No driver is required

> how to load the elf?
Use command line

kisvegabor, 2022-06-02:
I hoped for a project which contains a hello world LVGL label with an example Makefile. I still see too many questions to get started with it (on Linux) without a super basic example. :slightly_frowning_face: 

Is there any chance that a project like this will be available?

GorgonMeducer, 2022-06-02:
To be honest, I am not a Linux user and not good at writing cmake or makefile, so it will take time to create such an example project. Please allow me more time.

kisvegabor, 2022-06-07:
Thank you in advance. I'm also not a makefile hero but [this one](https://github.com/lvgl/lv_port_pc_eclipse/blob/master/Makefile) can be a good starting point.

GorgonMeducer, 2022-06-07:
I will use Arm-DS to create such a project because Arm-DS can generate a makefile automatically and it supports cmsis-pack.

kisvegabor, 2022-11-14:
The discussion continued in #3443 so I close it.

closed 2022-11-14 by kisvegabor