lvgl/lvgl #3032: Physical display resolution doesn't work

GorgonMeducer, 2022-01-20:
### Perform all steps below and tick them with [x]
- [X] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I tried to port demo/widgets to a development board that has a 320*240 LCD.
It appears that the resolution is too small for the widgets:

[image]

According to the [porting document](https://github.com/lvgl/lvgl/blob/master/docs/porting/display.md#optional-fields), 

> physical_hor_res horizontal resolution of the full / physical display in pixels. Only set this when not using the full screen (defaults to -1 / same as hor_res).
> physical_ver_res vertical resolution of the full / physical display in pixels. Only set this when not using the full screen (defaults to -1 / same as ver_res).

I should be able to use the LCD to display part of a physical display, for example, the left corner of the demo, i.e. 

[image]

by applying a bigger physical resolution:

```c
    static lv_disp_drv_t disp_drv;                                              /*Descriptor of a display driver*/
    lv_disp_drv_init(&disp_drv);                                                /*Basic initialization*/

    /*Set up the functions to access to your display*/

    /*Set the resolution of the display*/
    disp_drv.hor_res = GLCD_WIDTH;
    disp_drv.ver_res = GLCD_HEIGHT;

//#if LV_USE_DEMO_WIDGETS
    disp_drv.physical_hor_res = 480;
    disp_drv.physical_ver_res = 320;
//#endif

    /*Used to copy the buffer's content to the display*/
    disp_drv.flush_cb = disp_flush;

    /*Set a display buffer*/
    disp_drv.draw_buf = &draw_buf_dsc_1;

    /*Required for Example 3)*/
    //disp_drv.full_refresh = 1

    /* Fill a memory array with a color if you have GPU.
     * Note that, in lv_conf.h you can enable GPUs that has built-in support in LVGL.
     * But if you have a different GPU you can use with this callback.*/
    //disp_drv.gpu_fill_cb = gpu_fill;

    /*Finally register the driver*/
    lv_disp_drv_register(&disp_drv);
}
```

But this doesn't work. Do I miss anything? or is this a defect?

### To Reproduce

### Expected behavior
[image]

### Screenshots or video
[image]

GorgonMeducer, 2022-01-23:
You can reproduce this with https://github.com/lvgl/lv_port_mps3_an547_cm55

In MDK RTE dialog, unselect Benchmark and select Demo:Widgets as shown below:

[image]

Compile and Debug directly.

kisvegabor, 2022-01-24:
`physical_hor/ver_res` is not used by LVGL itself at all. They can be used by drivers if needed. 

Do you get the same result if you comment out 
```c
    disp_drv.physical_hor_res = 480;
    disp_drv.physical_ver_res = 320;
```

GorgonMeducer, 2022-01-24:
@kisvegabor Yes, the result is the same. 
I thought there was some internal clipping mechanism that uses physical resolutions. 

Do you have a plan to add this feature? or maybe you can update the documents to avoid misleading.

GorgonMeducer, 2022-01-25:
@kisvegabor I tried to use 480 * 240 as the screen resolution but cannot get the expected result.

[image]

https://github.com/lvgl/lv_port_mps3_an547_cm55/blob/master/RTE/LVGL/lv_port_disp_template.c

NOTE: the GLCD_DrawBitmap is also updated and tested.

kisvegabor, 2022-01-25:
I'm trying to understand what could be the problem. Can you enable [logging and asserts](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L197-L239)?

Is there any change if you increase `LV_MEM_SIZE`?

GorgonMeducer, 2022-01-26:
> I'm trying to understand what could be the problem. Can you enable [logging and asserts](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L197-L239)?
> 
> Is there any change if you increase `LV_MEM_SIZE`?

Hi @kisvegabor 

I have increased the LV_MEM_SIZE to 64KB. It doesn't help. I think asserts are enabled by default because when I set the resolution to 480 * 240, the malloc failed and was caught by assert. 

Can you replicate the issue on yourside?

kisvegabor, 2022-01-27:
> Can you replicate the issue on your side?

What is the actual of GLCD_WIDTH and GLCD_HEIGHT?

It worked for me with any resolution I tried from 240x240 to 1960x1080.

GorgonMeducer, 2022-01-27:
GLCD_WIDTH is 320
GLCD_HEIGHT is 240.

GorgonMeducer, 2022-01-27:
> > Can you replicate the issue on your side?
> 
> What is the actual of GLCD_WIDTH and GLCD_HEIGHT?
> 
> It worked for me with any resolution I tried from 240x240 to 1960x1080.

Can you share me your lv_conf.h ?

kisvegabor, 2022-01-28:
> Can you share me your lv_conf.h ?

It's the as it is template, I only enabled the demos.

Does LVGL freezes when you see that strange screen? (the arcs should be animated if LVGL is running)

GorgonMeducer, 2022-01-28:
[image]

I cannot see the arcs. 

I can pause the running, everything looks fine.

I enabled the log:

```c
/*Enable the log module*/
#define LV_USE_LOG 1
#if LV_USE_LOG

    /*How important log should be added:
    *LV_LOG_LEVEL_TRACE       A lot of logs to give detailed information
    *LV_LOG_LEVEL_INFO        Log important events
    *LV_LOG_LEVEL_WARN        Log if something unwanted happened but didn't cause a problem
    *LV_LOG_LEVEL_ERROR       Only critical issue, when the system may fail
    *LV_LOG_LEVEL_USER        Only logs added by the user
    *LV_LOG_LEVEL_NONE        Do not log anything*/
    #define LV_LOG_LEVEL        LV_LOG_LEVEL_WARN

    /*1: Print the log with 'printf';
    *0: User need to register a callback with `lv_log_register_print_cb()`*/
    #define LV_LOG_PRINTF           1

    /*Enable/disable LV_LOG_TRACE in modules that produces a huge number of logs*/
    #define LV_LOG_TRACE_MEM        1
    #define LV_LOG_TRACE_TIMER      1
    #define LV_LOG_TRACE_INDEV      1
    #define LV_LOG_TRACE_DISP_REFR  1
    #define LV_LOG_TRACE_EVENT      1
    #define LV_LOG_TRACE_OBJ_CREATE 1
    #define LV_LOG_TRACE_LAYOUT     1
    #define LV_LOG_TRACE_ANIM       1

#endif  /*LV_USE_LOG*/
```

There is only one warning:

[image]

So I enable the 
```c
#define LV_FONT_MONTSERRAT_20     1
```

After that, there is no warning or error but the problem still exists.

GorgonMeducer, 2022-01-28:
@kisvegabor sorry, I find the issue. My fault. In the cmsis-pack, I forgot to include lv_extra.c. 

I will fix it immediately. I am really sorry about this.

[image]

Now everything works well as expected.

kisvegabor, 2022-01-29:
Great! But I still see 2 issues on the image:
1. Why only 2 tabs are visible? It seems the image is clipped.
2. There is a grayish horizontal line above the "Log out" button.

GorgonMeducer, 2022-01-29:
> Great! But I still see 2 issues on the image:
> 
> 1. Why only 2 tabs are visible? It seems the image is clipped.
- it is because I set physical resolution to 480 * 240. After changing back to 320 * 240 everything looks fine

[image]

> 2. There is a grayish-greenish horizontal line above the "Log out" button.
- It is a glitch in the FVP LCD display. You can ignore it.

kisvegabor, 2022-01-29:
All great then :slightly_smiling_face: 

BTW, I thought you will see a compiler error if `lv_extra.c` is missing. 

Anyway, feel free to close this issue if you see no more problems.

GorgonMeducer, 2022-01-29:
> BTW, I thought you will see a compiler error if lv_extra.c is missing
in (lv_cmsis_pack.c)[https://github.com/lvgl/lvgl/blob/master/env_support/cmsis-pack/lv_cmsis_pack.txt], there is a weak function. This function helps people to customize LVGL and decouple LVGL from the content inside extras.

```c
/* when people don't want to use src/extra */
__attribute__((weak))
void lv_extra_init(void)
{
}
```

> Anyway, feel free to close this issue if you see no more problems.
Sure, as #3056 is merged, there is no reason to keep this one.

Thank you for your support!!!

closed 2022-01-29 by GorgonMeducer