lvgl/lvgl #3038: V8.1 SDL GPU crash with any draw

iankellogg, 2022-01-23:
### Perform all steps below and tick them with [x]
- [x ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
LVGL crashes in simulator when using SDL GPU when any lv_draw is called

In lv_draw_sdl_init_ctx the current repository code calls
draw_ctx_sdl->renderer = ((lv_draw_sdl_drv_param_t *) disp_drv->user_data)->renderer;

this puts the wrong address in draw_ctx_sdl->renderer

I was able to fix this with
draw_ctx_sdl->renderer = (lv_draw_sdl_drv_param_t *) disp_drv->user_data;

kisvegabor, 2022-01-24:
cc @mariotaku

mariotaku, 2022-01-25:
Please pass `lv_draw_sdl_drv_param_t` instead of `SDL_Renderer` to `disp_drv->userdata`. This struct is designed to hold more fields when needed.

And which simulator repo you are referring to? `lv_sim_vscode_sdl` right?

iankellogg, 2022-01-25:
Using something based on the eclipse simulator but has been rebuilt using
latest lv_driver, and latest lvgl with a new main.c
both the eclipse and vscode simulator don't run a version of LVGL that has
SDL_GPU support

the disp_drv->userdata to monitor.renderer in sdl_gpu_disp_drv_init in
sdl_gpu.c
If i change disp_drv->userdata to lv_draw_sdl_drv_param_t in sdl_gpu.c that
would require modifications all over sdl_gpu.c as that uses user_data
everytime for renderer. I can make those changes, but seems excessive
considering lv_draw_sdl_drv_param_t only contains SDL_Renderer.

jonbarnettk, 2022-01-25:
I agree with Iankellog.

I actually had this problem before and solved it in this way, but then I
had to add my own data to lv_draw_sdl_drv_param_t which then got
overridden when I went to master.

I feel this is inflexible because we lose user_data (those use it). For
example I kept SDL_Window in user_data and also created gl_context, because
it was useful for my purposes.

If this is the design, can we add a user_data to lv_draw_sdl_drv_param_t to
enable adding whatever we need, or what may be more sensible is create a
new property in the driver structure for drawparams and leave user_data?

I know this design is still in the works, so I am not sure the additional
plans for this struct, but some userdata property would be useful.

mariotaku, 2022-01-25:
@iankellogg @jonbarnettk I have updated [lvgl](https://github.com/lvgl/lvgl/pull/3045) and [lv_driver](https://github.com/lvgl/lv_drivers/pull/197), does that work for you?

jonbarnettk, 2022-01-25:
Good for me, thank you and thank you for all your awesome work on the sdl
stuff

iankellogg, 2022-01-29:
@mariotaku  you're branches look like they work for me. THanks

kisvegabor, 2022-01-29:
> you're branches look like they work for me. 

:+1: 

I close this issue then.

closed 2022-01-29 by kisvegabor