lvgl/lvgl #3039: V8.1 SDL GPU canvas does not display

iankellogg, 2022-01-23:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
After doing the fix in #3038 
Canvas crashes simulator when using SDL GPU
I added 
    #if LV_USE_GPU_SDL
        sdl_gpu_disp_drv_init(disp->driver);
    #else
    lv_disp_drv_init(disp->driver);
    #endif
to init_fake_disp
to fix the canvas crash but now canvas does not display in program
Images and other objects display just fine

### Expected behavior
Expected to see rendered canvas object in program when using SDL GPU. 

### Screenshots or video
[image]
when using SDL GPU

[image]
when using SDL

kisvegabor, 2022-01-24:
cc @mariotaku

embeddedt, 2022-01-25:
@iankellogg I believe both of your issues are fixed in `master`.

kisvegabor, 2022-02-11:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.