lvgl/lvgl #3041: Image cache and decoder optimization

FASTSHIFT, 2022-01-24:
### Introduce the problem
* The current image cache uses the number of images to limit how many images can be opened at the same time. This may cause two problems: if there are many low-resolution images, the memory cannot be fully used to reduce the open/close operations of the decoder; If the user uses a higher resolution image, it will easily run out of memory.

*  I have a GPU on my hardware that can accelerate image rendering, but this GPU requires that the width of the image must be an integer multiple of 16. At present, my approach is to decode the image and copy the original image into a blank image aligned to a width of 16px, and pass the copied image to lvgl. I have encountered a problem, when using the bin format image, I can customize the decoder to convert the image. But when I wanted to use png decoding, I had to modify the source code of the `lv_png` decoder, insert and call my own image conversion function to convert the image into a new image with a width aligned to 16 pixels. Is there a better solution?

### Examples and cases

### Suggested solution
* Limit the total memory size occupied by the image cache, rather than the number of open images. The memory occupied by the decoder that is opened each time can provide statistics in two ways:
1 ) lvgl records the image occupancy directly according to the `width * height * (bytes per pixel)` of the image.
2 ) The memory occupied by the custom decoder report itself. 
The default mode can be the first one.

* Is it possible to add a decoder secondary processing interface, which is called after each deocder `open_cb` is completed, to adapt to the situation that the hardware has special requirements for the image size?

kisvegabor, 2022-01-27:
Hi,

I agree with both things.

For the cache, I think "The memory occupied by the custom decoder report itself." is more reasonable as in some cases the images are read line-by-line or in block so the memory usage is much smaller than width x height.   
#2904 is a huge rework on the decoder interface. I suggest waiting with this feature until it gets merged.

Regarding the stride, so far I thought this issue is related only to the built-in "byte array" format but you are right, decoded PNG images  also can have stride constraints. Maybe I missed something, but it seems your suggestion will require `width x height` RAM for the normal decode plus `(width + padding) x height` for the alignment and then the first `width x height`  can be freed. 

I have no good idea for it, it'd be great to avoid having two image sized array allocated even for a short time.

X-Ryl669, 2022-01-28:
> If the user uses a higher resolution image, it will easily run out of memory.

Depends on the decoder. Some (most ?) decoder can deal with this case by only rendering line by lines. Some don't. 

Right now, the image cache is being reworked. I haven't accounted for image stride in my work, but if you need it, it shouldn't be hard to add. If I understand correctly, you'll need to add a pointer to a function `img_alloc_stride_requirements` or something in the draw context and have your GPU code fill this pointer to a function returning 16 or something.

Please notice that the stride and the allocation are 2 different things here: one is the number of pixels and the other might be the buffer size (in bytes, depends on the color format). Some decoders (many?) have their own allocation requirement and you'll have to copy between both format (so in the end, for the whole image, it'll be inefficient). The decoders that read line by line will be the most simple to fix, since they absolutely don't care about the buffer they are rendering into.

Concerning the image cache size, right now the easiest solution was implemented. We have thought about how to improve it and the conclusion was that we should make it "pixel" based (instead of "image" based). However, I haven't found a way to avoid fragmenting the cache (like I did for gradient by memcpy'ing the buffers around). So, right now, I've left this to some time later.

FASTSHIFT, 2022-02-02:
OK, I will try to modify my decoder first. By the way, when will the new decoder interface be ready?

FASTSHIFT, 2022-02-09:
@X-Ryl669

X-Ryl669, 2022-02-09:
You can rebase on #2904 for how it's going. Right now, there isn't any stride in the draw_img interface, so if you need to add one, you'll need to modify both lv_img_decoder.h and lv_draw_sw_blend.h to include (in the former) and account for (in the latter) a stride member in their structures.
I haven't thought about this, but in a bird eye view, you'll need to add a `uint16_t stride` to the decoder's context's dec_ctx_t type. 

In each decoder, when allocating the buffer, you'll check if the picture's width is a multiple of the stride (beware of a 0 stride). This can be done with `alloc_width = stride ?  (width  + stride - 1) & ~(stride - 1) : width` and allocate with the `alloc_width` instead of the width 

Then, since you're using a GPU, you'll need to provide your own `draw_img_decoded()` or blend function (I don't know your details here). If the GPU just need 16 bytes aligned buffer, and you're drawing line by line (very likely), then there might be no modification required, just make sure you're progressing in `alloc_width` steps in your buffer instead of `width`.

In the the sw function however, it needs to  account for the stride so a similar change is required.

In decoder decoding lines by lines (like many of those), you'll just need to make sure the provided buffer is correctly aligned with the right padding. It happens in `lv_draw_img.c`.

Hope it helps.

FASTSHIFT, 2022-02-15:
@X-Ryl669 
Any ideas on how to manage the cache based on "limiting the total size of memory occupied by the image"? If memory is sufficient, it may be better to buffer the entire image.

X-Ryl669, 2022-02-16:
We had a (short) discussion about this before. I don't exactly remember the exact conclusions but it was like this:
 1. The image cache should be changed not to store an number of entries, but to adapt to a given memory pool (so if you have plenty of small images it can fit in the cache, but not if you have one big picture).
 2. Component/widget API might be modified to snapshot themselves to speed up rendering if they are not modified in different render passes.

None of the above is done yet. What you have now is some decoders that are able to render to the `img_data` member of the decoder context. If this member (which is saved in the cache) is filled, then the complete picture is stored here and rendered, bypassing the call to decoder's `read_line`.

kisvegabor, 2022-02-18:
> The image cache should be changed not to store an number of entries, but to adapt to a given memory pool (so if you have plenty of small images it can fit in the cache, but not if you have one big picture).

It seems doable, but @X-Ryl669 pointed out fragmentatin can be big issue here as there will be large images too required large continuous memory.  

> Component/widget API might be modified to snapshot themselves to speed up rendering if they are not modified in different render passes.

It's definitely on the roadmap and I'm planning to work on it in the near future.

X-Ryl669, 2022-02-18:
Just a side question. I'm wondering how stable is the TLFS allocator in LVGL ? Can we use it actually without fear for fragmentation or will it fail to merge back to a some (efficient) state if we free in different order than the allocations ?

kisvegabor, 2022-02-21:
> Just a side question. I'm wondering how stable is the TLFS allocator in LVGL ? Can we use it actually without fear for fragmentation or will it fail to merge back to a some (efficient) state if we free in different order than the allocations ?

AFAIK, TLSF is very stable and we can use it without worrying abut performance or fragmentation.  It surely joins adjacent free block regardless to the `free` order.

closed 2022-03-15 by lvgl-bot