lvgl/lvgl #3046: lvgl 8.1.0 add rounder_cb() to fit the oled driver ic ,but the problem still exist in some case.

LianKaFoHaLi, 2022-01-25:
@LianKaFoHaLi 

How did you initialize the display driver? Please copy/paste including the display buffer size an initialization.

(Please send the code not an image to make it easy to copy into my project.)
----------------------------------------------------------------------------------------------------------------------------------------

Thank you for your reply,the following is lv_port_disp.c file and description.

To solve this problem ,I define another full screen size buffer in oled driver,when refresh area don't meet the requirement,I called 'oled_write_buf()' function that write data into another screen size buffer and complement data to fit the requirement,then use spi dma send data to screen.if refresh area fit the requirement,called 'oled_write_buf_refresh()' function that write data into another screen size buffer,meanwhile use spi dma send data to screen.
because of 'my_rounder_cb' function added,above 90% refresh events are fit the requirement. But it's cost too many ram resources to define another full screen size buffer.
I still want to do some measure that make the lvgl refresh area fit the oled requirement in 'disp_flush()' function.

LianKaFoHaLi, 2022-01-25:
`/**
 * @file lv_port_disp_templ.c
 *
 */

 /*Copy this file as "lv_port_disp.c" and set this value to "1" to enable content*/
#if 1

/*********************
 *      INCLUDES
 *********************/
#include "lv_port_disp.h"
#include "lvgl.h"
// #include "userdef.h"
/*********************
 *      DEFINES
 *********************/
#define MY_DISP_HOR_RES 126
#define MY_DISP_VER_RES 294  

/**********************
 *      TYPEDEFS
 **********************/

/**********************
 *  STATIC PROTOTYPES
 **********************/
static void disp_init(void);

static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p);
//static void gpu_fill(lv_disp_drv_t * disp_drv, lv_color_t * dest_buf, lv_coord_t dest_width,
//        const lv_area_t * fill_area, lv_color_t color);

/**********************
 *  STATIC VARIABLES
 **********************/

/**********************
 *      MACROS
 **********************/

/**********************
 *   GLOBAL FUNCTIONS
 **********************/

void lv_port_disp_init(void)
{
    /*-------------------------
     * Initialize your display
     * -----------------------*/
    disp_init();

    /*-----------------------------
     * Create a buffer for drawing
     *----------------------------*/

    /**
     * LVGL requires a buffer where it internally draws the widgets.
     * Later this buffer will passed to your display driver's `flush_cb` to copy its content to your display.
     * The buffer has to be greater than 1 display row
     *
     * There are 3 buffering configurations:
     * 1. Create ONE buffer:
     *      LVGL will draw the display's content here and writes it to your display
     *
     * 2. Create TWO buffer:
     *      LVGL will draw the display's content to a buffer and writes it your display.
     *      You should use DMA to write the buffer's content to the display.
     *      It will enable LVGL to draw the next part of the screen to the other buffer while
     *      the data is being sent form the first buffer. It makes rendering and flushing parallel.
[... 107 lines cut ...]
//{
//    /*It's an example code which should be done by your GPU*/
//    int32_t x, y;
//    dest_buf += dest_width * fill_area->y1; /*Go to the first line*/
//
//    for(y = fill_area->y1; y <= fill_area->y2; y++) {
//        for(x = fill_area->x1; x <= fill_area->x2; x++) {
//            dest_buf[x] = color;
//        }
//        dest_buf+=dest_width;    /*Go to the next line*/
//    }
//}
void my_rounder_cb(struct _lv_disp_drv_t * disp_drv, lv_area_t * area){

    area->x1 = (area->x1/2)*2;
    area->y1 = (area->y1/2)*2;
    area->x2 = (area->x2/2)*2+1;
    area->y2 = (area->y2/2)*2+1;
}

#else /*Enable this file at the top*/

/*This dummy typedef exists purely to silence -Wpedantic.*/
typedef int keep_pedantic_happy;
#endif`

LianKaFoHaLi, 2022-01-25:
This is driver ic requirement.
[image]
[image]

MaJerle, 2022-01-25:
Use triple ` for code posting. https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

LianKaFoHaLi, 2022-01-25:
> Use triple ` for code posting. https://www.markdownguide.org/extended-syntax/#fenced-code-blocks
[@file lv_port_disp_templ.md](https://github.com/lvgl/lvgl/files/7931272/%40file.lv_port_disp_templ.md)
I add code to md file

kisvegabor, 2022-01-25:
Please edit your post, to format the code as @MaJerle suggested.

LianKaFoHaLi, 2022-01-25:
```c
/**
 * @file lv_port_disp_templ.c
 *
 */

 /*Copy this file as "lv_port_disp.c" and set this value to "1" to enable content*/
#if 1

/*********************
 *      INCLUDES
 *********************/
#include "lv_port_disp.h"
#include "lvgl.h"
// #include "userdef.h"
/*********************
 *      DEFINES
 *********************/
#define MY_DISP_HOR_RES 126
#define MY_DISP_VER_RES 294  

/**********************
 *      TYPEDEFS
 **********************/

/**********************
 *  STATIC PROTOTYPES
 **********************/
static void disp_init(void);

static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p);
//static void gpu_fill(lv_disp_drv_t * disp_drv, lv_color_t * dest_buf, lv_coord_t dest_width,
//        const lv_area_t * fill_area, lv_color_t color);

/**********************
 *  STATIC VARIABLES
 **********************/

/**********************
 *      MACROS
 **********************/

/**********************
 *   GLOBAL FUNCTIONS
 **********************/

void lv_port_disp_init(void)
{
    /*-------------------------
     * Initialize your display
     * -----------------------*/
    disp_init();

    /*-----------------------------
     * Create a buffer for drawing
     *----------------------------*/

    /**
     * LVGL requires a buffer where it internally draws the widgets.
     * Later this buffer will passed to your display driver's `flush_cb` to copy its content to your display.
     * The buffer has to be greater than 1 display row
     *
     * There are 3 buffering configurations:
     * 1. Create ONE buffer:
     *      LVGL will draw the display's content here and writes it to your display
     *
     * 2. Create TWO buffer:
     *      LVGL will draw the display's content to a buffer and writes it your display.
     *      You should use DMA to write the buffer's content to the display.
     *      It will enable LVGL to draw the next part of the screen to the other buffer while
[... 109 lines cut ...]
//    /*It's an example code which should be done by your GPU*/
//    int32_t x, y;
//    dest_buf += dest_width * fill_area->y1; /*Go to the first line*/
//
//    for(y = fill_area->y1; y <= fill_area->y2; y++) {
//        for(x = fill_area->x1; x <= fill_area->x2; x++) {
//            dest_buf[x] = color;
//        }
//        dest_buf+=dest_width;    /*Go to the next line*/
//    }
//}
void my_rounder_cb(struct _lv_disp_drv_t * disp_drv, lv_area_t * area){

    area->x1 = (area->x1/2)*2;
    area->y1 = (area->y1/2)*2;
    area->x2 = (area->x2/2)*2+1;
    area->y2 = (area->y2/2)*2+1;
}

#else /*Enable this file at the top*/

/*This dummy typedef exists purely to silence -Wpedantic.*/
typedef int keep_pedantic_happy;
#endif
```

LianKaFoHaLi, 2022-01-25:
> Please edit your post, to format the code as @MaJerle suggested.

Sorry,you can see the code in my new comment.

LianKaFoHaLi, 2022-01-25:
‘’‘
[18:01:38.529]收←◆[INFO ] main: FreeRTOS V10.2.1
[INFO ] main: Fw Version: v0.7.0000
[INFO ] StartAppTask: Running...

[18:01:38.784]收←◆[INFO ] disp_flush: area ( 0  0 16 293)
[INFO ] disp_flush: area (17  0 33 293)
[INFO ] disp_flush: area (50  0 66 293)
[INFO ] disp_flush: area (67  0 83 293)
[INFO ] disp_flush: area (100  0 116 293)
[INFO ] disp_flush: area (117  0 125 293)

[18:01:40.780]收←◆[INFO ] disp_flush: area ( 0  0 16 293)
[INFO ] disp_flush: area (17  0 33 293)
[INFO ] disp_flush: area (50  0 66 293)
[INFO ] disp_flush: area (67  0 83 293)
[INFO ] disp_flush: area (100  0 116 293)
[INFO ] disp_flush: area (117  0 125 293)

[18:01:51.062]收←◆[INFO ] disp_flush: area ( 0  0 16 293)
[INFO ] disp_flush: area (17  0 33 293)

[18:01:51.085]收←◆[INFO ] disp_flush: area (50  0 66 293)
[INFO ] disp_flush: area (67  0 83 293)
[INFO ] disp_flush: area (100  0 116 293)
[INFO ] disp_flush: area (117  0 125 293)

[18:01:52.561]收←◆[INFO ] disp_flush: area ( 0  0 16 293)
[INFO ] disp_flush: area (17  0 33 293)
[INFO ] disp_flush: area (50  0 66 293)
[INFO ] disp_flush: area (67  0 83 293)
[INFO ] disp_flush: area (100  0 116 293)
[INFO ] disp_flush: area (117  0 125 293)

[18:01:53.548]收←◆[INFO ] disp_flush: area ( 0  0 16 293)
[INFO ] disp_flush: area (17  0 33 293)
[INFO ] disp_flush: area (50  0 66 293)
[INFO ] disp_flush: area (67  0 83 293)
[INFO ] disp_flush: area (100  0 116 293)
[INFO ] disp_flush: area (117  0 125 293)

[18:01:55.046]收←◆[INFO ] disp_flush: area ( 0  0 16 293)
[INFO ] disp_flush: area (17  0 33 293)
[INFO ] disp_flush: area (50  0 66 293)
[INFO ] disp_flush: area (67  0 83 293)
’‘’
this is some refresh events that don't fit the requirement
[image]
[image]

LianKaFoHaLi, 2022-01-26:
@kisvegabor  @MaJerle 
Hi,I find the reason that why refresh area don't fit the requirement .
because of the application requirement,GUI need 90° rotation.Yesterday,@glin00 said he has been sloved his problem by adding _rounder_cb()_ ,But his lvgl version is 7.6.1. So,I compared lvgl 8.1.0  that about _rounder_cb()_  code with 7.6.1,the result
is there is no change in this piece between the two versions. Today, I suspect the rotate caused the problem.So I canceled the rotation code,just as below code showing.
```
void lv_port_disp_init(void)
{
    /*-------------------------
     * Initialize your display
     * -----------------------*/
    disp_init();

    /*-----------------------------
     * Create a buffer for drawing
     *----------------------------*/

    /**
     * LVGL requires a buffer where it internally draws the widgets.
     * Later this buffer will passed to your display driver's `flush_cb` to copy its content to your display.
     * The buffer has to be greater than 1 display row
     *
     * There are 3 buffering configurations:
     * 1. Create ONE buffer:
     *      LVGL will draw the display's content here and writes it to your display
     *
     * 2. Create TWO buffer:
     *      LVGL will draw the display's content to a buffer and writes it your display.
     *      You should use DMA to write the buffer's content to the display.
     *      It will enable LVGL to draw the next part of the screen to the other buffer while
     *      the data is being sent form the first buffer. It makes rendering and flushing parallel.
     *
     * 3. Double buffering
     *      Set 2 screens sized buffers and set disp_drv.full_refresh = 1.
     *      This way LVGL will always provide the whole rendered screen in `flush_cb`
     *      and you only need to change the frame buffer's address.
     */

    /* Example for 1) */
    static lv_disp_draw_buf_t draw_buf_dsc_1;
    static lv_color_t buf_1[MY_DISP_HOR_RES*120];                          /*A buffer for 10 rows*/
    lv_disp_draw_buf_init(&draw_buf_dsc_1, buf_1, NULL, MY_DISP_HOR_RES*120);   /*Initialize the display buffer*/

    /* Example for 2) */
    // static lv_disp_draw_buf_t draw_buf_dsc_2;
    // static lv_color_t buf_2_1[MY_DISP_HOR_RES * 60];                        /*A buffer for 10 rows*/
    // static lv_color_t buf_2_2[MY_DISP_HOR_RES * 60];                        /*An other buffer for 10 rows*/
    // lv_disp_draw_buf_init(&draw_buf_dsc_2, buf_2_1, buf_2_2, MY_DISP_HOR_RES * 60);   /*Initialize the display buffer*/

    /* Example for 3) also set disp_drv.full_refresh = 1 below*/
    // static lv_disp_draw_buf_t draw_buf_dsc_3;
    // static lv_color_t buf_3_1[MY_DISP_HOR_RES * MY_DISP_VER_RES];            /*A screen sized buffer*/
    // static lv_color_t buf_3_2[MY_DISP_HOR_RES * MY_DISP_VER_RES];            /*An other screen sized buffer*/
    // lv_disp_draw_buf_init(&draw_buf_dsc_3, buf_3_1, buf_3_2, MY_DISP_HOR_RES * MY_DISP_VER_RES );   /*Initialize the display buffer*/

    /*-----------------------------------
     * Register the display in LVGL
     *----------------------------------*/

    static lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
    lv_disp_drv_init(&disp_drv);                    /*Basic initialization*/

    /*Set up the functions to access to your display*/

    /*Set the resolution of the display*/
    disp_drv.hor_res = 126;
    disp_drv.ver_res = 294;

    /*Used to copy the buffer's content to the display*/
    disp_drv.flush_cb = disp_flush;

    /*Set the screen Rotation*/
    // disp_drv.sw_rotate=1;
    // disp_drv.rotated=LV_DISP_ROT_90;

    /*Set a display buffer*/
    disp_drv.draw_buf = &draw_buf_dsc_1;

    disp_drv.rounder_cb = my_rounder_cb;
    /*Required for Example 3)*/
    // disp_drv.full_refresh = 1;

    /* Fill a memory array with a color if you have GPU.
     * Note that, in lv_conf.h you can enable GPUs that has built-in support in LVGL.
     * But if you have a different GPU you can use with this callback.*/
    //disp_drv.gpu_fill_cb = gpu_fill;

    /*Finally register the driver*/
    lv_disp_drv_register(&disp_drv);
}
```
```
static void disp_flush(lv_disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p)
{
    /*The most simple case (but also the slowest) to put all pixels to the screen one-by-one*/

    // int32_t x;
    // int32_t y;
    // for(y = area->y1; y <= area->y2; y++) {
    //     for(x = area->x1; x <= area->x2; x++) {
    //         /*Put a pixel to the display. For example:*/
    //         /*put_px(x, y, *color_p)*/
    //         color_p++;
    //     }
    // }
    if((area->x1%2!=0||area->y1%2!=0)||(area->x2%2!=1||area->y2%2!=1)){
        oled_write_buf(area->x1,area->y1,area->x2,area->y2,(oled_color16_t*)color_p,disp_drv);
        LOGI("area (%2d %2d %2d %2d)",area->x1,area->y1,area->x2,area->y2);
        // lv_disp_flush_ready(disp_drv);
    }else{
        oled_write_buf_refresh(area->x1,area->y1,area->x2,area->y2,(oled_color16_t*)color_p,disp_drv);
    }
    // oled_Write_color(area->x1,area->y1,area->x2,area->y2,(oled_color16_t*)color_p);
    // oled_Write_Color_Dma(area->x1,area->y1,area->x2,area->y2,(oled_color16_t*)color_p,disp_drv);
    // oled_write_buf(area->x1,area->y1,area->x2,area->y2,(oled_color16_t*)color_p,disp_drv);
    // oled_write_buf_dma(area->x1,area->y1,area->x2,area->y2,(oled_color16_t*)color_p,disp_drv);
    /*IMPORTANT!!!
     *Inform the graphics library that you are ready with the flushing*/
    // lv_disp_flush_ready(disp_drv);
}
```

**Accoring to the  log showing,refresh area are all fit the screen requirement,that meaning function _my_rounder_cb()_ take effect.**
[image]
[image]

kisvegabor, 2022-01-28:
Please also print the "input" and the modified area from the `rounder_cb` to see it get's called and calculates properly.

LianKaFoHaLi, 2022-02-12:
The problem has been solved. The reason is display buffer size is less than _LV_DISP_ROT_MAX_BUF_ size when enabling software rotation.

closed 2022-02-12 by LianKaFoHaLi

h3lpd3sk, 2023-03-29:
@LianKaFoHaLi Sorry for cross posting but any chance you're willing to share your RM69310 driver code? 
Really struggeling to find any references on that ic 👎