lvgl/lvgl #3049: Text optimised "hello World" example

ammaree, 2022-01-25:
### Perform all steps below and tick them with [x]
- [ x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I am trying to create a basic template for displaying TEXT ONLY on an ESP32 WROVER DevKit v4.1
I have tried starting with one of examples and trying to reduce it to just display text with minimal RAM & FLASH requirements.
Can you please provide some guidelines, preferably a "Hello World" example that we can modify
 
### To Reproduce

### Expected behavior

### Screenshots or video

embeddedt, 2022-01-25:
I haven't compiled this but it should work.

```c
lv_obj_t * label = lv_label_create(lv_scr_act());
lv_label_set_text(label, "Hello world");
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
```

----------------------
We use GitHub issues for development-related discussions.
Please use the [forum](https://forum.lvgl.io/) to ask questions.

closed 2022-01-25 by embeddedt

ammaree, 2022-01-26:
Thanks for the feedback @embeddedt but I am afraid I need a bit more help.
I am using ESP IDF master with "lvgl" and "lvgl_esp32_drivers" master repos, but get errors relating to missing LV_HOR_RES_MAX definitions and various initialisation functions.

Is it possible to get a short code segment to initialise lvl for the ESP32. I need to display about 2000 bytes of text listing tasks, priorities, stack space and related variables? This text is simply printed to a buffer but instead of sending it out via UART need to display on LCD, refreshed every 500-1000mSecs.

kisvegabor, 2022-01-27:
@ammaree 
If's related to ESP32 please open an issue in https://github.com/lvgl/lv_port_esp32