lvgl/lvgl #3051: Change the drop-down box to Chinese, and the drop-down arrow is gone

RiceChen0, 2022-01-27:


LianKaFoHaLi, 2022-01-27:
Did you add Chinese font to your project?

RiceChen0, 2022-01-27:
> Did you add Chinese font to your project?

Yes, my project needs to switch between Chinese and English. There are arrows in English mode, but no arrows in Chinese mode

LianKaFoHaLi, 2022-01-27:
Maybe your Chinese font doesn't have LV_SYMBOY_DOWN icon font.
[image]
You can use _lv_dropdown_set_symbol()_ to define a custom image.
```
    /*Use a custom image as down icon and flip it when the list is opened*/
    LV_IMG_DECLARE(img_caret_down)
    lv_dropdown_set_symbol(dropdown, &img_caret_down);
```

RiceChen0, 2022-01-27:
I still don't know how to deal with it. Could you please provide an example

LianKaFoHaLi, 2022-01-27:
May be you can refer to this.
[https://docs.lvgl.io/master/widgets/core/dropdown.html](Drop down list) 
[image]

embeddedt, 2022-01-27:
FYI, `img_caret_down` is located [in `examples/assets`](https://github.com/lvgl/lvgl/blob/master/examples/assets/img_caret_down.c).

kisvegabor, 2022-01-27:
Have you added the symbols to your Chinese font as described [here](https://lvgl.io/tools/fontconverter)?

RiceChen0, 2022-01-28:
Thanks for your support, I have solved the problem

closed 2022-01-28 by RiceChen0

hayschan, 2023-03-08:
> Thanks for your support, I have solved the problem

How did you solve the problem?