lvgl/lvgl #3052: Attach the NXP PXP/VGLITE GPUs back to the new draw context (architecture changes)

nicusorcitu, 2022-01-27:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Draw context architecture changed and now the GPU rendering is done by implementing the supported acceleration callbacks.
The NXP GPUs hooks are gone and we need to add the support back.

### To Reproduce
NXP GPUs support is currently broken on the master branch.

### Expected behavior
NXP PXP or VGLITE can be enabled and used as accelerators.

nicusorcitu, 2022-01-27:
I am currently working on this issue, the PXP attachment is complete and tested.

kisvegabor, 2022-01-27:
Thank you for opening this issue and working on it. 
Feel free to send a PR so that I can also try it out.

gokhannsahin, 2022-01-27:
I wonder why PXP and VGlite are limited to 16bits and why rotating is not used with vglite? The performance of IMXRT1170 is lower than expected because the gpu drivers poor.

gokhannsahin, 2022-02-01:
Also, using CPU to rotate performance is hungry, why don't use VGLite to rotate any image?

nicusorcitu, 2022-02-07:
Thank you for your interest to the NXP products. We were about to update the GPU with more functionalities but now we have to rework the patches due to the new architecture.

We will add soon the support for ARGB 32 bit color format on both PXP and VGLITE, hopefully at the end of this ticket. Also the image rotation is in progress.

We will continue to add even more GPU acceleration support so that the performance gets better and better.

gokhannsahin, 2022-02-08:
That is good news.
Also, multilayer features can be added default so it provides very good performance using LCDifv2 alpha blendig.

nicusorcitu, 2022-02-15:
We are also studying if chroma key, alpha blending, multi-layers alpha blending can be accelerated using LCDIFv2.

kisvegabor, 2022-03-02:
@nicusorcitu 
What is the current state of the new drivers?

gokhannsahin, 2022-03-03:
I'm looking forward to your reply. Please upgrade the driver for new version because we don't use it in v8.2.

nicusorcitu, 2022-03-04:
The GPU support is back online and we added some extra widget acceleration features. All these changes are currently on our local NXP GTEC repo.

In these days I am doing a clear set of unit tests to validate all the changes, next week I will complete them.

Next I will start the procedure to push the GPU support to upstream. The ETA for upstream is the end of March, this process is taking a while because I need to fallow an even secured mechanism now. These days the upstream to github might also suffer some changes to prevent any security breach.

embeddedt, 2022-03-29:
Not stale.

nicusorcitu, 2022-03-29:
The upstream is still in progress.

jupeos, 2022-05-03:
What's the latest on this?
I'm waiting for this to be complete before I upgrade from V7 to V8.
Using IMXRT1064

nicusorcitu, 2022-05-04:
I finally got the approval for this patch set. I will create the PR soon, hopefully by the end of this week.

nicusorcitu, 2022-05-04:
I just opened the PR
https://github.com/lvgl/lvgl/pull/3322

jupeos, 2022-05-05:
> I just opened the PR #3322

Fantastic! Can't wait to try it out.

kisvegabor, 2022-05-28:
As the PR is merged I close this issue.

Thank you @nicusorcitu!

closed 2022-11-14 by kisvegabor