lvgl/lvgl #3053: need scroll to fixed coordinates when node is create

FarawayRay, 2022-01-27:
When I need to create a long page, but I set the page length to be fixed, the page will automatically add scrolling when the child node exceeds the display area. If I create something and immediately jump to a certain location, the page has to scroll to a certain place and it doesn't move the page

example:
`    lv_obj_t* cont = lv_obj_create(lv_scr_act());
    lv_obj_set_size(cont, 280, 456);
    lv_obj_set_flex_flow(cont, LV_FLEX_FLOW_COLUMN);
    for (int i = 0; i < 20; i++) {
        lv_obj_t* child = lv_obj_create(cont);
        lv_obj_set_size(child, 250, 40);
    }

    lv_obj_set_scrollbar_mode(cont, LV_SCROLLBAR_MODE_OFF);
    lv_obj_scroll_to_y(cont, 500, LV_ANIM_OFF);`

question:
[image]

When a node is created, I need to jump to a certain location immediately. What do I do?

thank you

embeddedt, 2022-01-27:
Try calling `lv_obj_update_layout` on `cont` first. [Here](https://docs.lvgl.io/master/overview/coords.html#postponed-coordinate-calculation) is a docs page with information on why this may help.

FarawayRay, 2022-01-27:
Thank you for your support

closed 2022-01-27 by FarawayRay