lvgl/lvgl #3057: default LV_MEM_SIZE is too small to run demo: widgets

GorgonMeducer, 2022-01-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The default value of LV_MEM_SIZE in the 'lv_conf_template.h' is too small to run our iconical demo, i.e. Widgets with a typical resolution 320 * 240. It requires 35KBytes. 

We have to update the demos/README.md.
Except for that, should we consider increase the default value ?

### To Reproduce

### Expected behavior

### Screenshots or video

kisvegabor, 2022-01-29:
I agree to increase the default `LV_MEM_SIZE`. To sure I suggest 48k.

GorgonMeducer, 2022-01-29:
OK. Will generate a PR.

kisvegabor, 2022-01-29:
Feel free to update #3032

closed 2022-01-29 by GorgonMeducer